Igalia/meta-webkit #1: raspberrypi2 support?

c0ffee, 2015-12-07:
sadly the build failed on the rpi2, is this platform supported?

i can provide more infos on request

clopez, 2015-12-08:
I didn't test the build on the rpi2.

Do you have a log of where it failed?

clopez, 2016-02-16:
Can you share your local.conf and bblayers.conf files?
Also provide info about what versions (branches) of each layer you were using?
Do you have any log of where it failed?

indisoluble, 2016-09-08:
Hi all,

I have also had some issues trying to build an image with this layer for RaspberryPi, more exactly for a **RaspberryPi 3**, although I finally managed to build one. This is going to be a long comment, so my apologies on advance.

First, I tried to compile an image with the **WebKitGTK+ engine** using the following configuration:
```
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:12eb72ee3b02f826a156ff4e396c770f2b93571e"
meta-raspberrypi  = "master:2745399f75d7564fcc586d0365ff73be47849d0e"
meta-webkit       = "master:63bf58b406d91dccbf58747bbb5d6e3c9a66b86a"
```
I also added to the end of my `local.conf`:
```
MACHINE ?= "raspberrypi3"
DISTRO_FEATURES_append = " opengl x11"
IMAGE_INSTALL_append = " webkitgtk-bin"
```

However, when I executed `bitbake core-image-sato`, I got the following error:
```
/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore/platform/graphics/GLContext.cpp:135:83: error: invalid conversion from 'GLNativeWindowType {aka long long unsigned int}' to 'EGLNativeWindowType {aka void*}' [-fpermissive]
     if (auto eglContext = GLContextEGL::createContext(windowHandle, sharingContext))
                                                                                   ^
In file included from /path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore/platform/graphics/GLContext.cpp:29:0:
/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore/platform/graphics/egl/GLContextEGL.h:38:42: note:   initializing argument 1 of 'static std::unique_ptr<WebCore::GLContextEGL> WebCore::GLContextEGL::createContext(EGLNativeWindowType, WebCore::GLContext*)'
     static std::unique_ptr<GLContextEGL> createContext(EGLNativeWindowType, GLContext* sharingContext = 0);
                                          ^
[ 62%] Building CXX object Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/GraphicsContext3DPrivate.cpp.o
Source/WebCore/CMakeFiles/WebCore.dir/build.make:58510: recipe for target 'Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/GLContext.cpp.o' failed
make[2]: *** [Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/GLContext.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cd /path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/build/Source/WebCore &&/path/to/yocto/build/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -DBUILDING_GTK__=1 -DBUILDING_WEBKIT -DBUILDING_WITH_CMAKE=1 -DBUILDING_WebCore -DDATA_DIR=\"/usr/share\" -DGETTEXT_PACKAGE=\"WebKit2GTK-4.0\" -DHAVE_CONFIG_H=1 -DMESA_EGL_NO_X11_HEADERS -DUSER_AGENT_GTK_MAJOR_VERSION=602 -DUSER_AGENT_GTK_MINOR_VERSION=1 -DUSE_VCHIQ_ARM -DWEBKITGTK_API_VERSION_STRING=\"4.0\" -DWEBKIT_DOM_USE_UNSTABLE_API -I/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore -I/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore/Modules/airplay ... -fPIC  -o CMakeFiles/WebCore.dir/platform/graphics/GraphicsContext3DPrivate.cpp.o -c /path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/webkitgtk-2.10.7/Source/WebCore/platform/graphics/GraphicsContext3DPrivate.cpp
make[2]: Leaving directory '/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/build'
CMakeFiles/Makefile2:607: recipe for target 'Source/WebCore/CMakeFiles/WebCore.dir/all' failed
make[1]: *** [Source/WebCore/CMakeFiles/WebCore.dir/all] Error 2
make[1]: Leaving directory '/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/build'
Makefile:152: recipe for target 'all' failed
make: *** [all] Error 2
ERROR: oe_runmake failed
ERROR: Function failed: do_compile (log file is located at/path/to/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/webkitgtk/2.10.7-r0/temp/log.do_compile.9545)
```
As far as I can see, I think the problem is located in `GLContext.h`:
```
#if USE(EGL) && !PLATFORM(GTK)
#include "eglplatform.h"
typedef EGLNativeWindowType GLNativeWindowType;
#elif PLATFORM(GTK) && PLATFORM(WAYLAND) && !defined(GTK_API_VERSION_2)
#include <wayland-egl.h>
#include <EGL/eglplatform.h>
typedef EGLNativeWindowType GLNativeWindowType;
#else
typedef uint64_t GLNativeWindowType;
#endif
```
`GLNativeWindowType` is set to `uint64_t`, instead of `EGLNativeWindowType`.

Now, on the bright side it is possible to build an image for **WebKitForWayland engine** with this same build configuration, however you have to add to the end of `local.conf`:
```
MACHINE ?= "raspberrypi3"
[... 14 lines cut ...]
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.2"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard vfpv4 cortexa7"
[... cut ...]
meta-raspberrypi  = "master:cc64d6324d1543f009f0c010d720e74b7cb9f5d5"
meta-webkit       = "jethro:a107c901f93d453f886a3014224ca2430d15fcf3"
```
I also had to  override some preferred providers in `local.conf`:
```
MACHINE ?= "raspberrypi3"
DISTRO_FEATURES_append = " opengl x11"
IMAGE_INSTALL_append = " webkitgtk-bin"
PREFERRED_PROVIDER_virtual/libgl = "mesa"
PREFERRED_PROVIDER_virtual/libgles2 = "mesa"
PREFERRED_PROVIDER_virtual/egl = "mesa"
PREFERRED_PROVIDER_virtual/mesa = "mesa"
```

Hope this is of any help.

clopez, 2016-09-08:
Regarding webkitforwayland on the raspberrypi I think you will have better luck using the dispmanx backend rather than the wayland one. 
At least until the new free mesa-based vc4 drivers for the raspberry are in better shape.
With this dispmanx backend ,w4w renders directly to the framebuffer (no display server needed) using the raspberry-specific dispmanx APIs.
The support for this is not on the main w4w repository, but on the metrological one. I have a working recipe for Yocto, just let me know if you are interested and I can upload it.

indisoluble, 2016-09-08:
About `opengl`, the answer is no, it does not work. I loaded the page you commented before and `MiniBrowser` is not able to show it.

Also, thank you for offering to upload another recipe, I do appreciate it. But, I built an image for wayland just out of curiosity; at least for the moment, I have to keep using X11.

Let me ask you one more question, should I build my image using `webgl` instead of `opengl`? Would I be able to render the page you mentioned before with `webgl`?

Thanks a lot.

clopez, 2016-09-08:
I think opengl (or webgl) with webkit is not going to work on the rpi with the standard drivers (the propietary ones).
Maybe with the new ones (vc4) it could work, but i don't know. I tried it once without luck.

To get a browser that works ok on that machine the best thing is just to disable opengl at all. Most websites work perfectly without it.

I have just uploaded a patch (68ad3d23526d5d57176222139464616c37770000) that adds an option to disable opengl on webkitgtk.

After pulling this new patch, add this on your local.conf:
```
PACKAGECONFIG_pn-webkitgtk = "x11"
```
And rebuild. It should build without any opengl support.

clopez, 2016-09-08:
Btw.. if you want wayland instead of x11, use:
```
PACKAGECONFIG_pn-webkitgtk = "wayland"
```

You can build support for both at the same time also

```
PACKAGECONFIG_pn-webkitgtk = "x11 wayland"
```

indisoluble, 2016-09-08:
Thanks! I'll try :smiley:

clopez, 2018-05-16:
I'm closing this issue.

There have been many changes on master and this issue is not longer relevant.
WPE (former WebKitForWayland) is now building fine on the RPi and i.MX6 boards (at least with rocko)
I added also documentation on the [wiki](https://github.com/Igalia/meta-webkit/wiki) about it, including a [detailed step-by-step-guide for the RPi (RPi2 or RPi3)](https://github.com/Igalia/meta-webkit/wiki/RPi)

Please open a new issue if you still have any trouble with the recipes in this layer.

closed 2018-05-16 by clopez