Igalia/meta-webkit #2: Obsolete

jsarenik, 2016-02-15:
Please write a note in README of this repo that `webkit` is in `openembedded-core/meta/recipes-sato/webkit` - that's a maintained one.

Thanks.

jsarenik, 2016-02-15:
https://github.com/openembedded/openembedded-core (this is the main RO mirror with better bandwidth, GitHub is not used for PullRequests)

clopez, 2016-02-15:
Hi, there is already a note about this on the README file:

    The name of the recipe is the same than the one available in oe-core (master), so you should select which version of webkitgtk you want to build in your local.conf

Although I agree the note can be a bit clearer. I will improve it.

Also, this meta layer is mainly meant for the following purposes:

 * Allowing people to build WebKit2GTK+ on older branches of Yocto/OE (for example: fido, etc).
 * Allowing people to build WebKit4Wayland (there is no recipe for this in oe-core).

clopez, 2016-02-16:
Clarified in 7b8f1226ae1116265e9102dd565c2c49c8e2c6b9

closed 2016-02-16 by clopez