Igalia/meta-webkit #3: no known conversion for argument 1 from 'gbm_surface* const' to 'EGLNativeWindowType {aka wl_egl_window*}'

4urele, 2016-04-25:
Hello

compilation on yocto 1.8 (fido) ends with the following logs (sry for the long paths): 
If anyone knows the issue, please let me know.

Best regards
Aurèle

| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_progress_report /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/build/CMakeFiles
| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/gbm/GBMSurface.cpp: In member function 'std::unique_ptr<WebCore::GLContextEGL> WebCore::GBMSurface::createGLContext() const':
| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/gbm/GBMSurface.cpp:65:84: error: no matching function for call to 'WebCore::GLContextEGL::createWindowContext(gbm_surface* const&, WebCore::GLContext*)'
|      return GLContextEGL::createWindowContext(m_surface, GLContext::sharingContext());
|                                                                                     ^
| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/gbm/GBMSurface.cpp:65:84: note: candidate is:
| In file included from /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/gbm/GBMSurface.cpp:31:0:
| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/egl/GLContextEGL.h:39:42: note: static std::unique_ptr<WebCore::GLContextEGL> WebCore::GLContextEGL::createWindowContext(EGLNativeWindowType, WebCore::GLContext*, std::unique_ptr<WebCore::GLContext::Data>&&)
|      static std::unique_ptr<GLContextEGL> createWindowContext(EGLNativeWindowType, GLContext* sharingContext, std::unique_ptr<GLContext::Data>&& = nullptr);
|                                           ^
| /home/aurele/workspace/yocto/moniteur-fido/build-moniteur2/tmp-glibc/work/cortexa9hf-vfp-neon-mx6qdl-oe-linux-gnueabi/webkitforwayland/wpe-20151019-r0/webkit-wpe-20151019/Source/WebCore/platform/graphics/egl/GLContextEGL.h:39:42: note:   no known conversion for argument 1 from 'gbm_surface* const' to 'EGLNativeWindowType {aka wl_egl_window*}'

ps Thanks for this layer

clopez, 2016-05-30:
It looks there is some problem  with your OpenGL/EGL libraries and WebkitforWayland...

Some questions:
 * Which target (machine/board) where you trying to compile this for?
 * Which GPU drivers and OpenGL libraries are you using?

4urele, 2016-06-20:
Hi 

First of all, thanks for your answer, and sorry to be late...
I'm building for a custom board based on a freescale IMX6 (something wich looks like the imx6 sabre)
I'm using the GPU drivers from freescale

Maybe I could try to open the subject on the freescale forum

Thanks
Aurèle

woutervanh, 2018-02-20:
I have the same issue, also on a imx6-board. Using morty as base.
Same situation as 4urele. GPU drivers from freescale are imx-gpu-viv (vivante drivers)

clopez, 2018-02-20:
> I have the same issue, also on a imx6-board. Using morty as base.
> Same situation as 4urele. GPU drivers from freescale are imx-gpu-viv (vivante drivers)

Which recipe are you building? webkitgtk?

woutervanh, 2018-02-20:
Well, to be complete:
I'm compiling core-image-weston. My distro is fslc-wayland
In my local.conf, I added:
> DISTRO_FEATURES_append = " opengl wayland"
> IMAGE_INSTALL_append = " wpebackend-rdk wpebackend-mesa wpelauncher wpewebkit"
> 
> IMAGE_INSTALL_append = " imx-gpu-viv"

And it's since I added the wpebackend-mesa, I have this issue.

The first error popping up was:

> | In file included from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/imx6qsabresd/usr/include/EGL/eglplatform.h:38:0,
> |                  from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/imx6qsabresd/usr/include/wpe/renderer-backend-egl.h:30,
> |                  from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-gbm.h:32,
> |                  from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/mesa.cpp:36:
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/imx6qsabresd/usr/include/EGL/eglvivante.h:270:2: error: #error "Platform not recognized"
> |  #error "Platform not recognized"
> |   ^~~~~
> | CMakeFiles/WPEBackend-mesa.dir/build.make:65: recipe for target 'CMakeFiles/WPEBackend-mesa.dir/src/mesa.cpp.o' failed
> | make[2]: *** [CMakeFiles/WPEBackend-mesa.dir/src/mesa.cpp.o] Error 1
> | make[2]: *** Waiting for unfinished jobs....
> | make[2]: Leaving directory '/opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/build'
> | CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/WPEBackend-mesa.dir/all' failed
> | make[1]: *** [CMakeFiles/WPEBackend-mesa.dir/all] Error 2
> | make[1]: Leaving directory '/opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/build'
> | Makefile:130: recipe for target 'all' failed
> | make: *** [all] Error 2
> | ERROR: Function failed: do_compile (log file is located at /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/temp/log.do_compile.61815)

Following patch gave a hint:
http://lists.busybox.net/pipermail/buildroot/2014-October/107598.html
So I added #define EGL_API_FB and #define LINUX on top of the eglvivante.h header

This error went away and gave me following error:

> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp: In lambda function:
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp:197:24: error: cannot convert 'gbm_surface*' to 'EGLNativeWindowType {aka _FBWindow*}' in return
> |          return target->surface;
> |                         ^~~~~~~
> | In file included from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp:34:0:
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/imx6qsabresd/usr/include/gbm.h:48:8: note: class type 'gbm_surface' is incomplete
> |  struct gbm_surface;
> |         ^~~~~~~~~~~
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp: In lambda function:
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp:267:24: error: cannot convert 'gbm_surface*' to 'EGLNativeWindowType {aka _FBWindow*}' in return
> |          return target->surface;
> |                         ^~~~~~~
> | In file included from /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/git/src/gbm/renderer-backend-egl-gbm.cpp:34:0:
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/imx6qsabresd/usr/include/gbm.h:48:8: note: class type 'gbm_surface' is incomplete
> |  struct gbm_surface;
> |         ^~~~~~~~~~~
> | CMakeFiles/WPEBackend-mesa.dir/build.make:281: recipe for target 'CMakeFiles/WPEBackend-mesa.dir/src/gbm/renderer-backend-egl-gbm.cpp.o' failed
> | make[2]: *** [CMakeFiles/WPEBackend-mesa.dir/src/gbm/renderer-backend-egl-gbm.cpp.o] Error 1
> | make[2]: *** Waiting for unfinished jobs....
> | make[2]: Leaving directory '/opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/build'
> | CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/WPEBackend-mesa.dir/all' failed
> | make[1]: *** [CMakeFiles/WPEBackend-mesa.dir/all] Error 2
> | make[1]: Leaving directory '/opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/build'
> | ERROR: oe_runmake failed
> | Makefile:130: recipe for target 'all' failed
> | make: *** [all] Error 2
> | WARNING: exit code 1 from a shell command.
> | ERROR: Function failed: do_compile (log file is located at /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpebackend-mesa/0.1-r0/temp/log.do_compile.69940)
> 
> 

And that's where I'm stuck now...

Mesa implementation sould be provided by imx-gpu-viv (vivante)

clopez, 2018-02-20:
Ok.. the error comes from wpebackend-mesa which is not going to work with the Vivante drivers in any case. For IMX you need to select wpebackend-rdk (instead of wpebackend-mesa).

If you want to test WPE on IMX I suggest the following:

 * (optional) Switch to the branch krogoth of this repository (its currently the most tested one for IMX)
 * (optional) If you want video support add also this layer to your bblayers.conf: https://github.com/clopez/meta-gstreamer1.0/ and also add the bbmask line indicated on the README.md
 * Add this to local.conf
```
IMAGE_INSTALL_append = " wpebackend-rdk wpelauncher wpewebkit wpewebkit-web-inspector-plugin"
IMAGE_INSTALL_append = " gstreamer1.0-plugins-base gstreamer1.0-plugins-base-meta"
IMAGE_INSTALL_append = " gstreamer1.0-plugins-good gstreamer1.0-plugins-good-meta"
IMAGE_INSTALL_append = " gstreamer1.0-plugins-bad  gstreamer1.0-plugins-bad-meta"
IMAGE_INSTALL_append = " gstreamer1.0-plugins-imx  gstreamer1.0-plugins-imx-meta"
PACKAGECONFIG_pn-wpewebkit = "deviceorientation fullscreenapi fetchapi gamepad geolocation indexeddb libinput mediasource notifications sampling-profiler shadowdom subtitle subtlecrypto udev video webaudio wayland touch gst_gl logs"
```

Once you have built WPE, you launch it from inside weston like:
```
WPELauncher http://google.com
or
WPELauncher file:///path/to/file.html
```
Its important to indicate the ```file:// ``` prefix if you want to load a local resource.
To change the window resolution use this env vars before starting WPELauncher
```
export WPE_INIT_VIEW_HEIGHT=800
export WPE_INIT_VIEW_WIDTH=480
```

woutervanh, 2018-02-20:
Great, thanks, I'll try that.
Seems you have experience with imx6 at a whole? Will the rdk-backend have drawbacks compared to mesa-backend in performance? What performance can I expect?

clopez, 2018-02-20:
> Great, thanks, I'll try that.
> Seems you have experience with imx6 at a whole? Will the rdk-backend have drawbacks compared to mesa-backend in performance? What performance can I expect?

mesa-backend requires libgbm (it requires /dev/dri/renderD128 to exist), which is not available with the IMX.6 Vivante driver. However, mesa-backend may work with the open source driver (Etnaviv) as it implements GBM I think, but I have not tried.

wpebackend-rdk has multiple backends inside, which are crafted for different crappy/embedded/propietary drivers. The one I'm using currently on IMX is the wayland-egl backend inside wpebackend-rdk. This backend works with any driver that provides just wayland-egl without any facility for cross-process buffer-sharing, but it does so by disabling buffer sharing between WPE-UIProcess and WPE-WebProcess by making the WPE-WebProcess render directly. It works fine for the use case of "just one webview without decorations from the UIProcess (like a navigation bar)", but it may not work for other use cases.

There is a new backend implementation (fdo which is an acronym from FreeDesktopOrg https://github.com/Igalia/WPEBackend-fdo ) which may also work with the Vivante driver and in theory it should manage the buffer sharing properly, but i have still neither packaged here neither tested it on IMX.

woutervanh, 2018-02-21:
Ok, so now I checked out the krogoth branch of this repo. (I suppose I can stay with morty for the rest of my yocto-environment?) I also had to change some layer-names, as I'm using the imx community bsp (meta-freescale in stead of fscl-arm-)
However, I bounce on following error (which I suppose to be a webkit fault actually):

> | DEBUG: Executing shell function do_compile
> | [1/1378] Generate bindings (WebCoreBindings)
> | [2/530] cd /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/build/Source/WebKit2 && /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/x86_64-linux/usr/bin/perl-native/perl /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/git/Source/WebKit2/Scripts/generate-forwarding-headers.pl --include-path /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/git/Source/WebKit2 --output /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/build/DerivedSources/ForwardingHeaders --platform wpe --platform soup
> | [3/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebCompiledContentExtensionData.cpp.o
> | [4/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebContextMenuItem.cpp.o
> | [5/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebEvent.cpp.o
> | [6/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o
> | FAILED: Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/sysroots/x86_64-linux/usr/bin/arm-fslc-linux-gnueabi/arm-fslc-linux-gnueabi-g++     @Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o.rsp -MD -MT Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o -MF Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o.d -o Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnection.cpp.o -c /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/git/Source/WebKit2/Shared/WebConnection.cpp
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/git/Source/WebKit2/Shared/WebConnection.cpp: In member function 'void WebKit::WebConnection::postMessage(const WTF::String&, API::Object*)':
> | /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/git/Source/WebKit2/Shared/WebConnection.cpp:55:10: error: 'Messages' has not been declared
> |      send(Messages::WebConnection::HandleMessage(messageName, UserData(transformObjectsToHandles(messageBody))));
> |           ^~~~~~~~
> | At global scope:
> | cc1plus: warning: unrecognized command line option '-Wno-expansion-to-defined'
> | [7/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebConnectionClient.cpp.o
> | [8/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebEventConversion.cpp.o
> | [9/530] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/Shared/WebCoreArgumentCoders.cpp.o
> | ninja: build stopped: subcommand failed.
> | WARNING: exit code 1 from a shell command.
> | ERROR: Function failed: do_compile (log file is located at /opt/yocto/fsl-community-bsp_morty/build-imx6qbsp/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/wpewebkit/1.20180205-r0/temp/log.do_compile.7598)

clopez, 2018-02-21:
> Ok, so now I checked out the krogoth branch of this repo. (I suppose I can stay with morty for the rest of my yocto-environment?) I also had to change some layer-names, as I'm using the imx [...]

mmm.. strange build error... perhaps its compiler related? that version of WPE is already a bit old and I'm not testing it with other GCC than 4.9.
Can you try with GCC-4.9? set this on local.conf and rebuild everything:
```
GCCVERSION = "4.9%"
```

I plan to commit an updated WPE to master in the coming weeks

woutervanh, 2018-02-27:
It was working after a -c cleansstate.
I suppose something got messed up because of a unwanted powercut of my pc

woutervanh, 2018-02-28:
Now it seam to work. Only the performance seems rather slow. I'm running on a 1920x1080 screen. And I never saw a fps above 30... Is that normal? Can this be increased on a imx6? (It's the dual)

clopez, 2018-02-28:
> Now it seam to work. Only the performance seems rather slow. I'm running on a 1920x1080 screen. And I never saw a fps above 30... Is that normal? Can this be increased on a imx6? (It's the dual)

How do you test FPS?

Can you try to open this webpage https://webkit.org/blog-files/3d-transforms/poster-circle.html and start
WPELauncher with this environment variable below so it prints FPS to the terminal?
```
export WPE_DISPLAY_FPS=1
```

woutervanh, 2018-02-28:
I just tested it via some demo pages which print the fps
Result of your test:

> [WPEInjectedBundle] Initialized.
> didClearWindowObjectForFrame
> [WPELauncher] Hello InjectedBundle ...
> [WPEInjectedBundle] Hello!
>         array item: Test1
>         array item: Test2
>         array item: Test3
> [WPELauncher] 16.91 FPS
> [WPELauncher] 27.06 FPS
> [WPELauncher] 27.05 FPS
> [WPELauncher] 27.05 FPS
> [WPELauncher] 27.07 FPS
> [WPELauncher] 27.06 FPS
> [WPELauncher] 27.02 FPS
> [WPELauncher] 26.89 FPS
> [WPELauncher] 26.90 FPS

clopez, 2018-02-28:
On an I.MX6 board I have at hand I get 40FPS on the poster circle demo. But 60FPS should be the target. It seems there is still quite room for optimization on this platform. 

The new version of WPE we will be committing soon likely has better performance

clopez, 2018-02-28:
Perhaps its related to the screen refresh rate? if vsync is working ok (which should) then its normal it doesn't generate more FPS than the screen can actually deliver. Try running the test eglgears from mesa-demos

woutervanh, 2018-02-28:
refresh rate of the screen is 60Hz

mode "1920x1080-60"
        # D: 148.500 MHz, H: 67.500 kHz, V: 60.000 Hz
        geometry 1920 1080 1920 2160 16
        timings 6734 148 88 36 4 44 5
        accel false
        rgba 5/11,6/5,5/0,0/0
endmode

woutervanh, 2018-02-28:
Sorry for drifting away a bit from wpe-webkit itself, do you mind? But I just discovered that, if I run the weston-simple-egl  -f -b (in which -f means fullscreen, -b don't sync to compositor redraw (eglSwapInterval 0)), I get a stunning 175 fps!!!
So it's in some way the weston compositor which is limiting the redraw frequency

woutervanh, 2018-03-06:
Hello, Now I wanted to add video, and added the repo you mentioned, including the line in local.conf.
However, I get issues and almost everything gets removed from my recipe...

> WARNING: No bb files matched BBFILE_PATTERN_core '^/opt/yocto/fsl-community-bsp_morty/sources/poky/meta/'
> WARNING: No bb files matched BBFILE_PATTERN_yocto '^/opt/yocto/fsl-community-bsp_morty/sources/poky/meta-poky/'
> WARNING: No bb files matched BBFILE_PATTERN_openembedded-layer '^/opt/yocto/fsl-community-bsp_morty/sources/meta-openembedded/meta-oe/'
> WARNING: No bb files matched BBFILE_PATTERN_multimedia-layer '^/opt/yocto/fsl-community-bsp_morty/sources/meta-openembedded/meta-multimedia/'
> WARNING: No bb files matched BBFILE_PATTERN_networking-layer '^/opt/yocto/fsl-community-bsp_morty/sources/meta-openembedded/meta-networking/'
> WARNING: No bb files matched BBFILE_PATTERN_meta-python '^/opt/yocto/fsl-community-bsp_morty/sources/meta-openembedded/meta-python/'
> WARNING: No bb files matched BBFILE_PATTERN_freescale-layer '^/opt/yocto/fsl-community-bsp_morty/sources/meta-freescale/'
> WARNING: No bb files matched BBFILE_PATTERN_fsl-arm-extra '^/opt/yocto/fsl-community-bsp_morty/sources/meta-freescale-3rdparty/'
> WARNING: No bb files matched BBFILE_PATTERN_fsl-demos '^/opt/yocto/fsl-community-bsp_morty/sources/meta-freescale-distro/'
> WARNING: No bb files matched BBFILE_PATTERN_webkit '^/opt/yocto/fsl-community-bsp_morty/sources/meta-webkit/'
> WARNING: No bb files matched BBFILE_PATTERN_gstreamer1.0 '^/opt/yocto/fsl-community-bsp_morty/sources/meta-gstreamer1.0/'
> NOTE: There are 3 recipes to be removed from sysroot x86_64_x86_64_linux, removing...
> NOTE: There are 3 recipes to be removed from sysroot x86_64_arm, removing...
> NOTE: There are 68 recipes to be removed from sysroot x86_64_linux, removing...
> NOTE: There are 14 recipes to be removed from sysroot allarch, removing...
> NOTE: There are 173 recipes to be removed from sysroot armv7at2hf-neon, removing...
> NOTE: There are 228 recipes to be removed from sysroot imx6qsabresd, removing...

Without adding this line, video doesn't play (or crashes)  either. I get some warnings on the console about appsrc

clopez, 2018-03-06:
> Hello, Now I wanted to add video, and added the repo you mentioned, including the line in local.conf.
> However, I get issues and almost everything gets removed from my recipe...

That's likely because that repo and conf was only tried and tested with Yocto krogoth and will not work with new yocto releases out of the box.

Hopefully we will have updated recipes for WPE here that work with last Yocto version soon (no specific ETA, but I bet they should be done before H2-2018)

woutervanh, 2018-03-06:
I see...
And what's the added value of this layer? As I just had a quick view, I seem to have most (if not all) of it provided by the meta-freescale layer already. Would I be able to debug why video is not working fine by now? Codecs and freescale-specific sinks are there... How does wpewebkit chooses what to use?

clopez, 2018-03-06:
> And what's the added value of this layer? As I just had a quick view, I seem to have most (if not all) of it provided by the meta-freescale layer already. 

Do you ask about the value of https://github.com/clopez/meta-gstreamer1.0?
Its just to provide updated gstreamer packages for Yocto Krogoth that are needed for WPE on IMX. 

To get accelerated video decoding with WPE on IMX basically you need:
1. gstreamer-plugins-bad 1.12.x (with opengl plugins built)
2. gstreamer-imx (last master)
3. build WPE with gstreamer-gl option enabled.

Note that for 1. I'm using the official gstreamer-plugins-bad and not the freescale fork.

> Would I be able to debug why video is not working fine by now? Codecs and freescale-specific sinks are there... 

I don't know about your background, but certainly is possible to debug why things don't work.
I have tested this on Yocto krogoth with the layer https://github.com/clopez/meta-gstreamer1.0 and that's why I know that it works there, so that's why I tell you to use that version.
I won't provide free consultancy to fix things on your version of Yocto. I'm already pretty busy. I hope you understand :)

> How does wpewebkit chooses what to use?

When WPE is built with gstreamer-gl it decodes with the decoder that has the higest rank (imxvpudec should be in this case) and connects that with the glupload element (from gstreamer-plugins-bad) to render the video in an OpenGL surface.

woutervanh, 2018-03-07:
> I don't know about your background, but certainly is possible to debug why things don't work.
> I have tested this on Yocto krogoth with the layer https://github.com/clopez/meta-gstreamer1.0 and [...]
Ok, well, it is playing some kind, but very slow. And if I look at the logging, I see it uses imxvpudec and webkit-gl-video-sink, which creates a context on vivante GC2000. But I don't see why it's running that bad. It's also eating up my (very limited) diskspace. It's creating huge ~/.cache/WPEWebProcess* files. How can I disable cache on disk completely?

> I won't provide free consultancy to fix things on your version of Yocto. I'm already pretty busy. I hope you understand :)
Off course I understand :-). I really appreciate all the help you already provided!

> > How does wpewebkit chooses what to use?
> When WPE is built with gstreamer-gl it decodes with the decoder that has the higest rank (imxvpudec should be in this case) and connects that with the glupload element (from gstreamer-plugins-bad) to render the video in an OpenGL surface.

Looks like that is indeed the case, only with very bad performance...
BTW: I'm currently trying to implement a wpe egl-backend for the Vivante stack, based on the one for bcm/rpi. So that I can avoid wayland/weston at a whole. Don't know if I'll succeed or have time to finish it, but if I do, you'll hear about it ;-)

clopez, 2018-03-07:
>Ok, well, it is playing some kind, but very slow. 
> And if I look at the logging, I see it uses imxvpudec and webkit-gl-video-sink, which creates a context on vivante GC2000.
> But I don't see why it's running that bad. 

Maybe the problem is not in WPE but on GStreamer or on the hardware ?.
I'm unsure if all i.MX6 hardware has the same VPU capable of accelerated video decoding.

Does playing the same video with GStreamer directly with gst-launch-1.0 work fine?

> It's also eating up my (very limited) diskspace. It's creating huge ~/.cache/WPEWebProcess* files. How can I disable cache on disk completely?

I think there are som API settings that allow to disable it. I'm unsure about the details of that now.

> BTW: I'm currently trying to implement a wpe egl-backend for the Vivante stack, based on the one for bcm/rpi. So that I can avoid wayland/weston at a whole. Don't know
> if I'll succeed or have time to finish it, but if I do, you'll hear about it ;-)

That sounds great!
Feel free to send a PR here to add the recipes for it :)

woutervanh, 2018-03-21:
I have a imx6qd, which has its dedicated vpudecoder and a few dedicated capable videosinks. It plays fine with a plain gst-launch1.0... In the browser however, it has issues.

I just launched you a PR with the addition of the wpe egl-backend for vivante/imx6 I made

clopez, 2018-04-12:
> But I don't know if it's really a wpe issue...
> Also with the fullscreen "weston-simple-egl", I only get
> 
>     139 frames in 5 seconds: 27.799999 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
>     136 frames in 5 seconds: 27.200001 fps
> 
> So I don't know what influences this. It's wayland/weston 1.11 if that matters...
> I found https://community.nxp.com/thread/391972 , but is rather old without solution...

I found that the low FPS may be caused because the default weston repaint window interval (7ms) is too low for WPE, and that causes that WPE sometimes fail to deliver the frame in time.

You can open this page with WPE that is a simple benchmark of ```requestAnimationFrame()``` https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html 
If you are already getting 60FPS there then ok, but if you are getting a lower value, then try to modify the weston repaint-window value. To do that edit your weston.ini file and add something like this
```
[core]
repaint-window=15
```
and restart weston. The value is in ms. The best value should be something between 7 and 1000/60. Try with 14 also.

Doing that changes allowed WPE to reach 60FPS with weston and the wayland backend on the imx6 board I have here (vivante drivers).

woutervanh, 2018-04-12:
Hey Carlos,
I already did those tests, and indeed, I was able to improve the fps doing so.
But in the end, I just dropped complete wayland/weston and created the egl vivante-backend.
I ended up having other issues. A plain memory webprocess crash. See also:
https://github.com/WebPlatformForEmbedded/WPEWebKit/issues/461
I have it on either wayland and viv-imx6-backend...
Maybe you can have a look at it, as you have apparently a imx6 board at your side at the moment? It would already be a help just to know I'm not the only one :-)

clopez, 2018-05-16:
I'm closing this issue.

There have been many changes on master and this issue is not longer relevant.
WPE is now building fine on the RPi and i.MX6 boards (at least with rocko)
I added also documentation on the [https://github.com/Igalia/meta-webkit/wiki](wiki) about it.

Please open a new issue if you still have any trouble with the recipes in this layer.

closed 2018-05-16 by clopez