Igalia/meta-webkit #12: dyz build failure

macpijan, 2018-05-14:
Hello.
I'm building for i.MX6 board (`meta-freescale`, rocko revision). Though it looks to me that this issue is not board-dependent. I'm building `core-image-weston`.

I switched to latest master commit in the `dyz` recipe (https://github.com/Igalia/dyz/commit/197fed66831ff2e5b16173662b48cae8476c1b72) as suggested in the: https://github.com/Igalia/meta-webkit/pull/10 in order to avoid issue with finding the `luajit` binary. 

Now the error is in finding LUA jit modules:

```
luajit -bg -a arm -n apps.singleview.main apps/singleview/main.lua apps/singleview/main_lua.o
luajit -bg -a arm -n entrypoint entrypoint.lua entrypoint_lua.o
luajit -bg -a arm -n lib.glib lib/glib.lua lib/glib_lua.o
luajit -bg -a arm -n lib.wpebackend lib/wpebackend.lua lib/wpebackend_lua.o
luajit: unknown luaJIT command or jit.* modules not installed
```

I can see some of modules are in the recipe sysroot:

```
./recipe-sysroot-native/usr/share/luajit-2.0.5/jit
./recipe-sysroot/usr/share/luajit-2.0.5/jit
```

```
bc.lua  bcsave.lua  dis_arm.lua  dis_mipsel.lua  dis_mips.lua  dis_ppc.lua  dis_x64.lua  dis_x86.lua  dump.lua  v.lua  vmdef.lua
```

I don't really know `lua` and this app to tell if those are we need there and how to point in the Makefile to find it. Any hints a re  appreciated.

clopez, 2018-05-14:
The new version of dyz will depend on wpebackend-fdo.
There is another launcher (cog, previously know as dinghy) that should support all backends.
And there is currently a WIP branch here named fdo-test that contains updated recipes for dyz and cog.
I have also wrote some doc on the wiki about all of this. Check:
https://github.com/Igalia/meta-webkit/wiki/WPE
https://github.com/Igalia/meta-webkit/wiki/RPi

I will be cherry-picking commits from this fdo-test branch into master once I can give them the deserved QA, and I will be improving as well the documentation on the wiki about this.

So, If possible test this with "fdo-test" branch and let me know if you have any issue

clopez, 2018-05-14:
Also (still WIP): https://github.com/Igalia/meta-webkit/wiki/i.MX6

macpijan, 2018-05-14:
@clopez Thanks a lot for those valuable information. I'm using mainline stack (Etnaviv, CODA ...). I can post the results once I got something.

macpijan, 2018-05-15:
@clopez 
The issue for `dyz` failure is no longer valid with the `fdo-test` branch: https://github.com/Igalia/meta-webkit/tree/fdo-test  so I believe that we can close this issue. I will open another one for potential further discussion.

clopez, 2018-05-15:
Most of the changes from fdo-test are already merged on master. Closing

closed 2018-05-15 by clopez