Igalia/meta-webkit #13: i.MX6 opensource stack support

macpijan, 2018-05-16:
Hi,

I'm curious whether anyone has been testing `wpewebkit` with imx6 (opensource stack).

I'm using latest NXP Yocto BSP (rocko). In addition, I have switched to
openembedded-core (poky/meta) master in order to get latest mesa (18.0.2
instead of 17.1 as in rocko NXP BSP). As claimed in the following article, mesa
before 17.1 may not work well with Etnaviv / Wayland:

https://www.pengutronix.de/en/2017-09-28-etnaviv-weston-mainline.html

My configuration:

```
## webkit configuration ##
IMAGE_INSTALL_append = " wpewebkit cog wpebackend"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"

PACKAGECONFIG_pn-wpebackend-rdk = "wayland"

PACKAGECONFIG_pn-wpewebkit = " \
    fetchapi \
    indexeddb \
    webaudio \
    "

# use mainline graphics stack in NXP BSP
MACHINEOVERRIDES .= ":use-mainline-bsp"
PREFERRED_PROVIDER_virtual/kernel = "linux-fslc"
```

My results so far:

* [Igalia fps counter](https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html)
  on `cog` with `rdk/wayland` backend gives maximum of 26-27 FPS with 15ms
  interval in `wayland.ini`

* with `fdo` backend the display image is scrambled as shown on the image (both `cog` and `dyz` launchers):

[image]

Let me know if this is fine spot for such discussion.

clopez, 2018-05-16:
So far I have been testing WPE on i.MX6 with the Vivante drivers mainly. 

I'm adding @psaavedra and @philn to the CC that have tested WPE on i.MX6 with Etnaviv recently to see if they have any suggestion.

philn, 2018-05-16:
I haven't tried the rdk backend with etnaviv. Only the wpebackend-fdo, and it works :)

macpijan, 2018-05-16:
@philn Thanks for your answer. I'm glad it does. So I believe that I should focus on bringing the `wpebackend-fdo`  to operation.  I'm wondering if you guys have any notes of the configuration you were testing against (yocto layers branches, kernel, hardware, webkit / launcher configuration or anything). I can see that the `meta-webkit`'s conponents (as well as `meta-webkit` itself) is under quite rapid development and it is not an easy task to jump in and get started.

elima, 2018-05-16:
@macpijan, you need [linux-dmabuf](https://github.com/Igalia/WPEBackend-fdo/commits/linux-dmabuf) of WPEBackend-fdo on etnaviv. However, cog also needs changes to enable the dmabuf interface, I will write a patch for it.

Regarding the low FPS, did you mean you set "repaint-window=15" in *weston.ini* (not wayland.ini) ?

elima, 2018-05-16:
@macpijan, I have created a cog [linux-dmabuf](https://github.com/Igalia/cog/commits/linux-dmabuf) branch which, together with its counterpart in -fdo, should get the scrambled image fixed.

I tested it quickly in my laptop and should be fine, but please let me know if you find issues.

macpijan, 2018-05-16:
Thanks for such a rapid response @elima 
I've built the image with cog and `fdo` backend from `linux-dmabuf` branches. It does display properly now (scrambled display is gone). However, the performance seems to be the same as in case of the `rdk` backend.
It's around 20FPS for 5ms repaint and 25-27FPS for 15ms repaint.

And yes, I believe that I have modified the correct file:

* proof of loading the config:

```
root@cubox-i:~# cat /var/log/weston.log | grep config
[14:39:52.087] Using config file '/etc/xdg/weston/weston.ini'
```

* proof of loading the value from config:

```
root@cubox-i:~# cat /var/log/weston.log  | grep paint 
[14:39:52.088] Output repaint window is 15 ms maximum.
```

The hardware uses i.MX6DL SoC

philn, 2018-05-16:
I haven't tested this SoC, here I use a QuadPlus.

clopez, 2018-05-16:
> I've built the image with cog and fdo backend from linux-dmabuf branches. It does display properly now (scrambled display is gone). However, the performance seems to be the same as in case of the rdk backend.
> It's around 20FPS for 5ms repaint and 25-27FPS for 15ms repaint.

Have you checked that other apps running over weston can get 60 FPS? es2_gears / eglgears from package mesa-demos or glmark2 can be good tests to check in order to be sure that the issue is not on Weston or on the screen refresh rate configuration itself.

macpijan, 2018-05-17:
@clopez 

Actually, I have tried running the mesa backend already as those device files are present in case of Etnaviv.
However, I got following error. I can try finding out where does this symbol come from.

```
(cog:660): Cog-WARNING **: 07:30:16.136: Could not load: libcogplatform-mesa.so (possible cause: No such file or directory).

wpe: could not load the impl library. Is there any backend installed?: /usr/lib/libWPEBackend-default.so: undefined symbol: glDrawArrays
```

And some results from the `glmark2`:

DRM:

```
=======================================================
    glmark2 2017.07
=======================================================
    OpenGL Information
    GL_VENDOR:     etnaviv
    GL_RENDERER:   Vivante GC880 rev 5106
    GL_VERSION:    OpenGL ES 2.0 Mesa 18.0.2
=======================================================
[build] use-vbo=false: FPS: 58 FrameTime: 17.241 ms
[build] use-vbo=true: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=nearest: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=linear: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=mipmap: FPS: 60 FrameTime: 16.667 ms
[shading] shading=gouraud: FPS: 60 FrameTime: 16.667 ms
[shading] shading=blinn-phong-inf: FPS: 30 FrameTime: 33.333 ms
[shading] shading=phong: FPS: 30 FrameTime: 33.333 ms
[shading] shading=cel: FPS: 20 FrameTime: 50.000 ms
[bump] bump-render=high-poly: FPS: 30 FrameTime: 33.333 ms
[bump] bump-render=normals: FPS: 50 FrameTime: 20.000 ms
[bump] bump-render=height: FPS: 30 FrameTime: 33.333 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 10 FrameTime: 100.000 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 3 FrameTime: 333.333 ms
[pulsar] light=false:quads=5:texture=false: FPS: 43 FrameTime: 23.256 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 4 FrameTime: 250.000 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] effect=shadow:windows=4: FPS: 20 FrameTime: 50.000 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 20 FrameTime: 50.000 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 20 FrameTime: 50.000 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 20 FrameTime: 50.000 ms
[ideas] speed=duration: FPS: 31 FrameTime: 32.258 ms
[jellyfish] <default>: FPS: 12 FrameTime: 83.333 ms
[terrain] <default>:error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
 FPS: 0 FrameTime: inf ms
[shadow] <default>: FPS: 14 FrameTime: 71.429 ms
[... 52 lines cut ...]
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
error: compile failed!
etna_draw_vbo:221: compiled shaders are not okay
 FPS: 0 FrameTime: inf ms
[shadow] <default>: FPS: 10 FrameTime: 100.000 ms
[refract] <default>: FPS: 3 FrameTime: 333.333 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 35 FrameTime: 28.571 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 12 FrameTime: 83.333 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 37 FrameTime: 27.027 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 22 FrameTime: 45.455 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 12 FrameTime: 83.333 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 21 FrameTime: 47.619 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 21 FrameTime: 47.619 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 12 FrameTime: 83.333 ms
=======================================================
                                  glmark2 Score: 22
=======================================================
```

Generally, the first tests give 60FPS on the DRM, and there is a significant delay while running over Wayland / Weston (repaint-window=15)

macpijan, 2018-05-17:
`wepbackend-mesa` can be run with Etnaviv, but it needs to be linked against `GLESv2` library (in current state it is not). The result is scrambled display as on the first picture. Performance on weston in console log  is roughly the same.

It cannot be run without Weston:

```
root@cubox-i:~# env
EDITOR=vi
HZ=100
HUSHLOGIN=FALSE
USER=root
PWD=/home/root
HOME=/home/root
SHELL=/bin/sh
TERM=vt102
SHLVL=1
LOGNAME=root
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
PS1=\u@\h:\w\$ 
_=/usr/bin/env
root@cubox-i:~# ./mesa.xh 

(cog:675): Cog-WARNING **: 10:40:31.420: Could not load: libcogplatform-mesa.so (possible cause: No such file or directory).

Cog-Message: 10:40:31.747: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Load started.

(cog:675): Cog-WARNING **: 10:40:31.782: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 10:40:32.216: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Load started.
Cog-Message: 10:40:32.232: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Loading...

(cog:675): Cog-WARNING **: 10:40:32.244: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 10:40:32.491: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Load started.
Cog-Message: 10:40:32.500: <https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html> Loading...
```

macpijan, 2018-05-18:
@clopez My last log shows that I was not able to run it without `weston` so far.  As you can see there are no `WESTON_*` variables in the env. Running the `cog` (`cog -P mesa -v https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html`) results in above erros repeating in a loop. Nothing is displayed on the screen.

clopez, 2018-05-18:
@macpijan What's inside mesa.xh script? You don't need to pass any platform parameter to cog. Simply run it like `cog https://google.com`
Can you try to export the environment variable `G_MESSAGES_DEBUG=all` before running cog? that way it will print debug messages

clopez, 2018-05-18:
BTW.. check also that on your mesa build, you have the `PACKAGECONFIG` options `gbm`,  `dri` and `dri3` enabled.

macpijan, 2018-05-21:
@clopez The scipt's content is basically the one line which runs cog provided in my previous post.

mesa configuration is as follows: `PACKAGECONFIG_append_pn-mesa = "etnaviv imx gallium dri egl gles xa gbm"`
Are you sure about the `dri` and `dri3` stuff. I can see that they both depend on `xorg`. Actually, it is not possible to make a build with `dri3` enabled without `X11` in `DISTRO_FEATURES` due to `libxshmfence` dependency.

Thanks for the debug switch info. I will sure try it out.

clopez, 2018-05-21:
@macpijan I'm not sure about dri and dri3 being strictly needed. But sure it won't hurt.

macpijan, 2018-05-22:
When trying to tun over `mesa` backend without the `Weston`, debug messages does give too much additional info:

```
root@cubox-i:~# G_MESSAGES_DEBUG=all cog https://google.com
(cog:498): Cog-DEBUG: 08:26:40.279: platform_setup: Platform name: (null)
(cog:498): Cog-DEBUG: 08:26:40.281: Instantiating default WPE backend as fall-back.
Cog-Message: 08:26:40.607: <https://google.com/> Load started.

(cog:498): Cog-WARNING **: 08:26:40.652: <https://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 08:26:41.082: <https://google.com/> Load started.
Cog-Message: 08:26:41.095: <https://google.com/> Loading...

(cog:498): Cog-WARNING **: 08:26:41.109: <https://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 08:26:41.360: <https://google.com/> Load started.
Cog-Message: 08:26:41.368: <https://google.com/> Loading...

(cog:498): Cog-WARNING **: 08:26:41.391: <https://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 08:26:41.649: <https://google.com/> Load started.
Cog-Message: 08:26:41.657: <https://google.com/> Loading...
```

clopez, 2018-05-22:
Ok. It looks like the backend crashed or msething like that. Maybe it still doesn't work with etnaviv without wayland. Not sure. I guess running the process with `strace -f` or generating a coredump and a backtrace can help to know where its the issue.

ch-f, 2018-10-07:
Hi,

 I stumbled across the same issue: a scrambled image of cog (looks the same as the screenshot above).

The etnaviv weston open-source-stack is working fine here.

This is the wpewebkit config:
```

IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```

So I tried to use linux-dmabuf branches of repos wpebackend-fdo and cog. But they are currently outdated in regards to the other repos from current meta-webkit and don't compile:

wpebackend-fdo (branch linux-dmabuf):
<snip>
```
| -- Checking for modules 'wayland-client;wayland-server'
| --   Found wayland-client, version 1.14.0
| --   Found wayland-server, version 1.14.0
| -- Found WAYLAND: wayland-client;wayland-server
| -- Checking for module 'wayland-egl'
| --   Found wayland-egl, version 17.1.7
| -- Found WAYLAND_EGL: wayland-egl;wayland-client
| -- Could NOT find WPEBACKEND (missing:  WPEBACKEND_INCLUDE_DIRS WPEBACKEND_LIBRARIES)
| CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
| Please set them or make sure they are set and tested correctly in the CMake files:
```
</snip>

What's a fast way to solve this and get wpe working on etnaviv?

I also created a rather long strace from wpebackend-mesa which can be found here:
https://paste.ubuntu.com/p/KG2HCYJSZH/

shiftee, 2018-10-07:
Try adding wpebackend-fdo to IMAGE_INSTALL.

I got this working with Poky master (eventually).
With poky master I didn't need any dmabuf specific changes but I did make the following changes:

 recipes-core/images/core-image-minimal.bbappend 
>  IMAGE_INSTALL += "wayland weston weston-init"
> DISTRO_FEATURES_append = " wayland"
> 
> DISTRO_FEATURES_append = " opengl"
> 
> IMAGE_INSTALL += "cog"
> 
> IMAGE_INSTALL += "wpewebkit"
> 
> IMAGE_INSTALL += "wpebackend-fdo"
> 
> DISTRO_FEATURES_remove = "x11"

recipes-graphics/mesa/mesa_%.bbappend
> PACKAGECONFIG_append = " etnaviv imx gallium"

recipes-graphics/wayland/files/weston.ini
> [core]
> 
> idle-time=60
> shell=desktop-shell.so
> 
> [shell]
> locking=false
> background-image=/biotector/www/images/logo_colour.jpg
> client=/biotector/bin/browser_start.sh
> 

 recipes-graphics/wayland/weston_%.bbappend

> FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
> 
> SRC_URI += "file://weston.ini"
> 
> do_install_append() {
>     install -d ${D}${sysconfdir}/xdg/weston
>     install -m 0644 ${WORKDIR}/weston.ini ${D}${sysconfdir}/xdg/weston
> }
> 
> FILES_${PN} += "${sysconfdir}"

ch-f, 2018-10-08:
Thanks for your input shiftee. I'm pretty much stuck on rocko, but I can update/backport individual packages from master. That leads me to my next question: What packages do I need to update to circumvent dmabuf specific changes? Is it mesa 18 instead of 17?

ch-f, 2018-10-09:
that's what I did to get here in the first place

shiftee, 2018-10-09:
I'd try libdrm and the kernel then. I think the dmabuf stuff is now in the master branches of wpebackend-fdo and cog

marex, 2018-10-09:
Just curious, is it possible to build the wpe with wpebackend-mesa and run it on pure drm/kms console instead of having to use wayland ? I managed to build it, but cog indicates the renderer process crashed, which I believe is because cog tries to use the -fdo platform. Is there a launcher for the -mesa platform ?

ch-f, 2018-10-09:
After upgrading mesa, wayland and kernel I now do get a working cog wpewebkit.

But on a 800x600 screen it's pretty slow: 
[http://browserbench.org/MotionMark/](http://browserbench.org/MotionMark/ ) shows a score of 1.00 on all tests. Do you get similar low scores?

Adapting "repaint-window" option in weston.ini doesn't help much. Any ideas?

As marex, I'm also interested in the mesa backend hoping for a performance boost.

marex, 2018-10-10:
Cog seems to have only -fdo platform plugin in it, so it is unlikely to support the -mesa platform.

macpijan, 2018-10-10:
@marex I think it does. I think the reason for that there is no platform for other backends is explained in the wiki: https://github.com/Igalia/meta-webkit/wiki/WPE

```
Contrary to other wpebackends, the support for how the image is presented on the screen (e.g.: wayland) is implemented on the launcher itself (e.g.: on cog or dyz) rather than on the backend.
```

marex, 2018-10-10:
See the sentence right before the text you quoted -- 'Currently the launchers supporting it (cog and dyz) do that requiring a Wayland session with an EGL driver implementing the extension EGL_WL_bind_wayland_display, but support for running directly from a frame-buffer (DRM/KMS) or from X11 may get implemented on the future.'

macpijan, 2018-10-10:
And yet I cannot see from this sentence why the `cog` would not work with backends other than the `fdo`.
What I can tell is that I was running cog with `wpebackend-fdo`, `wpebackend-rdk (both Wayland and imx6 variants`, and `wpebackend-mesa` (Wayland wariant, but image was scrambled as on the first photo in this thread).
As I stated above, the `wpebackend-mesa` on imx6 does not seem to start properly without Weston. You are experiencing the same crash as I was. I was not debugging it any further then, so I do not know what the issue was.

ch-f, 2018-10-10:
After updating the [PerformanceTips wiki](https://github.com/Igalia/meta-webkit/wiki/PerformanceTips) I do get a benchmark score of ~1.26 by [MotionMark](http://browserbench.org/MotionMark/).

My next step is to test video playback.

The following note on [meta-webkit/wiki/i.MX6](https://github.com/Igalia/meta-webkit/wiki/i.MX6) concerns me a bit:

> Note: The instructions above have been only tested with the proprietary stack (Vivante). In theory it should be also possible to enable accelerated hardware decoding using the open source stack by leveraging the gstreamer v4l2 plugin and the kernel CODA driver. The steps for that should be pretty much the same, but installing the v4l2 gstreamer plugin instead of the gstreamer1.0-plugins-imx one.

Has anyone tried this yet?
Any tips, ideas, guidelines?
Are there special cog/wpewebkit/... compile flags I should have an eye on to enable VPU h264 support and play youtube videos?

philn, 2018-10-10:
You might need some GStreamer/Mesa/Kernel patches that are not upstream yet. Anyway a good start would be to enable the gst v4l2 plugin and try playback with `gst-play-1.0`... I'd advise to use the 1.14.1 release at least.

aperezdc, 2018-10-11:
@macpijan Regarding Cog, it should work fine with WPE backends other than FDO.  Just don't pass any `--platform=…` option in the command line, and it will fall-back to use `libWPEBackend-default.so` and using the default method for creating the backend instance (I know it works for the RDK backend, which is the non-FDO one that I have been testing with).

ch-f, 2018-10-16:
regarding video playpack on etnaviv i.mx6:
`gst-launch-1.0 playbin uri=file:/test_h264.avi video-sink=waylandsink`
works and no CPU is used. But a video size of 1080p fails with:

```
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecode
Bin:decodebin0/v4l2h264dec:v4l2h264dec0: Failed to allocate required memory.
```

Playing videos within wpewebkit (cog -P fdo) works now too, but there is no VPU support and videos bigger than 480p are pretty slow.

Any idea how to tell wpewebkit to use gstreamer plugin v4l2h264dec ?

philn, 2018-10-16:
If playbin outside of WPE already auto-plugs v4l2h264dec then playbin inside WPE should also do it. You can check the generated pipeline by setting `GST_DEBUG_DUMP_DOT_DIR=/tmp` for Cog and then convert the dot files in /tmp/ to png.

Does gst-launch with glimagesink instead of waylandsink also work?

psaavedra, 2018-10-24:
> After updating the [PerformanceTips wiki](https://github.com/Igalia/meta-webkit/wiki/PerformanceTips) I do get a benchmark score of ~1.26 by [MotionMark](http://browserbench.org/MotionMark/).
>
> My [...]

Mesa 18 could be fine. You can try with this (mesa 18 can be obtained from `sumo`):

```
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "libwpe"
PREFERRED_VERSION_cog = "git%"
PREFERRED_VERSION_wpewebkit = "2.21%"
PREFERRED_VERSION_wayland = "1.14.0"
PREFERRED_VERSION_wayland-protocols = "1.13"
PREFERRED_VERSION_wpebackend-fdo = "1.0.0"
PREFERRED_VERSION_mesa = "18.%"
```

psaavedra, 2020-04-01:
Two years with no activity. I'm going to close the issue. Feel free to reopen it again in case that you need it.

closed 2020-04-01 by psaavedra