Igalia/meta-webkit #15: compiler error of wpewebkit 2.20.0

13824125580, 2018-06-13:
do you know how to deal with the compiler error on the latest meta-webkit layer ？
2997 /home/caozilong/WorkSpace/yakoto/build/tmp/work/aarch64-poky-linux-musl/wpewebkit/2.20.0-r0/wpewebkit-2.20.0/Source/JavaScriptCore/runtime/SamplingProfiler.cpp:357:48: error: 'framePoint     er' is not a member of 'JSC::MachineContext'                                                                                                                                              
2998     ¦   ¦   ¦   ¦machineFrame = MachineContext::framePointer(registers);

clopez, 2018-06-13:
I have not found that build error.

Which version of GCC are you using (or which version of yocto/poky)?

gil0109, 2018-06-26:
@13824125580 I had something similar and I had to increase my RAM and all built well.

psaavedra, 2018-11-15:
If it is a total memory amount issue, I'd suggest include these settings in the `local.conf`:

```
    # Default to setting automatically based on cpu count
    BB_NUMBER_THREADS = "8"

    # Default to setting automatically based on cpu count
    PARALLEL_MAKE = "-j 8"
```

Note that you need ~2GB per core (aprox.) to get success building WPE.

closed 2018-11-15 by psaavedra