Igalia/meta-webkit #18: MP4 not playing within Webkit

gil0109, 2018-06-26:
We create an MP4 using powerpoint presentation to display on our digital signage.  The MP4 is approx. 700mb.  Webkit launches the webpage but does not play the .mp4.

When I point cog to the mp4, it says loading, then error in the player.  Is there a way to see what the error is?

clopez, 2018-06-26:
Can you play the .mp4 file with the command ```gst-play-1.0``` ?

if not then likely you are missing the decoders for it. If you want software decoders simply install in the image the package gstreamer1.0-libav. If you want hardware-based decoders then it depends on your hardware

gil0109, 2018-06-26:
Yes, it plays great on gst-play-1.0.

clopez, 2018-06-27:
That's strange. 

Can you play the mp4 video on this [test example page](https://people.igalia.com/clopez/wkbug/video/simplevideo.html) with cog? If not, double-check that you are passing the video option in the list of values for the wpewebkit PACKAGECONFIG

gil0109, 2018-06-27:
Yes that one worked. 

Although I can't give the HTML page that opens the MP4, I can give you the url to the mp4.

https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc.mp4

It works on a raspberry pi with Raspbian Stretch and chromium no problem.  

Any chance, we can see the error logs?

clopez, 2018-06-27:
Which wpebackend are you using?

gil0109, 2018-06-27:
This is my local.conf setup:

MACHINE = "raspberrypi3"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"
IMAGE_INSTALL_append = " wpewebkit cog"
DISTRO_FEATURES_append = " opengl wayland"
IMAGE_INSTALL_append = " gstreamer1.0-omx"
LICENSE_FLAGS_WHITELIST_append = " commercial"

clopez, 2018-06-27:
Some comments about the videos themselves ... that is likely unrelated with the issue, but in any case i think is worth commenting:

That videos can be optimized by processing them with the command ```qt-faststart``` to include the header at the start of the file, that will make streaming easier/faster for any player.

You can check the current video is not "streamable"
```
$ wget https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc_small.mp4

$ mediainfo -f sbc.mp4 |grep -i streama
IsStreamable                             : No
```
You can fix that, by processing them with qt-faststart
```
# After running via qt-faststart
$ qt-faststart sbc.mp4 sbc_fast.mp4
ftyp          0 24
uuid         24 40
mdat         64 644076521
moov  644076585 170625
 patching stco atom...
 writing ftyp atom...
 writing moov atom...
 copying rest of file...

$ mediainfo -f sbc_fast.mp4 |grep -i streama
IsStreamable                             : Yes
```

clopez, 2018-06-27:
I have tried to play the videos with cog and wpewebkit and it worked fine for me, but I'm testing on the desktop. On the RPi it may be different as on the RPi you are using the hardware-based gstreamer-omx decoders.

> Any chance, we can see the error logs?

To get the gstreamer error logs you can export this environment variables before starting cog
```
export GST_DEBUG="*:3"
```
To get even more logging, [raise the number as indicated here](https://gstreamer.freedesktop.org/documentation/tutorials/basic/debugging-tools.html#the-debug-log)

gil0109, 2018-06-27:
do you think its because of the size of the rootfs?  Is there an easy way to increase the size?

clopez, 2018-06-27:
> do you think its because of the size of the rootfs? Is there an easy way to increase the size?

Now that you mention it, if there isn't space enough to temporally store the video file it can cause issues.

Try setting on local.conf something like this:

```
# Add 5GB of extra free space to the image
IMAGE_ROOTFS_EXTRA_SPACE = "5242880"
```
Ref: https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#var-IMAGE_ROOTFS_EXTRA_SPACE

gil0109, 2018-06-27:
Converted file and tried again with image including extra rootfs space.   Still doesn't work.

https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc_new.mp4

Any other ideas? 

Here is the debugging output.

[image]

gil0109, 2018-06-27:
Your instincts were right.  That is not the issue.  Problem still exists.
[image]

clopez, 2018-06-27:
Maybe some more useful error gets logged with ```export GST_DEBUG="*:5" ```? (or 4 instead of 5 if that is too verbose) ?

gil0109, 2018-06-27:
Is there an easy way to pipe the log output to a file?

gil0109, 2018-06-27:
I figured how to pipe to file.  I hope this helps...
[karim.txt](https://github.com/Igalia/meta-webkit/files/2139895/karim.txt)

gil0109, 2018-06-27:
Also, tried with weston (Mesa VC drivers) and got this error:

[image]

gil0109, 2018-06-27:
I followed the instructions from the wiki which uses rocko

philn, 2018-06-27:
Can you try with `gst-play-1.0 --videosink glimagesink` too?
If this is a RPi, have you made sure to allocate enough men for the GPU at boot? https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md
Do you use wpebackend-rdk/rpi-dispmanx or wpe-backend-fdo with VC4? I doubt the latter works yet for video playback.

philn, 2018-06-27:
If there are buffering issues this is likely a WebKit issue..
Can you try this? `GST_DEBUG_FILE=gst.log GST_DEBUG_NO_COLOR=1 GST_DEBUG=webkit*:6 cog ...` and attach gst.log here?

philn, 2018-06-27:
Another thing to try, disable this code: https://github.com/WebKit/webkit/blob/master/Source/WebCore/platform/graphics/gstreamer/MediaPlayerPrivateGStreamerBase.cpp#L162-L166

Then another element will be used to read data from the http server, likely souphttpsrc.

philn, 2018-06-27:
https://github.com/WebKit/webkit/blob/master/Source/WebCore/platform/graphics/gstreamer/MediaPlayerPrivateGStreamerBase.cpp#L164 needs to be kept though.

gil0109, 2018-06-27:
GST_DEBUG_FILE=gst.log GST_DEBUG_NO_COLOR=1 GST_DEBUG=webkit*:6 cog results:
[gst.log](https://github.com/Igalia/meta-webkit/files/2142973/gst.log)

For the testing of the changes to the code, I will need to do them in a few hours.

gil0109, 2018-06-27:
Created local video.html file.  Launched with the following command: cog file:video.html.  And it worked fine.

clopez, 2018-06-28:
AFAIK You can't modify the source files directly, you have to create a patch and add it to the recipe in the SRC_URI list

gil0109, 2018-06-28:
Any chance you can walk me through the steps?

guessing I start with: devtool ?

gil0109, 2018-06-28:
I think I figured it out using this: https://wiki.yoctoproject.org/wiki/TipsAndTricks/Patching_the_source_for_a_recipe

I will use this recipe:  devtool modify wpewebkit

gil0109, 2018-06-28:
@clopez 

I made the changes and rebuilding with the following changes, please confirm:

#    GRefPtr<GstElementFactory> srcFactory = adoptGRef(gst_element_factory_find("webkitwebsrc"));
#    if (!srcFactory) {
        GST_DEBUG_CATEGORY_INIT(webkit_media_player_debug, "webkitmediaplayer", 0, "WebKit media playe$
#        gst_element_register(0, "webkitwebsrc", GST_RANK_PRIMARY + 100, WEBKIT_TYPE_WEB_SRC);
#    }

    return true;

gil0109, 2018-06-28:
Would you like me to build with sumo or master instead?

[image]

philn, 2018-06-28:
Comments in C/C++ are either 
`// this is a one-line comment`

or `/* multiple
lines */`

gil0109, 2018-06-28:
@philn @clopez  thanks for the tips.  

Sadly, the removal of those lines did not resolve the issue.  The mp4 player controls that shows up, and instead of staying loading (for 20sec) then error, it goes straight to error.

I appreciate all your help, any ideas what to do next?

gil0109, 2018-06-28:
I also built without GPU Acceleration and that didn't seem to make any difference.

Removing these lines from local.conf:

DISTRO_FEATURES_append = " opengl wayland"
IMAGE_INSTALL_append = " gstreamer1.0-omx"
LICENSE_FLAGS_WHITELIST_append = " commercial"

philn, 2018-06-28:
If an error happens it might appear in the GStreamer logs. Please attach generate a log again.

gil0109, 2018-06-28:
Do you want it with the image with the "fix" and to confirm, I am getting the logs by entering the following command prior to going to the mp4 file:

GST_DEBUG_FILE=gst.log GST_DEBUG_NO_COLOR=1 GST_DEBUG=webkit*:6 cog

gil0109, 2018-06-28:
@philn Here is the log attached
[gst.log](https://github.com/Igalia/meta-webkit/files/2146348/gst.log)

philn, 2018-06-28:
Ah, sorry, forgot to mention this line should also be commented out: https://github.com/WebKit/webkit/blob/master/Source/WebCore/platform/graphics/gstreamer/MediaPlayerPrivateGStreamer.cpp#L237

gil0109, 2018-06-28:
Updated and rebuilt.  This time, the video control didn't show up.  The logs are empty but still no video playing.

gil0109, 2018-06-28:
I went to the page (restricted from the outside) to get details.  Attached is the log for your review @philn 
[gst.log](https://github.com/Igalia/meta-webkit/files/2146930/gst.log)

philn, 2018-06-28:
Logs are empty, really?

gil0109, 2018-06-28:
I know.. it was strange.  But there is no video control display either (with the play button.  I will let it sit for 5 min and check the logs again.

philn, 2018-06-28:
Error 12: Your GStreamer installation is missing a plug-in. (url=https://sbc-qsystem-dev.pathfinder.gov.bc.ca/qsystem/qsmartboard/resource/invite.mp3)

philn, 2018-06-28:
If you try to play an audio file, use `<audio controls ...>` for video... use `<video controls ...>`.

gil0109, 2018-06-28:
The sound playing is from another iframe.  I am not concerned about playing that.  When I do, I will fix it.

The video is in a separate iframe (url: vid2.html).  Note, I pointed to the mp4 directly and it crashed on the screen but still no logs.

philn, 2018-06-28:
Same error for the mp4... Please make sure your GStreamer installation is ok... Use `gst-play-1.0` to check playback outside of webkit.

gil0109, 2018-06-28:
You are correct, I will check my layer.conf and rebuild later today and report back.  Thanks again for your patience.

gil0109, 2018-06-29:
That worked!!  Do I create my own patch, or will you do a pull request to fix it moving forward?

philn, 2018-06-29:
No it was just a test to verify this is a WebKit bug. See https://bugs.webkit.org/show_bug.cgi?id=187171

gil0109, 2018-06-29:
Great, thank you for your help on this.  It's been a pleasure working with you.

gil0109, 2018-06-29:
@philn A couple of followup questions, if you don't mind:
1. Will this issue be closed one the bug fix has been applied to webkit?
2. What is the temp folder that cog uses?  Is there a way to move it as I am making the rootfs read-only

philn, 2018-06-30:
Yes this issue will be closed once the wpe recipe contains the webkit fix. About the cog temp folder, I'm not sure, I opened an issue: https://github.com/Igalia/cog/issues/44

gil0109, 2018-07-02:
@philn I think this issue is resolved.  I rebuild from scratch without modifying lines above and it seems to be working now.  Not sure if they fixed it with another pull request.

gil0109, 2018-07-02:
@philn I rebuilt the image again today and all seems to be working without the changes mentioned above.  Not sure if they fixed it on their end but I am good now.  So I am not sure if we want to close the issue with webkit and this one.  I will leave that up to you.

Up next: I am now trying to run with these variables set:

export XDG_CACHE_HOME="/tmp"
export XDG_DATA_HOME="/tmp"

Which is causing the image to be "corrupt" after about 15sec of playback.  I am thinking it's an issue with my tmpfs size.  Nothing is easy ;)

closed 2018-07-02 by philn

gil0109, 2018-07-02:
@philn Thank you for all your help :)

Ankur498, 2019-05-24:
Please help me on this quite a long time i am struggling with this 
I am getting below error 

root@raspberrypi-rdk-mc:~# gst-launch-1.0 playbin uri=http://35.198.234.156:9003/vr/
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Missing element: text/html decoder
WARNING: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: No decoder available for type 'text/html'.
Additional debug info:
../../../gst-plugins-base-1.10.4/gst/playback/gsturidecodebin.c(938): unknown_type_cb (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: Your GStreamer installation is missing a plug-in.
Additional debug info:
../../../gst-plugins-base-1.10.4/gst/playback/gsturidecodebin.c(1005): no_more_pads_full (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0:
no suitable plugins found:
../../../gst-plugins-base-1.10.4/gst/playback/gstdecodebin2.c(4592): gst_decode_bin_expose (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0:
no suitable plugins found:
Missing decoder: text/html (text/html)

ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
root@raspberrypi-rdk-mc:~#

I tried wpeLauncher also no video is getting loaded .

I am little naive on RDK , please let me know how to enable text/html decoder