Igalia/meta-webkit #22: Cog example to load full screen browser at startup

cchilumbu, 2018-07-05:
Hello,

I am successfully building and booting a core-weston-image, using meta-webkit master branch.

1) How can i get 'cog' to display full screen browser of my custom URL in kiosk mode at boot time?

Currently, I am invoking cog manually, but it does not display full screen at all. 

2) And how can I specify which browser to use? What is the default browser being used?

I am able to launch and access websites like youtube, but there is a fair amount of macro-blocking during playback. I suspect hardware acceleration is not enabled on my x86 target.

3) Is anyone using Chromium here? Is that being provided via meta-browser? 

Any help will be appreciated.

clopez, 2018-07-05:
> Hello,
> 
> I am successfully building and booting a core-weston-image, using meta-webkit master branch.
> 
>     How can i get 'cog' to display full screen browser of my custom URL in kiosk mode at boot time?
> 

That depends on the WPE backend used. Some backends allow configuration of the resolution via environment variables. Check https://github.com/Igalia/meta-webkit/wiki/WPE (read the description of the backends at the bottom, where it talks about the environment variables)

> Currently, I am invoking cog manually, but it does not display full screen at all.
> 
>     And how can I specify which browser to use? What is the default browser being used?
> 

Cog is the browser :)

> I am able to launch and access websites like youtube, but there is a fair amount of macro-blocking during playback. I suspect hardware acceleration is not enabled on my x86 target.
> 
>     Is anyone using Chromium here? Is that being provided via meta-browser?
> 

This layer doesn't provide chromium recipes. If you are looking for Chromium check meta-browser (and if more specifically you are looking for chromium on wayland check our fork of that at https://github.com/Igalia/meta-browser )

> Any help will be appreciated.

cchilumbu, 2018-07-05:
Thank you very much for that clarification.

I am using 'mesa' as the WPE backend because my platform is x86. Is this ok? 

I will look up the link provided to see how that meets my requirements.

But so far, it is looking good :)

clopez, 2018-07-05:
> I am using 'mesa' as the WPE backend because my platform is x86. Is this ok?

For backend mesa, there isn't any environment variable to set the resolution (AFAIK). You would have to patch it.

But with mesa-based graphics (x86), you can use instead of mesa backend the new wpebackend-fdo.
And then with Cog and wpebackend-fdo you can enable full-screen on cog by setting:
```
export COG_PLATFORM_FDO_VIEW_FULLSCREEN=1
```
or any custom resolution with
```
export COG_PLATFORM_FDO_VIEW_WIDTH=1920
export COG_PLATFORM_FDO_VIEW_HEIGHT=1080
```

cchilumbu, 2018-07-09:
Hi Carlos, when i change backend to be 'fdo', i get the following build error:

It builds ok with backend set to 'mesa'.

Any ideas please?

| ERROR: oe_runmake failed
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-gcc  -DCOG_INSIDE_COG__=1 -DG_LOG_DOMAIN=\"Cog\" -Dcogplatform_fdo_EXPORTS -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/build -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/wayland -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/wpe-webkit-0.1 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/libsoup-2.4 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/libxml2 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/glib-2.0 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/lib/glib-2.0/include -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/wpe-fdo-0.1 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/wpe-0.1 -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot/usr/include/libdrm -I/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/core  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2  --sysroot=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0=/usr/src/debug/cog/20180703-r0 -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot=   -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2  --sysroot=/home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/recipe-sysroot -fPIC   -pthread -std=gnu99 -o CMakeFiles/cogplatform-fdo.dir/platform/cog-platform-fdo.c.o   -c /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c: In function 'on_surface_frame':
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:997:9: warning: implicit declaration of function 'wpe_view_backend_exportable_fdo_egl_dispatch_release_image' [-Wimplicit-function-declaration]
|          wpe_view_backend_exportable_fdo_egl_dispatch_release_image (wpe_host_data.exportable,
|          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c: In function 'cog_platform_get_view_backend':
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1356:19: error: variable 'exportable_egl_client' has initializer but incomplete type
|      static struct wpe_view_backend_exportable_fdo_egl_client exportable_egl_client = {
|                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1357:9: error: unknown field 'export_egl_image' specified in initializer
|          .export_egl_image = on_export_egl_image,
|          ^
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1357:29: warning: excess elements in struct initializer
|          .export_egl_image = on_export_egl_image,
|                              ^~~~~~~~~~~~~~~~~~~
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1357:29: note: (near initialization for 'exportable_egl_client')
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1356:62: error: storage size of 'exportable_egl_client' isn't known
|      static struct wpe_view_backend_exportable_fdo_egl_client exportable_egl_client = {
|                                                               ^~~~~~~~~~~~~~~~~~~~~
| /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/git/platform/cog-platform-fdo.c:1361:9: warning: implicit declaration of function 'wpe_view_backend_exportable_fdo_egl_create' [-Wimplicit-function-declaration]
|          wpe_view_backend_exportable_fdo_egl_create (&exportable_egl_client,
[... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/cchilumbu/projects/yocto-dr0081/build-aaeon/tmp/work/corei7-64-poky-linux/cog/20180703-r0/temp/log.do_compile.132131)
ERROR: Task (/home/cchilumbu/projects/yocto-dr0081/sources/meta-webkit/recipes-browser/cog/cog_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5893 tasks of which 5286 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/cchilumbu/projects/yocto-dr0081/sources/meta-webkit/recipes-browser/cog/cog_git.bb:do_compile

cchilumbu, 2018-07-10:
Because of the build failure in trying to use 'wpebackend-fdo', I have reverted to using 'wpebackend-mesa' for now.   

However, I can not launch browser into full screen or resize it once launched.

I am also getting a lot of macro-blocking when playing youtube content. I suspect there are settings in 'Cog' that I can tweak to solve this? 

Any chance of help please?

cchilumbu, 2018-07-10:
Hello Carlos, 

With the suggested changes, the full screen seems to work for a short while before target crashes :(

Actually the target seems to always crash after attempting to playback youtube content.

And the macro-blocking is still there :( 

I am using 'core-image-weston', with rdk backend. 

I am happy to try out further suggestions.

Thanks.

clopez, 2018-07-10:
> Actually the target seems to always crash after attempting to playback youtube content.
> 
> And the macro-blocking is still there :(

Well, that's a different issue.

Just guessing, but that may be caused by missing the required gstreamer decoders for the media you are trying to play? If you don't have a hardware-based decoder supported by gstreamer, then you have to add at least the gstreamer software based decoders to the image. To do that, you have to add this in your conf/local.conf file
```
IMAGE_INSTALL_append = " gstreamer1.0-libav"
LICENSE_FLAGS_WHITELIST_append = " commercial"
```

cchilumbu, 2018-07-11:
Thanks Carlos,

The full screen is working now, and target is not crashing when I set it before launching cog.

However, I am still getting macro-blocking when I playback youtube content. Could it be that cog is not using gpu acceleration? How can I set this?

Also, while in youtube, if i enable 'Full Screen' mode, the target crashes.

Any ideas please?

Many thanks

cchilumbu, 2018-07-11:
Let me list pending issues:

1) Macro-blocking when using cog browser. I do not see this with Chromium or MPV.

2) Target crashes when youtube switches to full screen mode. I do not see this with Chromium or MPV.

3) Full screen mode is NOT working as expected as I reported earlier. Screen size remains the same, and background is blacked out. So Full Screen cog did NOT work.

4) While launching cog, I get a white screen first before page is loaded.

philn, 2018-08-14:
Regarding the macro-block issues,  can you try to play videos with `gst-play-1.0` and see if those glitches are also visible there? If it's the case, please report a bug to the meta-gstreamer1.0 maintainers.

philn, 2019-02-20:
The backlog is a bit hard to follow. Is there any issue left to address?

As a general remark, I would recommend to use the wpewebkit mailing list (https://lists.webkit.org/mailman/listinfo/webkit-wpe) for discussions instead of this issue tracker.

closed 2021-02-05 by psaavedra