Igalia/meta-webkit #23: iMX8 support

appcodr, 2018-07-09:
Hi,

We are able to use the WPE engine on iMX6 with the info in the wiki section.  Is WPE backend supported on iMX8 as well? We are planning to use on iMX8M and wanted to see if its supported or would be supported soon?

clopez, 2018-07-10:
That doesn't depend on meta-webkit. The hardware support depends on your hardware-BSP layers. More specifically, for i.MX8 you need to use the meta-freescale layers with version sumo or newer.

WPEWebkit itself should build and run fine on any Aarch64 (ARM64) hardware.
And regarding the WPE backends, the same backends that support the i.MX6 Vivante GPU should also do for to the i.MX8 Vivante one (AFAIK the graphic drivers are pretty much the same, aren't they?)

jnettlet, 2018-09-11:
Just a quick note here.  While the MX8M still supports galcore and Vivante's userspace binaries it is no longer using a fbdev interface.  Instead it is using drm/kms and gbm for display management and passing around buffers.  This may work with WPEBackend-mesa, but I haven't had the time to test yet.  I need to take some time and do some benchmarks of Vivante binaries vs drm-etnaviv.

miikeat, 2019-04-08:
Hi @appcodr 

I try to get meta-webkit running on a MX6 as well but had no luck till now. Which way did you configure your image? the wiki mentions multiple with and without wayland.

Could you take a look at my issues? Maybe you see whats wrong -> #61 for fdo and #90 for rdk

If you could share your configuration and layer versions for the imx6 board that would also be great!

nvl1109, 2019-11-28:
Hi @appcodr  and all,
I'm very curious on WPE + iMX8M. Do you have any status update on this? Thank you so much.

philn, 2019-11-28:
I have etnaviv, Weston (6), WPEBackend-FDO and Cog (in Weston with the FDO platform plugin and in FB with the DRM platform plugin) working on iMX8M.

Do you have a specific question about this?

nvl1109, 2019-11-29:
Hi @philn ,
It is great! I'm going to build WPE on TechNexion pico pi imx8mq. But there yocto repo latest is sumo and only have wiki for i.MX6.

So it is very helpful for me if you please share your configuration, information about your working WPE on iMX8M. Thank you :)

philn, 2019-11-29:
You can follow the same instructions as in the i.MX6 wiki page. the meta-webkit git master branch should build for Yocto Zeus (if not, please file issues :)).

closed 2021-03-11 by github-actions[bot]