Igalia/meta-webkit #24: MP4 not playing within Webkit again

gil0109, 2018-07-12:
The issue has shown up again (MP4 not playing within Webkit #18)  but in a different form so I am hoping it can be resolved.

Working site running cog: http://pauljadam.com/demos/autoplay-loop-muted-controls.html
Not working with my site (which is protected so I can't show)
Not working when pointing cog direct to mp4: https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc.mp4
Working using gst-play-1.0:  gst-play-1.0 https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc.mp4

** Note both sites prior did not work, then they both mysteriously worked, and now only the one site works.

I am using SUMO (gstreamer 1.12.4 & cog 0.1.0) with the following changes in the local.conf:

MACHINE = "raspberrypi3"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"
IMAGE_INSTALL_append = " wpewebkit cog"
DISTRO_FEATURES_append = " opengl wayland"
IMAGE_INSTALL_append = " gstreamer1.0-omx"
LICENSE_FLAGS_WHITELIST_append = " commercial"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""
ARTIFACTIMG_FSTYPE = "ext4"

gil0109, 2018-07-12:
Attached is a gst.log using the following command:
GST_DEBUG_FILE=gst.log GST_DEBUG_NO_COLOR=1 GST_DEBUG=webkit*:6 cog https://sbc-qsystem-dev.pathfinder.gov.bc.ca/sbc_new.mp4

Error on screen is: video controls show loading, then later it shows error.

[gst.log](https://github.com/Igalia/meta-webkit/files/2189635/gst.log)

philn, 2018-07-22:
Can you add 3 to GST_DEBUG? `GST_DEBUG=3,webkit*:6`

gil0109, 2018-07-22:
@philn I have decided to try building Chromium instead as I am getting more stable results for video and wifi network issues. Closing issue.  Thank you for all your help on this.

closed 2018-07-22 by gil0109