Igalia/meta-webkit #28: [2.21.91 wpewebkit] build failure with VIDEO disabled

macpijan, 2018-09-11:
I was trying the branch with upgraded packages from this PR: https://github.com/Igalia/meta-webkit/pull/27
My build is failing due to no `video` in `PACKAGECONFIG`

Error message:

```
Source/WebCore/page/Page.cpp:1521:27: error: 'class WebCore::ChromeClient' has no member named 'setUpPlaybackControlsManager'
```

Error reason:

This member is not compiled when the VIDEO is not enabled (347 line in `./Source/WebCore/page/ChromeClient.h`) :

```
#if ENABLE(VIDEO)
    virtual void enterVideoFullscreenForVideoElement(HTMLVideoElement&, HTMLMediaElementEnums::VideoFullscreenMode) { }
    virtual void setUpPlaybackControlsManager(HTMLMediaElement&) { }
    virtual void clearPlaybackControlsManager() { }
#endif
```

However, it is used unconditionally in the `Page.cpp` file.

By the way: is there any repo with the code of the `WPEwebkit`? Where do the releases of the https://wpewebkit.org/ come from?

philn, 2018-09-19:
Yeah @psaavedra recently contributed !VIDEO build fixes indeed. Feel free to close the issue if the new recipe builds for you!

macpijan, 2018-09-24:
The build with `VIDEO` disabled does not fail anymore

closed 2018-09-24 by macpijan