Igalia/meta-webkit #30: build failure with REMOTE_INSPECTOR is not enabled

mrstinky, 2018-09-13:
It looks like Remote Inspection code is still being called when REMOTE_INSPECTOR is not enabled.

I'm testing this on 2.20.1, but it looks like 2.21.92 would have the same problem.

Build error:
```wpewebkit/2.20.1-r0/wpewebkit-2.20.1/Source/WebKit/UIProcess/API/glib/WebKitAutomationSession.cpp:87:79: error: 'webkitWebContextWillCloseAutomationSession' was not declared in this scope```

The function is only in scope if REMOTE_INSPECTOR is enabled
```
Surce/WebKit/UIProcess/API/glib/WebKitWebContextPrivate.h-#if ENABLE(REMOTE_INSPECTOR)
Surce/WebKit/UIProcess/API/glib/WebKitWebContextPrivate.h:void webkitWebContextWillCloseAutomationSession(WebKitWebContext*);
Surce/WebKit/UIProcess/API/glib/WebKitWebContextPrivate.h-#endif
```

The origin of the call is traced back to didDisconnectFromRemote(*this); which hasn't been wrapped in #if ENABLED(REMOTE_INSPECTOR). But I'm not sure if it needs to be.

```
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-#if ENABLE(REMOTE_INSPECTOR)
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-    if (Inspector::FrontendChannel* channel = m_remoteChannel) {
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-        m_remoteChannel = nullptr;
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-        m_frontendRouter->disconnectFrontend(channel);
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-    }
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-    setIsPaired(false);
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-#endif
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-    if (m_client)
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp:        m_client->didDisconnectFromRemote(*this);
Source/WebKit/UIProcess/Automation/WebAutomationSession.cpp-}
```

shiftee, 2018-10-16:
Still happening for me on 2.22:

> wpewebkit-2.22.0/Source/WebKit/UIProcess/API/glib/WebKitAutomationSessionPrivate.h:27:115: error: 'RemoteInspector' in namespace 'Inspector' does not name a type

philn, 2018-10-16:
Forwarded upstream, https://bugs.webkit.org/show_bug.cgi?id=190623

philn, 2018-10-16:
On track for 2.22.3: https://trac.webkit.org/wiki/WebKitGTK/2.22.x

shiftee, 2018-12-18:
This issue appears to have been fixed. We now have 2.22.2

closed 2018-12-18 by philn