Igalia/meta-webkit #31: Allwinner (Mali-400) support

macpijan, 2018-09-13:
Hi,
I'm playing with the [orange-pi-pc-plus board](http://www.orangepi.org/orangepipcplus/) and I'd like to share
my results so far.
The GPU performance on the mainline kernel inside Wayland/Weston (of course libMali is still closed blob)
seems encouraging:

```
root@orange-pi-pc-plus:~# glmark2-es2-wayland  --fullscreen
=======================================================
    glmark2 2017.07
=======================================================
    OpenGL Information
    GL_VENDOR:     ARM
    GL_RENDERER:   Mali-400 MP
    GL_VERSION:    OpenGL ES 2.0
=======================================================
[build] use-vbo=false: FPS: 120 FrameTime: 8.333 ms
[build] use-vbo=true: FPS: 155 FrameTime: 6.452 ms
[texture] texture-filter=nearest: FPS: 130 FrameTime: 7.692 ms
[texture] texture-filter=linear: FPS: 130 FrameTime: 7.692 ms
[texture] texture-filter=mipmap: FPS: 128 FrameTime: 7.812 ms
[shading] shading=gouraud: FPS: 123 FrameTime: 8.130 ms
[shading] shading=blinn-phong-inf: FPS: 102 FrameTime: 9.804 ms
[shading] shading=phong: FPS: 74 FrameTime: 13.514 ms
[shading] shading=cel: FPS: 60 FrameTime: 16.667 ms
[bump] bump-render=high-poly: FPS: 76 FrameTime: 13.158 ms
[bump] bump-render=normals: FPS: 105 FrameTime: 9.524 ms
[bump] bump-render=height: FPS: 85 FrameTime: 11.765 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 25 FrameTime: 40.000 ms
libpng warning: iCCP: known incorrect sRGB profile
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 12 FrameTime: 83.333 ms
[pulsar] light=false:quads=5:texture=false: FPS: 150 FrameTime: 6.667 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 12 FrameTime: 83.333 ms
libpng warning: iCCP: known incorrect sRGB profile
[desktop] effect=shadow:windows=4: FPS: 53 FrameTime: 18.868 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 23 FrameTime: 43.478 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 45 FrameTime: 22.222 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 26 FrameTime: 38.462 ms
[ideas] speed=duration: FPS: 102 FrameTime: 9.804 ms
[jellyfish] <default>: FPS: 40 FrameTime: 25.000 ms
Error: SceneTerrain requires Vertex Texture Fetch support, but GL_MAX_VERTEX_TEXTURE_IMAGE_UNITS is 0
[terrain] <default>: Unsupported
[shadow] <default>: FPS: 40 FrameTime: 25.000 ms
[refract] <default>: FPS: 13 FrameTime: 76.923 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 120 FrameTime: 8.333 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 54 FrameTime: 18.519 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 120 FrameTime: 8.333 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 60 FrameTime: 16.667 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 37 FrameTime: 27.027 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 60 FrameTime: 16.667 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 60 FrameTime: 16.667 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 60 FrameTime: 16.667 ms
=======================================================
                                  glmark2 Score: 75 
=======================================================
```

Since the EGL extension `EGL_WL_bind_wayland_display` is available (as shown in the [weston.log](https://pastebin.com/uQXe26vJ)), trying out `wpebackend-fdo` seemed like a good idea.

With `fdo` it is failing at:

```
root@orange-pi-pc-plus:~# cog -P fdo google.pl
(cog:423): Cog-DEBUG: platform_setup: Platform name: fdo
(cog:423): Cog-DEBUG: platform_setup: Platform plugin: libcogplatform-fdo.so
Wayland: Got a wl_compositor interface
Wayland: Got a wl_seat interface
Wayland: Got an xdg_shell interface
Wayland: Got a wl_shell interface
Cog-INFO: EGL version 1.4 initialized.
xkbcommon: ERROR: couldn't find a Compose file for locale "C"
(cog:423): Cog-DEBUG: platform_setup: Platform = 0x3d930
Seat caps: Keyboard 
Seat name 'default'
New XDG toplevel configuration: (0, 0)
New XDG toplevel configuration: (1920, 1080)
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.

(cog:423): Cog-WARNING **: <http://google.pl/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.

(cog:423): Cog-WARNING **: <http://google.pl/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
```

The [strace.log](https://pastebin.com/aMwpA19H) seems to be revealing `SEGV`, which I guess could be traced further with the `gdb` (once I include it in the image):

```
[pid   808] ioctl(11, _IOC(_IOC_READ, 0x84, 0x2, 0x10), 0xaa6fea00) = 0
[pid   808] ioctl(11, _IOC(_IOC_READ|_IOC_WRITE, 0x83, 0, 0x28), 0xaa6fe880) = 0
[pid   808] mmap2(NULL, 262144, PROT_READ|PROT_WRITE, MAP_SHARED, 11, 0x10000000) = 0xb000c000
[pid   808] --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
[pid   807] <... poll resumed> <unfinished ...>) = ?
[pid   806] <... futex resumed>)        = ?
[pid   805] <... poll resumed> <unfinished ...>) = ?
[pid   804] <... poll resumed> <unfinished ...>) = ?
[pid   802] <... rt_sigtimedwait resumed> <unfinished ...>) = ?
[pid   801] <... ioctl resumed> <unfinished ...>) = ?
[pid   800] <... futex resumed>)        = ?
[pid   799] <... poll resumed> <unfinished ...>) = ?
[pid   798] <... poll resumed> <unfinished ...>) = ?
[pid   797] <... poll resumed> <unfinished ...>) = ?
[pid   796] <... futex resumed>)        = ?
[pid   794] <... futex resumed>)        = ?
[pid   798] +++ killed by SIGSEGV +++
[pid   807] +++ killed by SIGSEGV +++
[pid   806] +++ killed by SIGSEGV +++
[pid   805] +++ killed by SIGSEGV +++
[pid   804] +++ killed by SIGSEGV +++
[pid   802] +++ killed by SIGSEGV +++
[pid   801] +++ killed by SIGSEGV +++
[pid   800] +++ killed by SIGSEGV +++
[pid   799] +++ killed by SIGSEGV +++
[pid   797] +++ killed by SIGSEGV +++
[pid   796] +++ killed by SIGSEGV +++
```

The `rdk/wayland` backend seems to be going further:

```
root@orange-pi-pc-plus:~# cog https://google.com                 
(cog:979): Cog-DEBUG: platform_setup: Platform name: (null)
(cog:979): Cog-DEBUG: Instantiating default WPE backend as fall-back.
xkbcommon: ERROR: couldn't find a Compose file for locale "C"
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
Cannot create EGL WPE context: EGL_SUCCESS
Cog-Message: <https://google.com/> Load started.
Cog-Message: <https://www.google.com/> Redirected.
Cog-Message: <https://www.google.com/> Loading...
Cog-Message: <https://www.google.com/> Loaded successfully.
```

But still, nothing appears on the Weston screen.

Any suggestions with further ideas to try out are highly appreciated.

philn, 2018-09-13:
This error might be important?
`Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.`

macpijan, 2018-09-13:
@philn I'm wondering. Do you know of any list of extensions needed by the (WPE)Webkit?
From what I've read, those errors are more like a debug messages to inform that given extension is not supported, but in most cases, the application should fall back to another one. A similar message appears when running Weston:

```
[17:39:08.117] EGL_KHR_surfaceless_context unavailable. Trying PbufferSurface
```

where `PbufferSurface` is used as a fallback. 
In case of the (WPE)Webkit, in our case the Wayland context should be created I think: https://github.com/WebPlatformForEmbedded/WPEWebKit/blob/wpe-20170728/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp#L194
(through `wl_egl_window_create`).

philn, 2018-09-14:
Perhaps @elima or @zdobersek would like to chime in here. :)

zdobersek, 2018-09-21:
@macpijan can you attach output of running cog, using WPEBackend-fdo, with the WAYLAND_DEBUG=1 env exported?

macpijan, 2018-09-21:
@zdobersek I've yet to go anywhere further.
Here's the log: https://pastebin.com/AwZrYaXG

zdobersek, 2018-09-21:
@macpijan is it possible to provide a backtrace? or even sprinkle around logging messages to see what goes wrong where?

The wl_surface object seems to be created properly, and that's needed for the window-based EGL context. But the "Cannot create EGL surfaceless context" message is seen after wl_surface is created, which makes me think that for some reason the window-based EGL context creation is failing. It's also not clear what's then used for the sharing context.

macpijan, 2018-09-24:
@zdobersek Thanks for the support. Absolutely. I plan to do that this week.

macpijan, 2018-09-27:
@zdobersek 

The first bactrace I was able to retreive with the breakpoint at:

```
Thread 1 "cog" hit Breakpoint 2, cog_handle_web_view_web_process_terminated (web_view=0x81898 [WebKitWebView], reason=WEBKIT_WEB_PROCESS_CRASHED, userdata=0x0)
    at /usr/src/debug/cog/gitAUTOINC-r0/git/core/cog-webkit-utils.c:125
```

It could not fit the `WPEWebkti` debug symbols into the memory. I guess trying to debug it on target was not the best idea :)

```
(gdb) bt full
#0  0xb6fba87c in cog_handle_web_view_web_process_terminated (web_view=0x81898 [WebKitWebView], reason=WEBKIT_WEB_PROCESS_CRASHED, userdata=0x0) at /usr/src/debug/cog/gitAUTOINC-r0/git/core/cog-webkit-utils.c:125
        message = <optimized out>
        title = <optimized out>
#4  0xb548f2ec in <emit signal ??? on instance 0x3dc30 [GParamInt]> (instance=<optimized out>, signal_id=<optimized out>, detail=0) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/gobject/gsignal.c:3447
        var_args = {__ap = 0xbefffa24}
    #1  0xb54728a0 in g_closure_invoke (closure=0x871e0, return_value=0x0, n_param_values=2, param_values=0xbefff8d8, invocation_hint=0xbefff894)
    at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/gobject/gclosure.c:804
                marshal = 0xb5474fa8 <g_cclosure_marshal_VOID__ENUM>
                marshal_data = 0x0
                real_closure = 0x871d0
                __func__ = "g_closure_invoke"
    #2  0xb5485e08 in signal_emit_unlocked_R (node=node@entry=0x80ce0, detail=3204446336, detail@entry=0, instance=0x3dc30, instance@entry=0x81898, emission_return=0x80ca8, 
    emission_return@entry=0x0, instance_and_params=0xbefff8d8) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/gobject/gsignal.c:3635
                tmp = <optimized out>
                handler = 0x38af0
                accumulator = 0xb54a6a64
                emission = {next = 0x0, instance = 0x81898, ihint = {signal_id = 44, detail = 0, run_type = G_SIGNAL_RUN_FIRST}, state = EMISSION_RUN, chain_type = 4}
                handler_list = 0x38af0
                return_accu = 0x0
                accu = 
                      {g_type = 0, data = {{v_int = 0, v_uint = 0, v_long = 0, v_ulong = 0, v_int64 = 0, v_uint64 = 0, v_float = 0, v_double = 0, v_pointer = 0x0}, {v_int = 0, v_uint = 0, v_long = 0, v_ulong = 0, v_int64 = 0, v_uint64 = 0, v_float = 0, v_double = 0, v_pointer = 0x0}}}
                signal_id = 3041551480
                max_sequential_handler_number = 530584
                return_value_altered = <optimized out>
    #3  0xb548eabc in g_signal_emit_valist (instance=<optimized out>, signal_id=<optimized out>, detail=detail@entry=2981076992, var_args=..., var_args@entry=...)
    at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/gobject/gsignal.c:3391
                instance_and_params = 0xbefff8d8
                signal_return_type = <optimized out>
                param_values = 0xbefff8f0
                i = <optimized out>
                n_params = <optimized out>
                __func__ = "g_signal_emit_valist"
#5  0xb598030c in NavigationClient::processDidTerminate(WebKit::WebPageProxy&, WebKit::ProcessTerminationReason) () at /usr/lib/libWPEWebKit-0.1.so.2
#6  0xb592b668 in WebKit::WebPageProxy::dispatchProcessDidTerminate(WebKit::ProcessTerminationReason) () at /usr/lib/libWPEWebKit-0.1.so.2
#7  0xb592b6aa in WebKit::WebPageProxy::processDidTerminate(WebKit::ProcessTerminationReason) () at /usr/lib/libWPEWebKit-0.1.so.2
#8  0xb59439f2 in WebKit::WebProcessProxy::processDidTerminateOrFailedToLaunch() () at /usr/lib/libWPEWebKit-0.1.so.2
#9  0xb6cb2f32 in WTF::RunLoop::performWork() () at /usr/lib/libWPEWebKit-0.1.so.2
#10 0xb6ccf606 in WTF::RunLoop::RunLoop()::{lambda(void*)#1}::_FUN(void*) () at /usr/lib/libWPEWebKit-0.1.so.2
#11 0xb53a8cf4 in g_main_dispatch (context=0x3c300) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/glib/gmain.c:3165
        dispatch = 0x0
        prev_source = <optimized out>
        was_in_call = 0
        user_data = 0xb1afa000
        callback = 0xb6ccf601 <WTF::RunLoop::RunLoop()::{lambda(void*)#1}::_FUN(void*)>
        cb_funcs = 0xb546359c <g_source_callback_funcs>
        cb_data = 0x3c3e0
        need_destroy = <optimized out>
        source = 0x3c388
        current = 0x3ce58
        i = 0
#12 0xb53a8cf4 in g_main_context_dispatch (context=context@entry=0x3c300) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/glib/gmain.c:3818
#13 0xb53a906c in g_main_context_iterate (context=context@entry=0x3c300, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/glib/gmain.c:3891
        max_priority = 2147483647
        timeout = 19628
        some_ready = 1
        nfds = <optimized out>
        allocated_nfds = <optimized out>
        fds = 0x41e10
#14 0xb53a9134 in g_main_context_iteration (context=0x3c300, may_block=1) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/glib/gmain.c:3952
        retval = <optimized out>
#15 0xb5559934 in g_application_run (application=application@entry=0x37048 [CogLauncher], argc=argc@entry=4, argv=argv@entry=0xbefffd24) at /usr/src/debug/glib-2.0/1_2.54.3-r0/glib-2.54.3/gio/gapplication.c:2401
        arguments = 0x35fe8
        status = 0
        context = 0x3c300
        acquired_context = <optimized out>
        __func__ = "g_application_run"
#16 0x000119e8 in main (argc=4, argv=0xbefffd24) at /usr/src/debug/cog/gitAUTOINC-r0/git/cog.c:415
        app = 0x37048
```

Right now I'm rebuilding the packages with optimization disabled and building the `SDK` to debug remotly with `gdbserver`.

macpijan, 2018-09-28:
@zdobersek A little bit more detailed backtrace: https://pastebin.com/xyeT1zVB
I'm hooking on the cog crash display function - do you think it makes sense?

I am also trying to hook inside the WPEWebkit (specifically on the context creating function in the `Source/WebCore/platform/graphics/egl/GLContextEGL.cpp` where the error `Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension` supposedly comes from.
I am setting breakpoints there, but the execution continues like it was not executed (?)

zdobersek, 2018-10-01:
@macpijan be sure to set breakpoints on the WPEWebProcess executable, as there's where this code is executed.

That's also where this crash occurs, so it would be ideal if you could provide a backtrace for the crash that happens there (and which is the reason why cog is triggering the webprocess-crashed handler).

macpijan, 2018-10-08:
@zdobersek 
It took me a moment to handle this - debugging multiple threads was not that obvious at first.
I think I have identified the segmentation to occur when stepping into the `target->interface->create()`:

```
Thread 3.2 "ositorWorkQueue" hit Breakpoint 1, wpe_renderer_backend_egl_offscreen_target_create ()
    at /usr/src/debug/libwpe/1.0.0-r0/libwpe-1.0.0/src/renderer-backend-egl.c:137
137     {
138         struct wpe_renderer_backend_egl_offscreen_target* target = calloc(1, sizeof(struct wpe_renderer_backend_egl_offscreen_target));
139         if (!target)
142         target->interface = wpe_load_object("_wpe_renderer_backend_egl_offscreen_target_interface");
143         target->interface_data = target->interface->create();

Thread 3.2 "ositorWorkQueue" received signal SIGSEGV, Segmentation fault.
wpe_renderer_backend_egl_offscreen_target_create ()
    at /usr/src/debug/libwpe/1.0.0-r0/libwpe-1.0.0/src/renderer-backend-egl.c:143
143         target->interface_data = target->interface->create();
```

Backtrace for this thread: https://pastebin.com/eenppQNH

Backtrace for all the spawned threads: https://pastebin.com/ns0UayFH

Further inspection reveals that the `target->interface` has 0x0 address:

```
$1 = (struct wpe_renderer_backend_egl_offscreen_target *) 0x3c620
$2 = {interface = 0x0, interface_data = 0x0}
```

Is it OK to not check here if the interface was correctly obtained?

The reason for that is it tries to load the `_wpe_renderer_backend_egl_offscreen_target_interface`,
which is not present in the `wpebackend-fdo` and the `nullptr` gets returned:
https://github.com/Igalia/WPEBackend-fdo/blob/master/src/fdo.cpp#L46

Any input on this? I'm not sure what does it mean, but `offscreen interface`
does not sound to me like the correct one to be picked up there?
What determines which of the interface is going to be loaded?

macpijan, 2018-10-10:
@philn I think so. I also think that the return value from the `wpe_renderer_backend_egl_offscreen_target_create()` should be verified by the caller in the: https://github.com/WebPlatformForEmbedded/WPEWebKit/blob/wpe-20170728/Source/WebCore/platform/graphics/egl/GLContextEGLWPE.cpp#L69
Where patches for the WPEWebkit which is building from the recipes in this repo can be submitted?
Should it be the one from the above link?

I am also looking at the WPEWebkit code a little bit more, and it seems that the `wpe_renderer_backend_egl_offscreen_target_create` is called whenever the `GLContextEGL::createWPEContext` gets executed. But it seems that the `wpebackend-fdo` does not support it?
What should be the execution path in a positive scenario? Which platform from the: https://github.com/WebPlatformForEmbedded/WPEWebKit/blob/wpe-20170728/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp#L183 should be used for context creation.

macpijan, 2018-10-10:
@zdobersek Understood. So it seems that it is crucial for the driver to implement the mentioned extension in order to run WPEWebkit?
Do you think that falling back to pbuffed-based context may provide some reasonable performance? I understand that this path was not verified yet?

The result is the same as in the case of `wpewebkit-rdsk/wayland` - it says the site is loaded, but nothing appears on the screen.

```
Wayland: Got a wl_compositor interface
Wayland: Got a wl_seat interface
Wayland: Got an xdg_shell interface
Wayland: Got a wl_shell interface
xkbcommon: ERROR: couldn't find a Compose file for locale "C"
Seat caps: Keyboard 
Seat name 'default'
New XDG toplevel configuration: (0, 0)
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
Cannot create EGL WPE context: EGL_SUCCESS
Cog-Message: <http://google.pl/> Load started.
Cog-Message: <http://www.google.pl/> Redirected.
Cog-Message: <https://www.google.pl/?gws_rd=ssl> Redirected.
Cog-Message: <https://www.google.pl/?gws_rd=ssl> Loading...
Cog-Message: <https://www.google.pl/?gws_rd=ssl> Loaded successfully.
```

And the same obscure message: `Cannot create EGL WPE context: EGL_SUCCESS` - success or fail?

zdobersek, 2018-10-11:
@macpijan falling back to pbuffer is utilized with other backends. It works, and there shouldn't be any difference in performance between pbuffer-based and surfaceless contexts. These are used for the sharing context (a token context enabling sharing GPU processes across other contexts) and for WebGL content.

I'd ignore the `EGL_SUCCESS` message and focus on this part:
```
New XDG toplevel configuration: (0, 0)
```

For some reason Weston is setting a 0x0 size for Cog's window. I don't know why, but it shouldn't be 0x0.

macpijan, 2018-10-11:
@zdobersek Yes

zdobersek, 2018-10-11:
@macpijan I suspect an issue in how Weston handles XDG surfaces. It works as expected on the RPi with Mesa drivers, using Weston 2.0.0.

Is this some standard release or a specific version that's necessary to work with the Mali drivers?

Can you try disabling the XDG shell interface binds in order to fall back to the wl_shell functionality?
https://github.com/Igalia/cog/blob/master/platform/cog-platform-fdo.c#L353
(Which shell protocol you end up using should be visible from WAYLAND_DEBUG=1 output.)

zdobersek, 2018-10-12:
@macpijan alright, I was using rocko releases so far, let me try with sumo.

zdobersek, 2018-10-22:
@macpijan I tested the sumo build on an RPi and an iMX6 board, but everything worked fine there, using Weston 4.0.92.

My suspicion remains that this is an issue around how the Wayland protocol (and related to this, Weston behavior) is integrated into the Mali GPU drivers. I'll be receiving an SBC with a Mali GPU sometime this week, and I'll investigate further then.

philn, 2019-05-27:
Anything left to fix here @macpijan ?

macpijan, 2019-05-28:
@philn Possibly. I am not aware of the current state of the Mali-400 with WPEWebkit. I left it from there and IIURC @zdobersek did as well.

I may find some time to test the current state of those platforms if you think anything may have changed for them since then.

closed 2021-03-11 by github-actions[bot]

trinukso, 2021-03-27:
What about PinePhone?

philn, 2021-03-28:
We had some success with the lima driver on orange-pi-pc-plus a couple months ago. This issue is closed though. Feel free to open a new one or a "discussion" thread regarding PinePhone.