Igalia/meta-webkit #32: cog won't build because it needs a version of wpewebkit not provide by this layer

mrstinky, 2018-09-13:
The error
```
ERROR: cog-20180809-r0 do_configure
...
| -- Checking for modules 'wpe-webkit-0.1>=2.21.90;wpe-0.2'
| --   Requested 'wpe-webkit-0.1 >= 2.21.90' but version of WPE WebKit is 2.20.1
| You may find new versions of WPE WebKit at https://wpewebkit.org
| --   No package 'wpe-0.2' found
| CMake Error at /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/cog/20180809-r0/recipe-sysroot-native/usr/share/cmake-3.8/Modules/FindPkgConfig.cmake:416 (message):
|   A required package was not found
```

The change https://github.com/Igalia/cog/commit/ed26f03900f9844b2a260a2ad92ad2d9038fdb28  that caused it

I guess the recipe could specify the particular version of cog that will build against this recipe, or I can trying using the latest wpewebkit version.

philn, 2018-09-13:
Can you give a try to this PR? https://github.com/Igalia/meta-webkit/pull/27

mrstinky, 2018-09-13:
Thanks. I ended up changing the recipe to specify the cog-0.1 branch
`SRC_URI = " git://github.com/Igalia/cog.git;protocol=https;branch=cog-0.1`

I'll swap my manifest over to your fork until the pull request is completed.

philn, 2018-09-20:
I think this can be closed now...

closed 2018-09-20 by philn