Igalia/meta-webkit #35: [2.21.91 wpewebkit] JavaScriptCore fails to build

mrstinky, 2018-09-18:
Building wpewebkit with `PACKAGECONFIG = "remote-inspector"`

Looks like the JavaScriptCore fails to compile

```
| /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.91-r0/build/DerivedSources/ForwardingHeaders/JavaScriptCore/ARM64Assembler.h:459:48: warning: 'JSC::ARM64Assembler::LinkRecord::<anonymous union>::RealTypes::m_compareRegister' is too small to hold all values of 'JSC::ARM64Assembler::RegisterID {aka enum JSC::ARM64Registers::RegisterID}'
|                  RegisterID m_compareRegister : 6;
|                                                 ^
```

philn, 2018-09-18:
I don't have an ARM64 board here, so won't be able to help much... This was likely fixed already in WebKit trunk, can you check maybe?

mrstinky, 2018-09-18:
I'm trying the Trunk but it looks like the network/SVN server is throttling the connection.  Taking a few hours to pull down that code.

mrstinky, 2018-09-18:
Looks like Trunk fails the same way. I'll have to go back to a stable version I think.

```
| /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/1.trunk-235780-r0/build/DerivedSources/ForwardingHeaders/JavaScriptCore/ARM64Assembler.h:459:48: warning: 'JSC::ARM64Assembler::LinkRecord::<anonymous union>::RealTypes::m_compareRegister' is too small to hold all values of 'JSC::ARM64Assembler::RegisterID {aka enum JSC::ARM64Registers::RegisterID}'
|                  RegisterID m_compareRegister : 6;
|                                                 ^
```

mrstinky, 2018-09-18:
Downstream also fails:

```
| CMake Error at /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/1.downstream-20180424-r0/recipe-sysroot-native/usr/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find WPEBackend (missing: WPE_INCLUDE_DIRS WPE_LIBRARIES)
| Call Stack (most recent call first):
|   /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/1.downstream-20180424-r0/recipe-sysroot-native/usr/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
|   Source/cmake/FindWPEBackend.cmake:47 (find_package_handle_standard_args)
|   Source/cmake/OptionsWPE.cmake:83 (find_package)
|   Source/cmake/WebKitCommon.cmake:61 (include)
|   CMakeLists.txt:154 (include)
```

dinfuehr, 2018-09-18:
Do you build with -Werror? I've built JSC for AArch64 just now, those errors are "just" warnings for me.

mrstinky, 2018-09-18:
Good point. Looks like I missed the actual error because it was further up the page, and off the screen:

```
/local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.92-r0/wpewebkit-2.21.92/Source/WebCore/page/Page.cpp:1520:33: error: incomplete type 'WebCo      re::HTMLMediaElement' used in nested name specifier
      if (auto bestMediaElement = HTMLMediaElement::bestMediaElementForShowingPlaybackControlsManager(MediaElementSession::PlaybackControlsPurpose::ControlsManager))
                                  ^~~~~~~~~~~~~~~~                               
 /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.92-r0/wpewebkit-2.21.92/Source/WebCore/page/Page.cpp:1520:101: error: 'MediaElementSession'       has not been declared
      if (auto bestMediaElement = HTMLMediaElement::bestMediaElementForShowingPlaybackControlsManager(MediaElementSession::PlaybackControlsPurpose::ControlsManager))
                                                                                                      ^~~~~~~~~~~~~~~~~~~
 /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.92-r0/wpewebkit-2.21.92/Source/WebCore/page/Page.cpp:1521:27: error: 'class WebCore::Chrome      Client' has no member named 'setUpPlaybackControlsManager'                                                                                                                            
          chrome().client().setUpPlaybackControlsManager(*bestMediaElement);     
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~                         
 /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.92-r0/wpewebkit-2.21.92/Source/WebCore/page/Page.cpp:1523:27: error: 'class WebCore::Chrome      Client' has no member named 'clearPlaybackControlsManager'
          chrome().client().clearPlaybackControlsManager();   
```

philn, 2018-09-18:
This might be fixed by https://github.com/Igalia/meta-webkit/pull/34

philn, 2018-09-20:
Is this fixed then?

mrstinky, 2018-09-21:
Sorry, not had a chance to go back and try yet.

philn, 2019-02-20:
Ping @mrstinky

mrstinky, 2019-02-20:
TL;DR: Looks like it built OK.

Sorry for the delay, we ended up going with 2.20.2 as it was the latest stable at the time, so I never went back to this, and I've since moved on from this project now. 

Is this fix a backported patch for that particular version? So I assume I have to build 2.21.91 to check this fix. Can you just confirm that I've done this right: 

I resynced, looked for the change in the log, then checked out the hash; 

```
$ git checkout 5c4a988561ca0fa3fb7fe0e8eab74797766fe856
Previous HEAD position was c1bcb24... Add new PACKAGECONFIG USE_OPENJPEG (#69)
HEAD is now at 5c4a988... Add upstreamed patches for wpewebkit_2.21.91 needed to build package
```
Confirmed that the patches appeared in the directory:

```
/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit
0001-REGRESSION-r217927-GTK-WPE-GSTREAMER_GL-Video-appear.patch
0001-REGRESSION-r217927-WPE-2.22-GSTREAMER_GL-Video-appea.patch
fix-189500.patch
fix-189506.patch
fix-189540.patch

```
Forced my recipe to use 2.21.91, started a rebake, and confirmed that it was building 2.21.91 
```
cat recipes-browser/wpewebkit/wpewebkit_2.21.91.bbappend
# set the configuration options, and build this version in preference
PACKAGECONFIG_pn-wpewebkit = "remote-inspector"
DEFAULT_PREFERENCE = "1"
```
Confirmed patches applied:
```
tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.21.91-r0/temp/log.do_patch

NOTE: Applying patch '0001-REGRESSION-r217927-WPE-2.22-GSTREAMER_GL-Video-appea.patch' (../meta-webkit/recipes-browser/wpewebkit/wpewebkit/0001-REGRESSION-r217927-WPE-2.22-GSTREAMER_GL-Video-appea.patch)
NOTE: Applying patch 'fix-189500.patch' (../meta-webkit/recipes-browser/wpewebkit/wpewebkit/fix-189500.patch)
NOTE: Applying patch 'fix-189506.patch' (../meta-webkit/recipes-browser/wpewebkit/wpewebkit/fix-189506.patch)
NOTE: Applying patch 'fix-189540.patch' (../meta-webkit/recipes-browser/wpewebkit/wpewebkit/fix-189540.patch)
```

rebake was successful

philn, 2019-02-20:
Yep, the `fix` patches were upstreamed. Let's close this then :) Thanks for the quick reply!

closed 2019-02-20 by philn