Igalia/meta-webkit #36: [1.0.0 wpebackend-fdo] Can't find EGL_WAYLAND_PLANE_WL

mrstinky, 2018-09-18:
I'm building wpebackend-fdo 1.0.0 with our custom EGL drivers. This is the build error:

```
| /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpebackend-fdo/1.0.0-r0/wpebackend-fdo-1.0.0/src/view-backend-exportable-fdo-egl.cpp: In member function 'virtual void {anonymous}::ClientBundleEGL::exportBuffer(wl_resource*)':
| /local/mnt/workspace/LE.UM.2.4.1/poky/build/tmp-glibc/work/aarch64-oe-linux/wpebackend-fdo/1.0.0-r0/wpebackend-fdo-1.0.0/src/view-backend-exportable-fdo-egl.cpp:101:13: error: 'EGL_WAYLAND_PLANE_WL' was not declared in this scope
|              EGL_WAYLAND_PLANE_WL, 0,
|              ^~~~~~~~~~~~~~~~~~~~
```

The wpebackend-fdo build can't find `EGL_WAYLAND_PLANE_WL`.  

view-backend-exportable-fdo-egl.cpp includes

```
#include <EGL/egl.h>                                                            
#include <EGL/eglext.h>
```

In which header file would it expect to find this symbol?

We do provide this symbol in a file called "eglwaylandext.h", so it's possible that our EGL authors haven't put the symbols in the expected files. But it looks to me like wpebackend-fdo needs to include a wayland specific EGL file.

philn, 2018-09-18:
wpebackend-fdo depends on wpebackend (aka libwpe) which depends on virtual/egl... Do you need to have a recipe providing virtual/egl... like Mesa, for instance.

mrstinky, 2018-09-18:
Yes, we have our own virtual/egl. Our virtual/egl  provides EGL/egl.h and EGL/eglext.h.  

I'm not sure if EGL/egl.h and EGL/eglext.h would normally define EGL_WAYLAND_PLANE_WL, or include the file which defines it. 

I can hack it in for now, but I'm not sure what to feedback to our virtual/egl provider so they can fix the problem at source.

mrstinky, 2018-09-18:
OK.  Looking at the Mesa library. It looks like:

eglext.h -- includes --> eglmesaext.h --defines--> EGL_WAYLAND_PLANE_WL,"

https://cgit.freedesktop.org/mesa/mesa/tree/include/EGL/eglext.h
https://cgit.freedesktop.org/mesa/mesa/tree/include/EGL/eglmesaext.h

mrstinky, 2018-09-21:
Problem was in our EGL libraries, will be fixed.

closed 2018-09-21 by mrstinky