Igalia/meta-webkit #38: [Request] Tag stable versions of meta-webkit

mrstinky, 2018-09-20:
Are there plans to release/tag stable versions of meta-webkit? 

e.g. a version that will successfully complete a build of wpewebkit + backend_x + cog?

wpewebkit now has stable releases, but building it with the backend and cog doesn't work "out of the box".

philn, 2018-09-20:
What are the remaining issues?

psaavedra, 2018-09-20:
Maybe, you mean aligned with OE/Yocto releases (rocko, sumo , sumo-next,  ...)?

mrstinky, 2018-09-21:
@philn  Maybe there aren't any remaining issues.  I haven't had the time to go back and try the latest builds.

I've managed to build wpewebkit 2.20.2 + wpebackend 0.2 + wpebackend-fdo 0.1 + cog 0.1. And I'm currently trying to get that running on target.

For the moment, my priority is get _a_ version of wpewebkit running on our hardware in order to prove that our EGL and Weston libraries meet the requirements.  Once that's done, then I'll look at trying a more bleeding edge version.

@psaavedra Releases aligned with OE/Yocto are always useful as well. For example, I'm on Rocko 2.4.3. I'm just assuming that the latest meta-webkit is suitable for Rocko.

philn, 2018-10-11:
There's a rocko branch now and soon a sumo branch too, once https://github.com/Igalia/meta-webkit/pull/47 is merged.

closed 2018-10-17 by philn