Igalia/meta-webkit #45: nothing provides gstreamer1.0-plugins-bad-opengl

cadubentzen, 2018-10-10:
As in https://github.com/OSSystems/meta-gstreamer1.0/issues/20, building a WPE 2.20.1 image with Yocto master yields this error.

```
Error: 
 Problem 1: conflicting requests
  - nothing provides gstreamer1.0-plugins-bad-opengl needed by wpewebkit-2.22.0-r0.cortexa7t2hf_neon_vfpv4
 Problem 2: package cog-git0-r0.cortexa7t2hf_neon_vfpv4 requires libWPEWebKit-0.1.so.2, but none of the providers can be installed
  - conflicting requests                                                
  - nothing provides gstreamer1.0-plugins-bad-opengl needed by wpewebkit-2.22.0-r0.cortexa7t2hf_neon_vfpv4
```

The answer in there already suggests a fix: 

> The OpenGL plugin was moved to -base in 1.14. So you need gstreamer1.0-plugins-base-opengl .

Should the logic for where to get the opengl plugins depending on the gstreamer version be added do the wpewebkit recipe?

Thanks!

philn, 2018-10-10:
Yes. I don't know if that's doable though? If gst 1.12 is selected we want opengl from -bad, otherwise if 1.14 is selected we want opengl from -base... If you find how to do that please file a PR :)

cadubentzen, 2018-10-11:
I asked about it on #yocto and they recommended:

> #khem cramalho: you should just track release branches and match them with oe-core release branches

With this workflow essentially this meta layer (ideally) would be branched out to the versions (rocko, sumo, etc) and master would have the `-bad` -> `-base` change to match `oe-core`'s gstreamer naming until `thud` is made stable and it is branched again.

Ok. A lot of rethinking on the repository workflow over one problem but I wonder if this is the usual yocto workflow? I'm new to Yocto btw.

It seems branches were created for krogoth, jethro and fido, but stopped doing this afterwards.

philn, 2018-10-11:
Yeah, right, release branches are a good approach indeed. Many layers do that. Until which release should we use 1.12? Rocko?

philn, 2018-10-11:
https://github.com/Igalia/meta-webkit/pull/47
rocko branch is already pushed.

cadubentzen, 2018-10-11:
> Yeah, right, release branches are a good approach indeed. Many layers do that. Until which release should we use 1.12? Rocko?

1.14 is quite recent, right? Up until sumo 1.12 is used. Now in master (which will become thud) 1.14 is used.

https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer?h=sumo

So I think until sumo 1.12 should be used. Then 1.14 in current master and forward.

philn, 2018-10-11:
Hm true, but OSSystems have a rocko branch shipping 1.14, maybe we could state in the README this is a highly recommended layer.

philn, 2018-10-11:
https://github.com/OSSystems/meta-gstreamer1.0/tree/rocko/recipes-multimedia/gstreamer

cadubentzen, 2018-10-11:
Yes, I think this layer would be really useful for compatibility with versions older than master. Especially for compiling with MSE, which now requires gst 1.14

philn, 2018-10-11:
As the README points to the Wiki, I edited the Wiki :) https://github.com/Igalia/meta-webkit/wiki

philn, 2018-10-12:
So if meta-gstreamer1.0 becomes a recommended dependency we can just assume gst 1.14 is used for rocko sumo & thud. That simplifies things.

closed 2018-10-15 by philn