Igalia/meta-webkit #50: Why is xkbcommon used with wpebackend-rdk?

miikeat, 2018-10-19:
Wpebackend-rdk depends on libinput and wpebackend-fdo on libxkbcommon.

I have an RPi and built my Image using RDK so when I start cog I see a LibInputServer starting.
The thing that confuses me: If i press a any button on my Keyboard, cog crashes with a segfault and an errormessag of xkbcommon complaining that the directory ```/usr/share/X11/xkb``` could not be found.
When I just copy that folder from my desktop on the sd-card everything works fine.

So two questions:

- Is it correct that libxkbcommon should be used when I built my image using wpebackend-rdk?
- What should normally create that directory and its content? Am I missing something when building the image?

I haven't done anything more really than some recipes to configure the screen and added a custom splash screen... So I pretty much have just built the image like explained in the Wiki for RPi.

psaavedra, 2018-10-24:
Maybe this is something related to https://github.com/WebPlatformForEmbedded/WPEBackend-rdk/pull/15/files

Could be useful to see the bitbake build logs to see how was configured wpebackend-rdk

philn, 2019-02-20:
@miikeat can you provide the bitbake build logs?

miikeat, 2019-04-05:
As an update to that issue:

I added the following packages to get that working:

```
IMAGE_INSTALL_append = "xkeyboard-config"
```

I've also removed x11 from the distro, so i wonder if that was maybe the reason... But I would wonder if I haven't tried with x11 as well to fix that issue.

philn, 2019-05-27:
So should `xkeyboard-config` be a dependency of the RDK backend? For which packageconfig option?

miikeat, 2019-06-03:
Yes, for RDK.

closed 2020-04-15 by psaavedra