Igalia/meta-webkit #51: krogoth build for Up Board fails to find LibEpoxy

jaripenttinen, 2018-10-19:
I am trying top build Yocto for UpBoard having also wpewebkit

MACHINE = "up-board"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
IMAGE_INSTALL_append = " wpewebkit dyz"

If I add package dyz, it fails to

Could NOT find LibEpoxy: Found unsuitable version "1.3.1", but required is at least "1.4.0" 

Any ideas how to fix?

philn, 2018-10-19:
Dyz is no longer maintained, it was replaced by Cog.

psaavedra, 2018-10-24:
> I am trying top build Yocto for UpBoard having also wpewebkit
>
> MACHINE = "up-board"
> PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
> IMAGE_INSTALL_append = " wpewebkit dyz"
>
> If I add package [...]

I suppose you are using `master` branch since `virtual/wpebackend` is something relatively new in this layer.

I'd suggest to you the usage of the `krogoth` branch (https://github.com/Igalia/meta-webkit/tree/krogoth/recipes-browser).

As @philn says we are not longer maintaining `dyz` but another idea that you can do it by yourself could be to try to add a `.bbappend` recipe in your custom layers patching the cmake rules for `dyz` to set as valid your older libepoxy library version ans see what happens.

jaripenttinen, 2018-10-24:
> I suppose you are using `master` branch since `virtual/wpebackend` is something relatively new in this layer.
>
> I'd suggest to you the usage of the `krogoth` branch [...]

Well actually I am using krogoth branch. I tried first pyro with master, and that virtual/wpebackend is just leftover from there.

I could try to do that thing you proposed, problem is that I am not familiar of Yocto system, so I am looking for ready solution I just could build :) 
But thank you for your proposal, I will check it in a near future. 

Any ideas if there is a support for Thud or Sumo coming in near future?

psaavedra, 2018-10-24:
It is not a golden rule but the here master is always against the current stable OE meta-layers (sumo at this moment). Once a new stable is announced we fork a branch with the previous branch (to keep the history easily accessible).

philn, 2019-05-27:
@jaripenttinen Have you upgraded to Thud or Sumo in the end? Please let us know if this issue should remain open.

closed 2021-03-11 by github-actions[bot]