Igalia/meta-webkit #59: Transparent background

abirza, 2018-11-28:
Is transparent background supported with the Cog Launcher?
I'm trying to run application on the imx6 where gstreamer is sendind video to the /dev/fb0 and I need to have a transparent UI that would be drawn on the /dev/fb1.

philn, 2018-11-28:
Is the GStreamer video rendered with a `<video>` element?
Is your use-case about overlaying html over foreign (non-WebKit app) video?

AndrisBB, 2018-11-28:
Second scenario.
GStreamer is ouputing video directly to the /dev/fb0. 
Webkit app is rendering UI to the /dev/fb1 (overlay).
IMX6 is merging both framebuffers together. Probbably it is better explained here starting from slide 35.
http://cache.freescale.com/files/training/doc/ftf/2014/FTF-CON-F0223.pdf
For examaple if I output png image with transparent parts to the FB1, then I will see video under image.

I'm new to the WPE and still in the process of understanding how does it work. What I found is that WPEBrowser have a setting to enable transparent background and was hoping that it is also possible with the Cog Launcher.
https://github.com/WebPlatformForEmbedded/meta-wpe/wiki/WPEFramework

philn, 2018-11-28:
Can you please reach out by e-mail? philn@igalia.com

philn, 2018-12-03:
This new API for WPE should solve the problem: https://bugs.webkit.org/show_bug.cgi?id=192305