Igalia/meta-webkit #61: i.MX6 - Cog fails to start - cannot open shared object file

miikeat, 2019-01-14:
I followed the wiki pages for i.MX6 to build my image so i have a pretty basic core-image-weston image based on the sumo release (except for meta-webkit which is master) but when I try to start cog it fails with the following error message:

    root@imx6qdlsabresd:~# cog www.google.com
    wpe: could not  load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory
    Aborted (core dumped)

When I try to start it with the parameter ```--platform=fdo```:

    root@imx6qdlsabresd:~# cog --platform=fdo www.google.com
    
    (cog:231): Cog-WARNING **: Platform setup failed: Could not open Wayland display

    ** (cog:231): CRITICAL **: WebKitWebViewBackend* webkit_web_view_backend_new(wpe_view_backend*, GDestroyNotify, gpointer): assertion 'backend' failed

    (cog:231): Cog-ERROR **: Could not instantiate any WPE backend.
    Trace/breakpoint trap (core dumped)

So it seems somethings isn't right with the Wayland setup... Sadly

Here's my ```local.conf```:

    MACHINE ??= 'imx6qdlsabresd'
    DISTRO ?= 'fslc-wayland'
    PACKAGE_CLASSES ?= "package_deb"
    EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
    USER_CLASSES ?= "buildstats image-mklibs image-prelink"
    PATCHRESOLVE = "noop"
    PACKAGECONFIG_append_pn-qemu-native = " sdl"
    PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
    CONF_VERSION = "1"
    
    DL_DIR ?= "${BSPDIR}/downloads/"
    ACCEPT_FSL_EULA = "1"
    
    B_NUMBER_THREADS ?= "16"
    PARALLEL_MAKE ?= "-j 12"
    
    # Accelerated video support
    IMAGE_INSTALL_append = " gstreamer1.0-plugins-imx  gstreamer1.0-plugins-imx-meta"
    
    # WPEWebkit and cog launcher
    IMAGE_INSTALL_append = " wpewebkit cog"
    
    # Backend selection
    PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"

And my ```bblayers.conf```:

    BBLAYERS = " \
      ${BSPDIR}/sources/poky/meta \
      ${BSPDIR}/sources/poky/meta-poky \
      \
      ${BSPDIR}/sources/meta-openembedded/meta-oe \
      ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
      \
      ${BSPDIR}/sources/meta-freescale \
      ${BSPDIR}/sources/meta-freescale-3rdparty \
      ${BSPDIR}/sources/meta-freescale-distro \
      \
      ${BSPDIR}/sources/meta-webkit \
      ${BSPDIR}/sources/meta-gstreamer1.0 \

philn, 2019-01-14:
Is weston running?

philn, 2019-01-14:
Ok, cog with the fdo plugin can't run without compositor. So you'll need to have a working Wayland compositor before trying cog again.

miikeat, 2019-01-14:
The journal for weston.service are not really helpful ...

    -- Logs begin at Mon 2019-01-14 21:29:16 UTC, end at Mon 2019-01-14 21:31:18 UTC. --
    Jan 14 21:29:19 imx6qdlsabresd systemd[1]: Starting Weston Wayland Compositor (on tty7)...
    Jan 14 21:29:19 imx6qdlsabresd systemd[1]: Started Weston Wayland Compositor (on tty7).
    Jan 14 21:29:28 imx6qdlsabresd systemd[1]: weston.service: Main process exited, code=exited, status=1/FAILURE
    Jan 14 21:29:28 imx6qdlsabresd systemd[1]: weston.service: Failed with result 'exit-code'.

After a little search I figured weston should be started with the correct tty so when I run ```weston --tty=7``` I just get an empty screen with a frozen cursor. Starting it with ```westion --tty=7 --backend=wayland-backend.so``` gives me the following error:

    Date: 2019-01-14 UTC
    [21:36:20.687] weston 4.0.0
                   http://wayland.freedesktop.org
                   Bug reports to: https://bugs.freedesktop.org/enter_bug.cgi?product=Wayland&component=weston&version=4.0.0
                   Build: 4.0.0-27-gbf74309d-dirty g2d-render: support src crop in viewport protocol (2018-06-08 16:23:42 +0800)
    [21:36:20.688] Command line: weston --tty=7 --backend=wayland-backend.so
    [21:36:20.688] OS: Linux, 4.1.46-Com, #16 SMP PREEMPT Wed Nov 14 20:05:33 CET 2018, armv7l
    [21:36:20.688] Starting with no config file.
    [21:36:20.689] Output repaint window is 16 ms maximum.
    [21:36:20.689] Loading module '/usr/lib/libweston-4/wayland-backend.so'
    [21:36:20.726] Error: Failed to connect to parent Wayland compositor: No such file or directory
                   display option: (none), WAYLAND_DISPLAY=(not set)
    [21:36:20.726] fatal: failed to create compositor backend

Could you give me a hint what could cause the image behave like that? Is there anything I missed in the documentation from the Wiki for I.MX6?

philn, 2019-01-14:
Sorry I'm not familiar with the proprietary Vivante driver in combination with Weston. I advise you to reach out to the broader Wayland/Weston community for proper support.

psaavedra, 2019-02-06:
Hi @miikeat ,AFAIK Vivante  only works for Weston2 at this moment.  the following settings it is the unique combination that I was able to make it work.

```
DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " egl"

PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "libwpe"

PREFERRED_VERSION_cog = "git%"
PREFERRED_VERSION_wpewebkit = "1.trunk%"
PREFERRED_VERSION_wpebackend-fdo = "1.0.0~git%"
PREFERRED_VERSION_libwpe = "1.0~git%"

PREFERRED_VERSION_wayland = "1.13.%"
PREFERRED_VERSION_wayland-protocols = "1.10"
PREFERRED_VERSION_weston = "2.%"
```

... for trunk, or:

```
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "wpebackend"
PREFERRED_VERSION_cog = "0.1%"
PREFERRED_VERSION_wpewebkit = "2.20%"
PREFERRED_VERSION_wayland = "1.13.%"
PREFERRED_VERSION_wayland-protocols = "1.10"
PREFERRED_VERSION_wpebackend-fdo = "0.1~git0.2"
PREFERRED_VERSION_weston = "2.%"
```

... for 2.20.

Unfortunately, Weston 2 is from Rocko but you can still get Weston 2 from https://gitlab.com/saavedra.pablo/meta-perf-browser/tree/sumo/recipes-graphics/wayland for Sumo .
In this meta-layer, I have defined some Yocto conf templates which I use for test vivante and etnaviv on Wandboards. You could try this meta-layer as follows:

```
mkdir -p ~/yocto-wandboard-wpe
cd ~/yocto-wandboard-wpe
repo init -u https://gitlab.com/saavedra.pablo/meta-perf-browser.git -m manifest.xml -b sumo
repo sync --force-sync
source setup-environment wandboard-vivante-wpe-2.22 --update-config
rm -rf tmp
bitbake core-image-weston-wpe
```

There are more templates available here: https://gitlab.com/saavedra.pablo/meta-perf-browser/tree/master/conf/

miikeat, 2019-02-08:
@psaavedra Wow! Thanks! As soon as I have the board back I'll give it another try.

philn, 2019-02-20:
@psaavedra maybe your explanation can move somewhere to the wiki? It would be nice I think :)
@miikeat please let us know how it goes!

macpijan, 2019-03-04:
@psaavedra What exactly are you testing? Are you running some kind of performance benchmarks? I don't have a wanbaord myself, but I think it would be extremely interesting to see some performance comparison (e.g. vivante vs etnaviv).

miikeat, 2019-04-08:
@psaavedra I tried now your recommended configuration but had a lot of issues:

I did not find the recipe for Weston 2.x in the repo you linked, there is only version 3. So i took it from openembedded-core - rocko. Also Weston 2 requires wayland-protocols 10.0 which i got from there as well.

Because ```imx-base.inc``` sets ```PREFERRED_VERSION_weston_mx6 ?= "4.0.0.imx"``` I had as well to set ```PREFERRED_VERSION_weston_mx6 = "2.0.0"```.

This caused weston 2.0.0 to fail with the following error:
```
| checking for libudev >= 136 libdrm >= 2.4.30 gbm mtdev >= 1.1.0... no
| configure: error: Package requirements (libudev >= 136 libdrm >= 2.4.30 gbm mtdev >= 1.1.0) were not met:
| 
| No package 'gbm' found
```
I only found meta-ti which had a libgbm package but adding that layer didn't work.

I found this bugrport https://bugreports.qt.io/browse/QTBUG-56504 so which claims kms should not be activated, so i removed kms from the packageconfig for wayland in the weston 2.0.0 recipe.

```
PACKAGECONFIG ??= "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'fbdev wayland egl', '', d)} \
```
instead of
```
PACKAGECONFIG ??= "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'kms fbdev wayland egl', '', d)} \
```

That caused gtk+3 to fail:
```
ERROR: gtk+3-3.22.28-r0 do_compile: oe_runmake failed
ERROR: gtk+3-3.22.28-r0 do_compile: Function failed: do_compile (log file is located at .../build/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/gtk+3/3.22.28-r0/temp/log.do_compile.21892)
...
Making all in gdk
| make[2]: Entering directory '.../build/tmp/work/armv7at2hf-neon-mx6qdl-fslc-linux-gnueabi/gtk+3/3.22.28-r0/build/gdk'
| Failed to open file “./gdk.gresource.xml”: No such file or directory
```
gtk+3 is only added because of gtk+3-demo so i removed that from the image
```
IMAGE_INSTALL_remove = "gtk+3-demo"
```
but then i gave up after getting an error with gstreamer:
```
ERROR: gstreamer1.0-plugins-bad-1.14.4-r0 do_compile: oe_runmake failed
ERROR: gstreamer1.0-plugins-bad-1.14.4-r0 do_compile: Function failed: do_compile 
```

My final configuration looked like that:
```
MACHINE ??= 'imx6qdlsabresd'
DISTRO ?= 'fslc-wayland'
PACKAGE_CLASSES ?= "package_deb"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"

DL_DIR ?= "${BSPDIR}/downloads/"
ACCEPT_FSL_EULA = "1"

DISTRO_FEATURES_append = " systemd opengl"
VIRTUAL-RUNTIME_init_manager = "systemd"

# for meta-webkit
# Accelerated video support
IMAGE_INSTALL_append = " gstreamer1.0-plugins-imx  gstreamer1.0-plugins-imx-meta"

# WPEWebkit and cog launcher
IMAGE_INSTALL_append = " wpewebkit cog"

# Backend selection
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"

PREFERRED_PROVIDER_virtual/libwpe = "wpebackend"
PREFERRED_VERSION_cog = "0.1%"
PREFERRED_VERSION_wpewebkit = "2.20%"
PREFERRED_VERSION_wayland = "1.13.%"
PREFERRED_VERSION_wayland-protocols = "1.10"
PREFERRED_VERSION_wpebackend-fdo = "0.1~git0.2"
PREFERRED_VERSION_weston_mx6 = "2.%"

IMAGE_INSTALL_remove = "gtk+3-demo"
```

I'm now going to test wpebackend-rdk/imx6 which should not need wayland.

So i guess i could change the distro to ```fsl-framebuffer``` and build the image ```core-image-base``` instead to get rid of wayland at all.

psaavedra, 2019-04-09:
> @psaavedra What exactly are you testing? Are you running some kind of performance benchmarks? I don't have a wanboard myself, but I think it would be extremely interesting to see some performance comparison (e.g. vivante vs etnaviv).

The original goal it was to provide a image for wandboards which for test and report results in https://browserperfdash.igalia.com/ . The progress in this goal target is not being as fast as I want. 

In the meantime the CI integrated in this BSP tries to keep in good shape the all the recipes of the meta-webkit testing the WPE builds against trunk and generating the SDK toolchains for crosscompiling

psaavedra, 2019-04-09:
> @psaavedra I tried now your recommended configuration but had a lot of issues:
> 
> I did not find the recipe for Weston 2.x in the repo you linked, there is only version 3. So i took it from openembedded-core - rocko. Also Weston 2 requires wayland-protocols 10.0 which i got from there as well.

I assume you are building for a Freescale board. Weston 2 is available in the rocko branch of the meta-freescale (https://github.com/Freescale/meta-freescale/tree/rocko/recipes-graphics/waylandrepo).

Anyway lately I was able to run Wa

> Because `imx-base.inc` sets `PREFERRED_VERSION_weston_mx6 ?= "4.0.0.imx"` I had as well to set `PREFERRED_VERSION_weston_mx6 = "2.0.0"`.
>
> This caused weston 2.0.0 to fail with the following error: [...]

closed 2021-04-12 by github-actions[bot]