Igalia/meta-webkit #65: error: redefinition of 'bool in JSCOptions.cpp building on trunk

psaavedra, 2019-02-03:
```
/home/igalia/psaavedra/yocto-wandboard-wpe/builds/wandboard-mesa-wpe-nightly/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/wpewebkit/nightly-AUTOINC+8d87c59385-r0/git/Source/JavaScriptCore/API/glib/JSCOptions.cpp:87:13: error: redefinition of 'bool valueFromGValue(const GValue*, size_t&)'
 static bool valueFromGValue(const GValue* gValue, size_t& value)
```

psaavedra, 2019-02-03:
Triggered by

```
commit a39285bfede3d35f4497b6a1e52eb87fde97cd8f
Author: carlosgc@webkit.org <carlosgc@webkit.org@268f45cc-cd09-0410-ab3c-d52691b4dbfc>
Date:   Thu Jan 24 11:09:54 2019 +0000

    [GLIB] Expose JavaScriptCore options in GLib public API
    https://bugs.webkit.org/show_bug.cgi?id=188742
    
    Reviewed by Michael Catanzaro.
    
    Source/JavaScriptCore:
    
    Add new API to set, get and iterate JSC options.
```

philn, 2019-02-04:
CC @carlosgcampos

psaavedra, 2019-02-05:
The yocto env uses GCC 8.2

psaavedra, 2019-02-05:
On systems using the GNU C Library, this will be unsigned int or unsigned long int.

Ref: https://www.gnu.org/software/libc/manual/html_node/Important-Data-Types.html

psaavedra, 2019-02-06:
this patch works but I guess it is just a workaround:

```
diff --git a/Source/JavaScriptCore/API/glib/JSCOptions.cpp b/Source/JavaScriptCore/API/glib/JSCOptions.cpp
index a9dc7e6c683..8dddffd42c5 100644
--- a/Source/JavaScriptCore/API/glib/JSCOptions.cpp
+++ b/Source/JavaScriptCore/API/glib/JSCOptions.cpp
@@ -73,6 +73,7 @@ static void valueToGValue(int32_t value, GValue* gValue)
     g_value_set_int(gValue, value);
 }
 
+#if !CPU(ARM_THUMB2) && !CPU(ARM64)
 static bool valueFromGValue(const GValue* gValue, unsigned& value)
 {
     value = g_value_get_uint(gValue);
@@ -83,6 +84,7 @@ static void valueToGValue(unsigned value, GValue* gValue)
 {
     g_value_set_uint(gValue, value);
 }
+#endif
 
 static bool valueFromGValue(const GValue* gValue, size_t& value)
 {
@@ -540,10 +542,12 @@ static JSCOptionType jscOptionsType(int)
     return JSC_OPTION_INT;
 }
 
+#if !CPU(ARM_THUMB2) && !CPU(ARM64)
 static JSCOptionType jscOptionsType(unsigned)
 {
     return JSC_OPTION_UINT;
 }
+#endif
 
 static JSCOptionType jscOptionsType(size_t)
 {
```

psaavedra, 2019-02-06:
Fixed on upstream: https://bugs.webkit.org/show_bug.cgi?id=194330

closed 2019-02-07 by psaavedra