Igalia/meta-webkit #70: wpewebkit 2.22.4 with experimental features doesn't build

philn, 2019-02-14:
`PACKAGECONFIG_append_pn-wpewebkit = " experimental-features"`

```
| CMake Error at Source/CMakeLists.txt:28 (add_subdirectory):
|   add_subdirectory given source "ThirdParty/openvr" which is not an existing                                                                                                                                      
|   directory.                                                                                                                                                                                                      
|                                           
```

philn, 2019-02-14:
@aperezdc FYI :)

philn, 2019-02-23:
This looks more like a workaround, right? Why isn't openjpg dependency enabled through the openjpeg packageconfig option which is enabled by default?

psaavedra, 2019-02-24:
OK. I  didn't understand well the problem.

psaavedra, 2019-02-24:
I can't replicate the issue. Could provide the output of `bitbake -e wpewebkit > output.log` and the `log.do_configure` ?

philn, 2019-02-24:
It looks like a packaging issue, the `openvr` directory is missing from the tarball. That's why I was mentioning Adrian.

I think you got confused with #72 which is the actual issue about openjpeg.

psaavedra, 2019-02-25:
> It looks like a packaging issue, the `openvr` directory is missing from the tarball. That's why I was mentioning Adrian.
> 
> I think you got confused with #72 which is the actual issue about openjpeg.

yes. I  just moved this comment to the right place. Sorry for the noise.

aperezdc, 2019-03-18:
We will need to start shipping the `openvr/` subdirectory in the tarballs. I have filed an [upstream bug](https://bugs.webkit.org/show_bug.cgi?id=195889) for this, and will try to have the fix applied to the 2.24 release branch as well.

aperezdc, 2019-03-18:
While we don't have a resolution upstream, something that may work for you as long as you are interested in experimental features which are *not* the VR support or WebRTC, I would suggest to add individual flags to the Bitbake recipe for each of the features you are interested on. Toggling them individually is more likely to work instead of using a catch-all flag.

commit a06311ad (2019-02-23): [wpewebkit] experimental-features requires openjpeg (Fix: #70)
PR #82 (2019-03-20): wpewebkit: Remove experimental features build support
Individual experimental features should be enabled one by one instead.

Fixes #70
commit 707a1e37 (2019-03-20): wpewebkit: Remove experimental features build support

Individual experimental features should be enabled one by one instead.

Fixes #70
closed 2019-03-20 by psaavedra