Igalia/meta-webkit #72: wpewebkit openjpeg build fails

philn, 2019-02-20:
```
| CMake Warning at Source/cmake/OptionsWPE.cmake:103 (find_package):
|   By not providing "FindOpenJPEG.cmake" in CMAKE_MODULE_PATH this project has
|   asked CMake to find a package configuration file provided by "OpenJPEG",
|   but CMake did not find one.
| 
|   Could not find a package configuration file provided by "OpenJPEG" with any
|   of the following names:
| 
|     OpenJPEGConfig.cmake
|     openjpeg-config.cmake
| 
|   Add the installation prefix of "OpenJPEG" to CMAKE_PREFIX_PATH or set
|   "OpenJPEG_DIR" to a directory containing one of the above files.  If
|   "OpenJPEG" provides a separate development package or SDK, be sure it has
|   been installed.
| Call Stack (most recent call first):
|   Source/cmake/WebKitCommon.cmake:56 (include)
|   CMakeLists.txt:159 (include)
| 
| 
| CMake Error at Source/cmake/OptionsWPE.cmake:105 (message):
|   libopenjpeg is needed for USE_OPENJPEG.
| Call Stack (most recent call first):
|   Source/cmake/WebKitCommon.cmake:56 (include)
|   CMakeLists.txt:159 (include)
| 
| 
| -- Configuring incomplete, errors occurred!
```

philn, 2019-02-20:
That's with a git snapshot of wpewebkit though

psaavedra, 2019-02-25:
I can't replicate this issue in my local environment. Could provide the output of `bitbake -e wpewebkit > output.log` and the `log.do_configure` ?

philn, 2019-02-25:
What do you need from `log.do_configure` that's not already in the description of the issue above? I'll share the env log privately.

psaavedra, 2019-02-25:
> What do you need from `log.do_configure` that's not already in the description of the issue above? I'll share the env log privately.

I'd like to check the cmake configure flags output against the one generates by my in local.

philn, 2019-03-13:
Which meta-openembedded branch do you use? I had to copy the openjpeg 2.3 recipe from master to make this issue disappear here: https://github.com/openembedded/meta-openembedded/tree/master/meta-oe/recipes-graphics/openjpeg

psaavedra, 2019-03-13:
Thud

philn, 2019-03-13:
So that explains why you didn't have the issue :) Would it be ok to backport that recipe in meta-webkit then?

psaavedra, 2019-03-13:
Actually master is usually oriented to run over current stable Yocto
release. Everytime that an new stable is released, we create a new branch.

I'd suggest to add the missing dependencies in the sumo branch and.

philn, 2019-03-13:
And what?

philn, 2019-03-13:
Our sumo branch doesn't have WPE 2.23 (upcoming 2.24)

psaavedra, 2019-03-13:
> And what?

...and nothing else :-)

psaavedra, 2019-03-13:
> Our sumo branch doesn't have WPE 2.23 (upcoming 2.24)

If we want to keep updated the `sumo` branch, one idea could be to pick the following changes (already in master)  into it:

```
* 5a8ab19 Revert "[wpewebkit] Temporary commenting out of bubblewrap PACKAGECONFIG"
* ff2f817 Various changes in the recipes
* e0d5fb3 [wpewebkit] Temporary commenting out of bubblewrap PACKAGECONFIG
* b1610c6 [bubblewrap][xdg-dbus-proxy] Fix native build
* d1994c2 Update README.md
* 745e224 recipes-qt: Use Yocto common license file for qt-wpe-simple-browser
* 7e5317c Use tarballs for bubblewrap and xdg-dbus-proxy (#77)
* 67e161c Add libseccomp as build dependency (#77)
* 93104f4 Add xdg-dbus-proxy as build dependency (#77)
* c9c9eb0 Add bubblewrap as build dependency (#77)
* 02095f7 [libwpe] Bump git recipe to 1.2
* 67cff94 Relabeling of the wpebackend-fdo_git version
* e7109c7 libWPEBackend-fdo-1.0.so installed but not shipped (fixes #75)
* 6177dc9 recipes-qt: Add a sample QtWPE mini browser application
* 75024a2 Revert "[wpewebkit] experimental-features requires openjpeg (Fix: #70)"
* a06311a [wpewebkit] experimental-features requires openjpeg (Fix: #70)
* 724b332 [1.0.0~git wpebackend-fdo] build failure: /bin/sh: wayland-scanner: command not found (Fix: #73)
* 7d74ffb wpebackend-fdo: Bump to 1.0.1
* 280f526 webkitgtk: Bump to 2.22.6
* f3fbf41 wpewebkit: Bump to 2.22.4
* c1bcb24 Add new PACKAGECONFIG USE_OPENJPEG (#69)
* f1f5ad9 wpewebkit: New build options for experimental features, service workers and webgl2
* 83d230c wpewebkit: Enable Ninja, ccache and Release build configuration
* a240d37 wpewebkit: Bump to 2.22.3
* 503e978 wpewebkit: QtWPE build support
* b6b94d2 cog: bump to 0.1.1
```

psaavedra, 2019-03-13:
Anyway, I'd prefer to keep aligned the Webkit releases respect the Yocto releases a long the time. If someone needs to use later versions always they can overwrite and extend the recipes in they custom layers.

philn, 2019-03-22:
OK so that means no WPE 2.24 in sumo?
I have a BSP I need to keep in sumo for the time being, unfortunately, it uses a forked Poky that still has GCC 6.5.0... :/ So if we could have 2.24 in sumo it would be great. OpenJPEG support could be disabled there, so no need to backport the update recipe.

psaavedra, 2019-03-22:
> OK so that means no WPE 2.24 in sumo?
> I have a BSP I need to keep in sumo for the time being, unfortunately, it uses a forked Poky that still has GCC 6.5.0... :/ So if we could have 2.24 in sumo it would be great. OpenJPEG support could be disabled there, so no need to backport the update recipe.

this sounds good to me

philn, 2019-05-27:
I'll move to Warrior very soon. Perhaps we can leave the sumo branch as-is then?

psaavedra, 2020-04-01:
Let's close the issue then?

philn, 2020-04-01:
Sure!

closed 2020-04-01 by philn