Igalia/meta-webkit #75: libWPEBackend-fdo-1.0.so installed but not shipped

psaavedra, 2019-02-26:
ERROR: recipe wpebackend-fdo-1.0.0~git-r0: task do_populate_sysroot: Succeeded
ERROR: wpebackend-fdo-1.0.0~git-r0 do_package: QA Issue: wpebackend-fdo: Files/directories were installed but not shipped in any package:
  /usr/lib/libWPEBackend-fdo-1.0.so
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid install

**Root cause: **

Bump API version to 1.0 in preparation for the 1.2.x series

https://github.com/Igalia/WPEBackend-fdo/commit/b5de2adbbbdba391b54d0c014f35fca431ecd520

psaavedra, 2019-02-26:
that's for WPE WebKit 2.24, @aperezdc  is going to merge it to the release branch today. Will be in wpebackend-fdo 1.2.x as well, also will make the release branch today.

aperezdc, 2019-02-26:
The [wpebackend-fdo-1.2 release branch](https://github.com/Igalia/WPEBackend-fdo/tree/wpebackend-fdo-1.2) is now available. Later I will be doing a 1.3.90 release candidate.

commit e7109c72 (2019-03-06): libWPEBackend-fdo-1.0.so installed but not shipped (fixes #75)
closed 2019-03-06 by psaavedra