Igalia/meta-webkit #76: Building a WPE Browser pkg-config

estobbart, 2019-02-28:
The pkg-config for wpe-webkit-0.1 doesn't seem to include the header locations needed to be able to compile a browser. Is there a recommendation for the include paths or example recipe or pkg-config with a demo browser that could be referenced?

philn, 2019-02-28:
Please try the cog recipe. It provides a minimal WPE-based browser

estobbart, 2019-03-05:
Thanks, the COG build helped. I was somehow able to build WPE without MediaSource (RPi), any pointers on what I might've misconfigured in the build?

philn, 2019-03-06:
Maybe I can help if you post the log of the failing wpewebkit (with MSE enabled) configure step.

estobbart, 2019-03-08:
```
d$ cat tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/temp/log.do_configure
DEBUG: Executing python function externalsrc_configure_prefunc
DEBUG: Python function externalsrc_configure_prefunc finished
DEBUG: Executing shell function do_configure
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- Check for working C compiler: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc
-- Check for working C compiler: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++
-- Check for working CXX compiler: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The CMake build type is: Release
-- Found Gperf: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/gperf (Required is at least version "3.0.1") 
-- Found Perl: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/perl-native/perl (found suitable version "5.28.1", minimum required is "5.10.0") 
-- Found PerlModules_JSON::PP: TRUE  
-- Found PerlModules: TRUE  found components:  JSON::PP 
-- Found PythonInterp: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/python-native/python2.7 (found suitable version "2.7.15", minimum required is "2.7.0") 
-- Found Ruby: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/recipe-sysroot-native/usr/bin/ruby (found suitable version "2.5.3", minimum required is "1.9") 
-- Performing Test C_COMPILER_SUPPORTS_-fno-strict-aliasing
-- Performing Test C_COMPILER_SUPPORTS_-fno-strict-aliasing - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-strict-aliasing
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-strict-aliasing - Success
-- Performing Test C_COMPILER_SUPPORTS_-fno-exceptions
-- Performing Test C_COMPILER_SUPPORTS_-fno-exceptions - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-exceptions
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-exceptions - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-rtti
-- Performing Test CXX_COMPILER_SUPPORTS_-fno-rtti - Success
-- Performing Test CXX_COMPILER_SUPPORTS_CXX14
-- Performing Test CXX_COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-std=c++14
-- Performing Test CXX_COMPILER_SUPPORTS_-std=c++14 - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wall
-- Performing Test C_COMPILER_SUPPORTS_-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wextra
-- Performing Test C_COMPILER_SUPPORTS_-Wextra - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wcast-align
-- Performing Test C_COMPILER_SUPPORTS_-Wcast-align - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wformat-security
-- Performing Test C_COMPILER_SUPPORTS_-Wformat-security - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wmissing-format-attribute
-- Performing Test C_COMPILER_SUPPORTS_-Wmissing-format-attribute - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wpointer-arith
-- Performing Test C_COMPILER_SUPPORTS_-Wpointer-arith - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wundef
-- Performing Test C_COMPILER_SUPPORTS_-Wundef - Success
-- Performing Test C_COMPILER_SUPPORTS_-Wwrite-strings
-- Performing Test C_COMPILER_SUPPORTS_-Wwrite-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall
-- Performing Test CXX_COMPILER_SUPPORTS_-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra
-- Performing Test CXX_COMPILER_SUPPORTS_-Wextra - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align
-- Performing Test CXX_COMPILER_SUPPORTS_-Wcast-align - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat-security
-- Performing Test CXX_COMPILER_SUPPORTS_-Wformat-security - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-format-attribute
-- Performing Test CXX_COMPILER_SUPPORTS_-Wmissing-format-attribute - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpointer-arith
-- Performing Test CXX_COMPILER_SUPPORTS_-Wpointer-arith - Success
-- Performing Test CXX_COMPILER_SUPPORTS_-Wundef
-- Performing Test CXX_COMPILER_SUPPORTS_-Wundef - Success
[... 192 lines cut ...]
-- Using platform-specific CMakeLists: /home/rdk/Development/rpi-yocto/rpi-wpe-build/workspace/sources/wpewebkit/Source/WebDriver/PlatformWPE.cmake
[... cut ...]
--  USE_WOFF2 ............................. ON
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ENABLE_WPE_QT_API
    USE_OPENJPEG

-- Build files have been written to: /home/rdk/Development/rpi-yocto/rpi-wpe-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.22.4-r0/wpewebkit-2.22.4
DEBUG: Shell function do_configure finished
DEBUG: Executing python function do_qa_configure
DEBUG: Python function do_qa_configure finished
```

estobbart, 2019-03-08:
`cog http://<ip>:8080/index.html`
```console.log(`typeof MediaSource ${typeof window.MediaSource}`);```
`log: typeof MediaSource undefined`

philn, 2019-03-08:
You need to enable MSE at runtime. With `cog --enable-mediasource=1` ... See also `cog --help-websettings`.

estobbart, 2019-03-09:
That's new... that fixed it though! Thanks!

closed 2019-03-09 by estobbart