Igalia/meta-webkit #79: WebRTC Webcam & Mic permission

mibstar, 2019-03-07:
Hi,

I'm looking for WebRTC support for Video calls and while I appreciate there seems to be on-going work on WebRTC I've just tried https://appr.tc / (based off https://github.com/webrtc/apprtc) and I only get a blank screen. 

I'm assuming the browser is asking permission to access the webcam and mic. It it possible to pre-bake or white list what sites have access?

I'm testing on an rpi3 with the following:

**Device Name**
buildroot S/N  WPEuCfrS13Z 
Identifier  WPEuCfrS13Z 
Version  1.0.100355f#100355f78713cabf8b8b40db0f20580587653763

Many thanks in advance.

philn, 2019-03-07:
I think you should report your issue in this project, https://github.com/WebPlatformForEmbedded/buildroot
meta-webkit is a Yocto layer for the upstream (wpewebkit.org) project and has nothing to do with buildroot :)

closed 2019-03-07 by philn

mibstar, 2019-03-07:
Many thanks @philn ... I was going to post on https://lists.webkit.org/mailman/listinfo/webkit-wpe but I've not been able to subscribe. 

Also rather annoyingly the https://github.com/WebPlatformForEmbedded/buildroot repo does not have any issues so I can't add 🙃

aperezdc, 2019-03-07:
@mibstar Make sure the confirmation e-mail you get when trying to subscribe didn't end up in your spam box. If your problem persists, please do send a message to `webkit-wpe-owner@lists.webkit.org` and we can handle it manually (unless it was you who just sent a little while ago today :wink:).