Igalia/meta-webkit #81: unable to build wpebackend-fdo

estobbart, 2019-03-17:
```| CMake Error at /home/es/Development/rpi-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpebackend-fdo/1.0.1-r0/recipe-sysroot-native/usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find WPE (missing: WPE_INCLUDE_DIRS WPE_LIBRARIES)```
I've seen this issue with a number of different configurations.. did something change upstream?

philn, 2019-03-17:
Which version of libwpe do you have in the sysroot?

estobbart, 2019-03-17:
It's not in the sysroot.. but it's in the depends..
`DEPENDS += " libwpe"`
is there a virtual/libwpe also?
`../meta-webkit/recipes-browser/libwpe/libwpe.inc:PROVIDES += "virtual/libwpe"`

philn, 2019-03-17:
It depends... which meta-webkit branch are you using?

estobbart, 2019-03-17:
`master`

estobbart, 2019-03-17:
Same issue with the rdk backend..
```
| CMake Error at /home/es/Development/rpi-yocto/rpi-ml-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpebackend-rdk/1.20180925-r0/recipe-sysroot-native/usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find WPE (missing: WPE_INCLUDE_DIRS WPE_LIBRARIES)
| Call Stack (most recent call first):
```

philn, 2019-03-17:
Add this (or similar) to local.conf? `PREFERRED_VERSION_virtual/libwpe = "1.0.1"`

estobbart, 2019-03-17:
You were right about the version issue..
I've got 1.0 in my sysroot & it's looking for 0.2

https://github.com/WebPlatformForEmbedded/WPEBackend-rdk/blob/master/cmake/FindWPE.cmake#L32

My fork fixes it..
https://github.com/estobbart/WPEBackend-rdk/commit/c275769547745e10b8fdb8bfb792d869c2469786
Am I mis-configured somehow or should I open this commit as a pull request?

philn, 2019-03-17:
Seems like -rdk needs a PR indeed. Is that right @aperezdc ?

estobbart, 2019-03-17:
It looks like a number of things are looking for 0.2
https://github.com/WebPlatformForEmbedded/WPEWebKit/commit/eaf2a3f93a64dc62b42b9ec8b71ef2b297d319e0

philn, 2019-03-17:
If you want to build the downstream flavour of WPEWebKit, you need to use this recipe: https://github.com/Igalia/meta-webkit/blob/master/recipes-browser/wpewebkit/wpewebkit_downstream.bb

What's the actual issue you're having, exactly?

estobbart, 2019-03-17:
just trying to get through a compile using the meta-webkit recipe set

estobbart, 2019-03-17:
It doesn't look like `libwpe_0.2.0.bb` exists.. I think that's where everything else is failing downstream.

philn, 2019-03-17:
You're confusing API version and package version... Version 1.0.0 provides API version 0.2. https://github.com/WebPlatformForEmbedded/libwpe/releases/tag/1.0.0

estobbart, 2019-03-17:
Ahh... so pulling in v. 1.1.0 is why I'm getting api 1.0
```
SET_PROJECT_VERSION(1 1 0)
set(WPE_API_VERSION "1.0")
```

philn, 2019-03-19:
Is there an issue left here?

estobbart, 2019-03-19:
Nope, thank you @philn

closed 2019-03-19 by estobbart