Igalia/meta-webkit #84: README and dependencies

court-jus, 2019-03-26:
Hi,

The README says we need the `master` branches for `openembedded-core` and `meta-openembedded` layers and also the `5.12` branch for `meta-qt5`.

There is no `5.12` branch or tag on the `meta-qt5` layer and also it seems to me that if building a `thud` version of yocto we should be able to use the `thud` versions of the layers.

Are there any dependencies in the `master` layers that are not available in `thud` branches ?

philn, 2019-03-26:
meta-qt5 does have a 5.12 branch: https://code.qt.io/cgit/yocto/meta-qt5.git/log/?h=5.12

I don't know the status of Qt5 in thud in general. Is there another layer providing 5.12 in a thud branch?

philn, 2019-03-26:
meta-qt5 5.12 branch has `LAYERSERIES_COMPAT_qt5-layer = "sumo thud"` so I think we're safe.

philn, 2019-03-29:
Closing then, please reopen if you have more feedback :)

closed 2019-03-29 by philn