Igalia/meta-webkit #88: Current status of master and sumo (or it's Documentation)

miikeat, 2019-04-05:
I tried to rebuild setups from about a month ago and suddently none of my images were working.
As mentioned in Issue #81 its "just" a version issue.

Could you please somewhere list which versions work together to get a working build or get rid of the ```_gitAUTOINC``` recipies. At the moment sumo and thud use the same version of cog which is probably the cause for these issues as it just got updated!

The error i got in thud were:
(without any ```PREFERRED_VERSION```)

```
ERROR: wpebackend-rdk-1.20180925-r0 do_configure: Function failed: do_configure
...
| CMake Error at .../build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpebackend-rdk/1.20180925-r0/recipe-sysroot-native/usr/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
|   Could NOT find WPE (missing: WPE_INCLUDE_DIRS WPE_LIBRARIES)
```
In sumo i got:
```
ERROR: cog-gitAUTOINC-r0 do_configure: Function failed: do_configure 
...
| -- Checking for module 'wpe-webkit-1.0>=2.23.91'
| --   No package 'wpe-webkit-1.0' found
| CMake Error at .../build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cog/gitAUTOINC-r0/recipe-sysroot-native/usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
|   A required package was not found
```
as the only other version of cog in sumo is 0.1.0 i selected that which lead to this error:
```
ERROR: cog-0.1.0-r0 do_configure: Function failed: do_configure 
...
| -- Checking for modules 'wpe-webkit-0.1;wpe-0.1'
| --   No package 'wpe-0.1' found
| CMake Error at .../build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/cog/0.1.0-r0/recipe-sysroot-native/usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
|   A required package was not found
```

I managed to get my build finishing again by copying ```cog_0.2.0.bb``` from thud to sumo and selecting that version.

I'm not sure if it's intended to pick the preferred version manually (if so could you add some documentation which versions work together) or if this is just a momentarily issue because of some version upgrades...

Wouldn't it be possible to make sample projects for RPi, i.MX6, x86,... and test them after each commit to master/sumo/rocko... . If testing them would be too much, just having them would be a great resource for people to start and it would be a great addition to the wiki. Also if such version issues occure again, it would be easy to spot them when the examples projects aren't working.

miikeat, 2019-04-06:
I just stumbled upon this table which might be useful:
 
WPE WebKit | libwpe | WPEBackend-fdo
-- | -- | --
2.24.x | 1.2.x | 1.2.x
2.22.x | 1.0.x | 1.0.x
2.20.x | < 1.0.0 | < 1.0.0

https://wpewebkit.org/release/schedule/ (scroll down to compatible components)

I just checked the different versions of cog and gave it a try to extend the above table by its dependencies:

For the different Versions i found the following version checks:

Cog 0.1.0, 0.1.1, 0.1.2: ```pkg_check_modules(WEB_ENGINE REQUIRED wpe-webkit-0.1 wpe-0.1)```
Cog 0.2.0: ```pkg_check_modules(WEB_ENGINE REQUIRED wpe-webkit-0.1>=2.21.90 wpe-0.2)```
Cog 0.3.0: ```pkg_check_modules(WEB_ENGINE REQUIRED wpe-webkit-1.0>=2.23.91)```

So i'd say this should be about right:

Cog | WPE WebKit | libwpe (API Verison) | WPEBackend-fdo | WPEBackend-rdk
-- | -- | -- | -- | --
0.3.x | 2.24.x | 1.2.x (API: 1.0) | 1.2.x | ?
0.2.x | 2.22.x | 1.0.x (API: 0.2)| 1.0.x | ? 
0.1.x | 2.20.x | < 1.0.0 (API: 0.1) | < 1.0.0 | ?

For WPEBackend-rdk i haven't found any dependencies... maybe someone else knows more and then this table could be added to the wiki.

HowardACornwell, 2019-04-07:
I'm having the same issue. As a total beginner in this realm, following the RPi docs has led me into dead-ends, and I've not got sufficient knowledge debug the issues myself.

psaavedra, 2019-04-23:
Thanks for the contribution. I think we could  include this information in the Readme file or in some page of the Wiki.

BTW, the  meta-perf-browser already includes a set of preset files which details these dependencies/versions as well:

https://gitlab.com/saavedra.pablo/meta-perf-browser/tree/master/conf/template/presets

For the RDK case, there is not yet a valid version for 2.14.x, this is the reason why we have userland-wpe-2_22.conf but not userland-wpe-2_24.conf yet

https://gitlab.com/saavedra.pablo/meta-perf-browser/blob/master/conf/template/presets/userland-wpe-2_22.conf

miikeat, 2019-04-23:
Just an idea... Would it be possible to add the meta-perf-browser configurations to meta-webkit and just include/require the preferred configuration in local.conf? Or is there even something like configurationgroups???

psaavedra, 2019-04-23:
> Just an idea... Would it be possible to add the meta-perf-browser configurations to meta-webkit and just include/require the preferred configuration in local.conf? Or is there even something like configurationgroups???
> 
Sounds like a interesting thing try to explore the idea that you suggest but I'm know if we could explicit the versions required with packagegroups. Alternatively, we could define images/distros for demo purpose in this repository as well. The motivation for meta-perf-browsers was in origin to provide the machine definitions and those configuration example for specific hardware that I use for test this repository. 

At this point, I prefer to keep separated the meta-webkit and the meta-perf-browser but maybe  it is time to update the Wiki pages, document much better the dependencies and provide Quick start guides 

> A Wiki page with the possible configurations and the differences would then be easy to maintain.

psaavedra, 2019-04-23:
> A Wiki page with the possible configurations and the differences would then be easy to maintain.

I wonder if  you can check the presets  that I link you before. If they are OK I'd glad to work on it documenting this configurations and dependencies in the Wiki

closed 2021-03-11 by github-actions[bot]