Igalia/meta-webkit #91: Since wpewebkit 2.24.x wpebackend-rdk/fdo needs to be manually installed

miikeat, 2019-04-08:
as discussed in #89 (in the end) when building meta-webkit with cog 0.3.0 which depends on wpewebkit 2.24.x you need to manually add the respective wpebackend to your image.

wpewebkit <= 2.22 seem to bepend on wpebackend via RCONFLICTS (it seems like RCONFLICTS is a specialized RDEPENDS)

philn, 2019-04-14:
Cog doesn't depend on any specific WPEBackend. The build-time dependency on -fdo is optional. At runtime the backend has to be selected. What's the issue with that?

miikeat, 2019-04-14:
The "issue" is, that only since wpewebkit 2.24.x you have to add ```IMAGE_INSTALL_append = " wpebackend-rdk"```. Before ```wpebackend-rdk``` was installed automatically.

It's no "issue" that it's required to install it manually, its just that nowhere in the wiki it's mentioned:

RPi-rdk:
```
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"
IMAGE_INSTALL_append = " wpewebkit cog"
```
RPi-fdo:
```
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL_append = " wpewebkit cog"
```
WPE Page:
```
IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```
Wandboard:
```
# Accelerated video support
IMAGE_INSTALL_append = " gstreamer1.0-plugins-imx  gstreamer1.0-plugins-imx-meta"

# WPEWebkit and cog launcher
IMAGE_INSTALL_append = " wpewebkit cog"

# Backend selection
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```

So from my point of view i'd say the issue is, that the wiki does not say so that i need to manually install it and till now it worked like that...

philn, 2019-04-15:
Actually I removed the `IMAGE_INSTALL_append = " wpebackend-fdo"` and the backend still gets installed, somehow.

miikeat, 2019-04-15:
In that case, maybe it still works for ```wpebackend-fdo```, but not for ```-rdk```

philn, 2019-04-15:
At least one problem is that wpewebkit doesn't `RDEPENDS` on `virtual/wpebackend`.

commit 9ddb737b (2019-04-23): recipes-browser: wpewebkit: Add runtime dependency for wpebackend

WPE can't run without backend. This change ensures the selected wpebackend
provider is installed on the rootfs image.

Fixes #91
commit e8477d7a (2019-04-23): recipes-browser: wpewebkit: Add runtime dependency for wpebackend

WPE can't run without backend. This change ensures the selected wpebackend
provider is installed on the rootfs image.

Fixes #91
closed 2019-04-23 by psaavedra