Igalia/meta-webkit #92: Low RPi3 B+ canvas performance / frame rate

HowardACornwell, 2019-04-10:
I've built the `wpebackend-rdk` and `wpebackend-fdo` images in the [RPi Wiki](https://github.com/Igalia/meta-webkit/wiki/RPi) and followed all the suggestions in [PerformanceTips](https://github.com/Igalia/meta-webkit/wiki/PerformanceTips), but my RPi3 B+ is struggling to get past 30fps.

The [igalia fps counter](https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html) reaches ~25-28fps. This [canvas performance test](https://smashcat.org/av/canvas_test/) briefly reaches 25fps, before locking up & rendering at ~2fps. The [webkit poster circle](https://webkit.org/blog-files/3d-transforms/poster-circle.html) renders >50fps on `-rdk` but is much slower using `-fdo` inside Weston.

I'm comparing my results to [this video](https://youtu.be/_KsCrg0kyac?t=30), where the canvas test at 0m30s gets ~70fps.

I've tried loading simple / empty THREE.js scenes that should easily reach 60fps, but they're all stuck around ~25-30 as well.

Utilization while running each of these is far from maxed out, if you trust `top`.

```shell
# raf_onload
Mem: 261092K used, 363052K free, 9292K shrd, 5464K buff, 78424K cached
CPU:   3% usr   2% sys   0% nic  93% idle   0% io   0% irq   0% sirq
Load average: 0.20 0.12 0.05 2/119 1991
```

```shell
# canvas_test
Mem: 289292K used, 334852K free, 9300K shrd, 5480K buff, 79088K cached
CPU:  10% usr   1% sys   0% nic  88% idle   0% io   0% irq   0% sirq
Load average: 0.56 0.33 0.14 2/121 2189
```

```shell
# poster-circle
Mem: 266468K used, 357676K free, 9300K shrd, 5472K buff, 78432K cached
CPU:   3% usr   1% sys   0% nic  95% idle   0% io   0% irq   0% sirq
Load average: 0.74 0.27 0.11 3/120 2152
```

Any ideas?

miikeat, 2019-04-11:
In the video you link to the author links to [his blog](https://medium.com/@decrocksam/building-wpe-webkit-for-raspberry-pi-3-cdbd7b5cb362) and there he is using [meta-wpe](https://github.com/WebPlatformForEmbedded/meta-wpe).

I haven't found out whats the difference between meta-webkit and meta-wpe except that they use a different launcher (Cog vs. "WPEWebKit Browser")

Maybe meta-wpe performs better than meta-webkit???

philn, 2019-04-13:
meta-webkit relies on https://github.com/WebPlatformForEmbedded/WPEWebKit which is a downstream fork of the upstream WPEWebKit port available from wpewebkit.org.

miikeat, 2019-04-14:
@HowardACornwell: Because I was curious I gave ```meta-wpe``` a try and i ran now the tests you ran on my RPi3 B (not plus) without overclocking and ```wpebackend-rdk```:

* The igalia fps counter: 54 fps
* The canvas performance starts at 35 fps and then goes up to 41 fps
* The webkit poster circle: 59 fps

So i'd say with your RPi3 B**+** you should get slightly higher fps than me and if you also overclock the Pi like Sam Decrock did it in his blog (checkout the config.txt he his in the image he offers for download) you should get even better values.

But that [the canvas performance test](https://smashcat.org/av/canvas_test/) only reaches 2fps is either some kind of issue with meta-webkit or the wiki is missing some configuration that is required!

The configs I used:

### meta-webkit:
On branch master but as wpebackend-rdk currently does not build with cog 0.3.0 / libwpe 1.2.0 / wpewebkit 2.24 I'm still one version behind.
```
MACHINE = "raspberrypi3"

GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"

IMAGE_FEATURES += " hwcodecs ssh-server-openssh"

PREFERRED_VERSION_cog = "0.2.0"
PREFERRED_VERSION_libwpe = "1.0.%"
PREFERRED_VERSION_wpewebkit = "2.22.%"

IMAGE_INSTALL_append = " wpebackend-rdk"

PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"

DISTRO_FEATURES_append = " opengl wayland"
IMAGE_INSTALL_append = " \
                        wpewebkit \
                        cog \
                        xkeyboard-config \
                        kbd-keymaps \
                        glibc-utils \
                        glibc-charmaps \
                        localedef \
                        packagegroup-core-ssh-openssh \
                        openssh-sftp-server \
                        htop \
                        gstreamer1.0-omx \
                        "
LICENSE_FLAGS_WHITELIST_append = " commercial"

PREFERRED_VERSION_cog = "0.2.0"
PREFERRED_VERSION_libwpe = "1.0.%"
PREFERRED_VERSION_wpewebkit = "2.22.%"

IMAGE_INSTALL_append = " wpebackend-rdk"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"

DISTRO_FEATURES_append = " opengl wayland"
IMAGE_INSTALL_append = " gstreamer1.0-omx"
LICENSE_FLAGS_WHITELIST_append = " commercial"
```

### meta-wpe:
For all branches I used ```thud``` and I built for my ```raspberrypi3``` a ```wpe-eglfs-image```.
The configuration is as generated by yoe and the additions neccessary for ```wpe-eglfs-image```
```
DISTRO_FEATURES_remove_rpi = "x11"
DISTRO_FEATURES_remove_rpi = "wayland"

WPE_BACKEND = "rpi libinput"

#to remove virtualinput from PACKAGECONFIG
PACKAGECONFIG_pn-wpebackend-rdk = "${WPE_BACKEND}"

WPE_PLATFORM = "egl touch"
```

HowardACornwell, 2019-04-17:
I've built using your two configs and get similar results. Better than I was getting, but not quite as fast as the meta-wpe build. I'll try overclocking and see if I can bring it any closer.

Closing this issue because I can't provide any useful information as to why my config was slower, or why the canvas performance test was struggling.

closed 2019-04-17 by HowardACornwell