Igalia/meta-webkit #100: [wpewebkit] build failure: 'IDBTransaction' has not been declared

mrstinky, 2019-04-29:
wpewebkit_2.24.1.bbappend:
PACKAGECONFIG_pn-wpewebkit = "remote-inspector"

local.conf:
IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"

Build fails with:
/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.24.1-r0/wpewebkit-2.24.1/Source/WebCore/testing/Internals.cpp: In member function 'unsigned int WebCore::Int
ernals::numberOfIDBTransactions() const':
/poky/build/tmp-glibc/work/aarch64-oe-linux/wpewebkit/2.24.1-r0/wpewebkit-2.24.1/Source/WebCore/testing/Internals.cpp:2388:12: error: 'IDBTransaction' has not been 
declared
     return IDBTransaction::numberOfIDBTransactions;
            ^~~~~~~~~~~~~~

philn, 2019-05-15:
The syntax is `PACKAGECONFIG_append_pn-wpewebkit = " remote-inspector"`

mrstinky, 2019-05-15:
Thanks! I see what happened. I moved the PACKAGECONFIG string from my local.conf in to the bbappend, and I forgot to remove the -wpewebkit. 

It still worked though, and I was able to build the previous version (wpewebkit_2.20.2, cog_0.1.0, wpebackend-rdk) without any problems.  I don't want to _append as this is a minimal build for a simple HTML/Javascript based display.

philn, 2019-05-15:
So is there an issue to fix? I can't reproduce this build error here.

mrstinky, 2019-05-15:
yes, but it's not a high priority issue for us, I was just trying to see if there was an easy upgrade path to using backend fdo, and a newer version of wpewebkit.

I'll try again in a few months once we get all our codebase and recipes moved from Yocto Rocko to Thud. I'm happy to close this if you can't reproduce it.

philn, 2019-05-15:
That's fine, let's keep this open for the time being :)

closed 2021-03-11 by github-actions[bot]