Igalia/meta-webkit #101: Google fonts not working in 2.24.1 ?

marhoy, 2019-04-30:
I have built an image with cog-0.3, libwpe-1.2 and wpewebkit-2.24.1, but the fonts look strange compared to my previous image with cog-0.2, libwpe-1.0 and wpewebkit-2.22.5.

I didn't find a way to take proper screenshots (ideas happily accepted), so these are iPhone-shots. Look closely at the date/time in the upper left, or the 1-digit in the temperatures:
- The [2.22.5 look](https://www.hoy.priv.no/Screen-2.22.5.jpg)
- The [2.24.1 look](https://www.hoy.priv.no/Screen-2.22.5.jpg)

The font on this page is delivered via Google Fonts in the .css:
```
@import url('https://fonts.googleapis.com/css?family=Noto+Sans');
body {
...
  font-family: 'Noto Sans', sans-serif;
}
```

It seems that for version 2.24.1, the Google font is not used and wpewebkit instead falls back to the locally installed Bitstream Vera Sans.

Why isn't the Google font used?
Anything I can compile/install/configure differently?

The relevant part of the image config looks like this:
```
IMAGE_INSTALL_append += " \
	fontconfig \
	fontconfig-utils \
	ttf-bitstream-vera \
	wpebackend-rdk \
	wpewebkit \
	cog \
	gstreamer1.0-plugins-base-alsa \
"
```

PS: The docker images with the compiled wpewebkit-versions are available here:
`marhoy/balena-wpe:raspberrypi3-2.22.5-220cac4`
`marhoy/balena-wpe:raspberrypi3-2.24.1-test2`
See [balena-wpe](https://github.com/balena-io-projects/balena-wpe) for one way to use them.

petrosagg, 2019-05-01:
@marhoy I think you accidentally linked the same image in both cases

marhoy, 2019-05-01:
> @marhoy I think you accidentally linked the same image in both cases

Argh. Fixed now...

philn, 2019-05-15:
Perhaps this is because variation fonts support is disabled? Check this snippet in OptionsWPE.cmake:

```
# Enable variation fonts when cairo >= 1.16, fontconfig >= 2.13.0, freetype >= 2.9.0 and harfbuzz >= 1.4.2.
if (("${PC_CAIRO_VERSION}" VERSION_GREATER "1.16.0" OR "${PC_CAIRO_VERSION}" STREQUAL "1.16.0")
    AND ("${PC_FONTCONFIG_VERSION}" VERSION_GREATER "2.13.0" OR "${PC_FONTCONFIG_VERSION}" STREQUAL "2.13.0")
    AND ("${FREETYPE_VERSION_STRING}" VERSION_GREATER "2.9.0" OR "${FREETYPE_VERSION_STRING}" STREQUAL "2.9.0")
    AND ("${PC_HARFBUZZ_VERSION}" VERSION_GREATER "1.4.2" OR "${PC_HARFBUZZ_VERSION}" STREQUAL "1.4.2"))
    WEBKIT_OPTION_DEFAULT_PORT_VALUE(ENABLE_VARIATION_FONTS PRIVATE ON)
endif ()
```

marhoy, 2019-05-18:
@philn Thanks for the tip!
I just tried to recompile with cairo-1.16.0, fontconfig-2.13.1, freetype-2.9.1 and harfbuzz-1.8.8.
But unfortunately, it didn't solve the problem.

bobby-beckmann, 2019-05-22:
I'm going to jump in on this issue because I think it's related. I'm running 2.24.1 and it seems all google properties are failing to load. I get: Error writing data to TLS socket: The specified session has been invalidated for some reason. for all google/youtube, etc. sites (including fonts). 

I'm going to try 2.24.2 to see if it's any better.

zdobersek, 2019-05-22:
> I'm going to jump in on this issue because I think it's related. I'm running 2.24.1 and it seems all google properties are failing to load. I get: Error writing data to TLS socket: The specified [...]

FWIW I was seeing this same issue, and it was fixed when upgrading the following packages:
- gnutls 3.6.4 to 3.6.6
- glib-networking 2.54 to 2.58
- libsoup 2.62.? to 2.64.2

I don't know what the underlying issue is, but this set of updates alone fixed it.

bobby-beckmann, 2019-05-22:
I accomplished the same by just downgrading gnutls to 3.6.1 from 3.6.4 (from sumo). Good to know the above probably works as well. Must be a mismatch b/w gnutls and glib-networking.

trvbueno, 2019-05-24:
I can confirm TLS is broken on thud, unless GnuTLS is downgraded from 3.6.4 to 3.6.1.

marhoy, 2019-05-25:
> FWIW I was seeing this same issue, and it was fixed when upgrading the following packages:
>
> * gnutls 3.6.4 to 3.6.6
> * glib-networking 2.54 to 2.58
> * libsoup 2.62.? to 2.64.2
>
> I don't know what the [...]

@zdobersek: I just tried to recompile using poky warrior with these versions, which should be more than sufficient:
- gnutls-3.6.7
- glib-networking-2.60.1
- libsoup-2.64.2
- wpewebkit-2.24.2

But it still doesn't work, and I get the following error which indicate there's problems with certificates:

```GLib-Net-WARNING **: 19:25:15.807: Failed to load TLS database: Failed to load system trust store: GnuTLS was not configured with a system trust```

**How do I configure GnuTLS with "system trust"?**

@trvbueno: I also tried downgrading gnutls to version 3.6.1, but that didn't solve it either.

marhoy, 2019-05-26:
Nailed it! I just compiled wpewebkit-2.24.2 with poky warrior and gnutls-3.6.7!

In order to tell GnuTLS to use "system trust", I added `--with-default-trust-store-dir=/etc/ssl/certs` to `gnutls_3.6.7.bb`.

Compiled docker image can be downloaded here: `marhoy/balena-wpe:2.24-latest`

philn, 2019-05-27:
So, is this still an issue?

marhoy, 2019-05-28:
@philn No, issue resolved :)

And it turned out to be a GnuTLS issue, not a wpewebkit issue.

closed 2019-05-28 by marhoy