Igalia/meta-webkit #106: Building issue in yocto warrior

woutervanh, 2019-05-23:
I'm trying to build in yocto warrior, but it's stopping with following error:

`In file included from DerivedSources/WebCore/unified-sources/UnifiedSource-3c72abbe-34.cpp:5:
/opt/yocto/fsl-community-bsp_warrior/build_m7/tmp/work/cortexa9t2hf-neon-imx-fslc-linux-gnueabi/wpewebkit/2.24.2-r0/wpewebkit-2.24.2/Source/WebCore/platform/graphics/opentype/OpenTypeVerticalData.cpp: In member function 'void WebCore::OpenTypeVerticalData::loadMetrics(const WebCore::FontPlatformData&)':
/opt/yocto/fsl-community-bsp_warrior/build_m7/tmp/work/cortexa9t2hf-neon-imx-fslc-linux-gnueabi/wpewebkit/2.24.2-r0/wpewebkit-2.24.2/Source/WebCore/platform/graphics/opentype/OpenTypeVerticalData.cpp:481:129: warning: cast from 'const WebCore::OpenType::VmtxTable::Entry*' to 'const Int16*' {aka 'const WebCore::OpenType::BigEndianShort*'} increases required alignment of target type [-Wcast-align]
         const OpenType::Int16* pTopSideBearingsExtra = reinterpret_cast<const OpenType::Int16*>(&vmtx->entries[countVmtxEntries]);
                                                                                                                                 ^
ninja: build stopped: subcommand failed.
`

woutervanh, 2019-05-23:
Ok, that was not the real error. I tend to have issues with cairo
`| In file included from DerivedSources/WebCore/unified-sources/UnifiedSource-3c72abbe-26.cpp:4:
| /opt/yocto/fsl-community-bsp_warrior/build_m7/tmp/work/cortexa9t2hf-neon-imx-fslc-linux-gnueabi/wpewebkit/2.24.2-r0/wpewebkit-2.24.2/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp: In member function 'virtual cairo_device_t* WebCore::GLContextEGL::cairoDevice()':
| /opt/yocto/fsl-community-bsp_warrior/build_m7/tmp/work/cortexa9t2hf-neon-imx-fslc-linux-gnueabi/wpewebkit/2.24.2-r0/wpewebkit-2.24.2/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp:483:21: error: 'cairo_egl_device_create' was not declared in this scope
|      m_cairoDevice = cairo_egl_device_create(m_display.eglDisplay(), m_context);
|                      ^~~~~~~~~~~~~~~~~~~~~~~
| /opt/yocto/fsl-community-bsp_warrior/build_m7/tmp/work/cortexa9t2hf-neon-imx-fslc-linux-gnueabi/wpewebkit/2.24.2-r0/wpewebkit-2.24.2/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp:483:21: note: suggested alternative: 'cairo_gl_surface_create'
|      m_cairoDevice = cairo_egl_device_create(m_display.eglDisplay(), m_context);
|                      ^~~~~~~~~~~~~~~~~~~~~~~
|                      cairo_gl_surface_create
`

philn, 2019-05-27:
As mentioned in wpewebkit.inc, did you add this to your local.conf? `PACKAGECONFIG_append_pn-cairo = " glesv2"`

philn, 2019-05-27:
2dcanvas support doesn't perform very well anyway (in my experience). Why do you enable it?

psaavedra, 2019-07-10:
Could your try again using on of this  `bbappend` depending on your `cairo` version?:

* https://gitlab.com/saavedra.pablo/meta-perf-browser/blob/nightly/recipes-graphics/cairo/cairo_1.14.%25.bbappend
* https://gitlab.com/saavedra.pablo/meta-perf-browser/blob/nightly/recipes-graphics/cairo/cairo_1.16.0.bbappend

closed 2021-03-11 by github-actions[bot]