Igalia/meta-webkit #119: RPI: Cog EGL issue

TomMettam, 2019-07-20:
When attempting to visit a page with a WebGL context on a raspberry pi, cog gives this error:

```
Cannot create EGL surfaceless context: missing EGL_KHR_surfaceless_{context,opengl} extension.
Cannot create EGL WPE context: EGL_SUCCESS
```
Here's my local.conf:
```
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "rpi"
IMAGE_INSTALL_append = " wpewebkit cog openssh-sftp-server nano userland"

PACKAGECONFIG_append_pn-cairo = " glesv2 egl"
PACKAGECONFIG_remove_pn-cairo = " opengl "
PACKAGECONFIG_append_pn-wpewebkit = " 2dcanvas"
PREFERRED_VERSION_virtual/libwpe = "1.2.0"

LICENSE_FLAGS_WHITELIST = "commercial"

INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "useradd -P raspberry root;"
EXTRA_IMAGE_FEATURES += "ssh-server-openssh"

MACHINE_FEATURES_append = " userland"
MACHINE_FEATURES_remove = " vc4graphics"

```

philn, 2019-07-22:
Please try with the FDO backend instead of rdk.

TomMettam, 2019-07-22:
I really don't want to use the fdo backend. Is it now a requirement for wpe?

philn, 2019-07-22:
Why don't you want to use it?
It's not a requirement but trying a different backend might help narrow down the issue.

TomMettam, 2019-07-22:
Because I don't want the overhead, really. Also I understand that the rpi framebuffer backend performs significantly better..

philn, 2019-07-22:
Can you clarify what you mean with "overhead"? Also please confirm which wpewebkit version you're using.

TomMettam, 2019-07-22:
The overhead being the memory and disk footprint of the freedesktop.org server and the latency incurred by the display server versus rending directly to the rpi framebuffer.

Honestly not sure which version of wpewebkit it compiled (i've since deleted the config and tried using meta-wpe which also doesn't work currently). It would have been whatever version was default in this repo 2 days ago.

zdobersek, 2019-07-23:
Despite the error, is the WebGL content displayed properly? Does http://get.webgl.org/ work? Is there a specific WebGL example that doesn't?

The reported error is otherwise harmless.

shorvat, 2019-08-07:
@TomMettam did you manage to solve the issue?

TomMettam, 2019-08-20:
I didn't, i'm afraid. No, no WebGL examples work with it. I've moved on though to another platform, so I won't update here.

psaavedra, 2020-04-14:
The description is incompleted. Closing this issue as `invalid`. @TomMettam please, feel you free to reopen again this issue providing more information about how to reproduce (test cse, WebKit version and cog version). Thanks.

closed 2020-04-14 by psaavedra