Igalia/meta-webkit #120: wpebackend-rdk and PowerVR GPU

andy-sdc, 2019-07-23:
Has anyone attempted to create a wpebackend-rdk integration with the OpenGL interface provided by the PowerVR GPU in the TI AM335x devices - as used on the Beaglebone Black?

I'm trying to find an option that starts up quickly and doesn't require the use of the Wayland compositor as it has quite a large footprint and takes an age to start.

-Andy.

clopez, 2019-07-24:
> Has anyone attempted to create a wpebackend-rdk integration with the OpenGL interface provided by the PowerVR GPU in the TI AM335x devices - as used on the Beaglebone Black?

Speaking about GPUs for embedded platforms, I have experience working with Vivante GPUs (i.MX6/8) and RPi GPUs (Broadcom).
Regarding PowerVR GPUs: I didn't tried WPE on it. Not sure if anyone else did. I'm not sure what you are going to find.

But its highly recommended that the GPU drivers of your platform provide support for the EGL Wayland extension, specifically support for ```EGL_WL_bind_wayland_display```

If that is not supported by your GPU drivers then you won't be able to use wpebackend-fdo and you will have to use/implement a custom backend for your platform. wpebackend-rdk implements custom backends for several platforms, not sure if anyone of those is going to work for you. 

> I'm trying to find an option that starts up quickly and doesn't require the use of the Wayland compositor as it has quite a large footprint and takes an age to start.

Weston should start in a matter of a few seconds, if it takes more then that looks like an issue with your platform or GPU drivers. Weston has several backends, double-check you are using the right one for your platform (example: fbdev vs drm)

TIP: You can include the utility waylandeglinfo in your image (provided by this layer) and run it inside weston to check what EGL extensions your driver defines.

andy-sdc, 2019-07-24:
> But its highly recommended that the GPU drivers of your platform provide support for the EGL Wayland extension, specifically support for `EGL_WL_bind_wayland_display`

Looking at the header files in [git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git](git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git) which appears to be where the meta-ti layer gets the GPU drivers from then it is a supported function.

> Weston should start in a matter of a few seconds, if it takes more then that looks like an issue with your platform or GPU drivers. Weston has several backends, double-check you are using the right one for your platform (example: fbdev vs drm)

The problem is that the system overall has to boot in a "matter of a few seconds".  With the TI AM335x devices I can use Falcon mode to bypass most of U-Boot and in a past project stripped the kernel down to less than 2MB which gave a very fast boot time.

What I have seen so far in testing (and can't have) is having to wait any significant period of time for Weston to start or have the grey background displayed whilst waiting for Cog to start.

I'm just waiting for a Yocto build to complete and I'll see how I get on!

-Andy.

philn, 2019-07-24:
As an alternative to Weston perhaps you can consider the wlroots Cage compositor.

andy-sdc, 2019-08-14:
> Speaking about GPUs for embedded platforms, I have experience working with Vivante GPUs (i.MX6/8) and RPi GPUs (Broadcom).

I've not used the i.MX6 devices before.  Is the Vivante GPU in all of the [i.MX6 variants](https://www.nxp.com/products/processors-and-microcontrollers/arm-processors/i.mx-applications-processors/i.mx-6-processors:IMX6X_SERIES) or only those towards the bottom of the list that state 3D support?

-Andy.

MagnaboscoL, 2020-06-27:
Hi @sdcsystems,

did you have any success using WPE with the PowerVR GPU without a Wayland compositor?

Luca

closed 2021-07-06 by github-actions[bot]