Igalia/meta-webkit #121: libWPEBackend version issue when building with Mesa backend - sumo

concreted, 2019-08-21:
Hi, I am trying out building WPE Webkit for the Intel Compute Stick and ran into a build error when running `bitbake core-image-weston`:
```
Error: 
 Problem: package libwpebackend-mesa0-git-r0.i586 requires libWPEBackend-0.1.so.0, but none of the providers can be installed
  - package wpewebkit-2.22.5-r0.i586 conflicts with wpebackend < 1.0 provided by libwpebackend-0.1-0-0.2.0-r0.i586
  - package cog-0.2.0-r0.i586 requires virtual/wpebackend, but none of the providers can be installed
  - conflicting requests
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages)
```
Looks like the Mesa backend requires an older version of `libWPEBackend` than `wpewebkit`? Is there any way to work around this or is there something I'm missing? Not sure where to inject an `--allowerasing` command since it seems to be coming from the `core-image-weston` steps, I'm new to Yocto. I'm using the sumo versions of all the layers.

I am using these layers in `bblayers.conf`:
```
BBLAYERS ?= " \
  /home/build/poky/meta \
  /home/build/poky/meta-poky \
  /home/build/poky/meta-yocto-bsp \
  /home/build/meta-openembedded/meta-oe \
  /home/build/meta-openembedded/meta-python \
  /home/build/meta-gstreamer1.0 \
  /home/build/meta-webkit \
  "
```

and have these values set in `local.conf`:
```
IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-mesa"
```

My repo with full details is https://github.com/mirainc/intel-wpe. The submodules I'm using are all pointing to the `sumo` branches: https://github.com/mirainc/intel-wpe#submodules-used. I can try with another version if that's the expected fix.

philn, 2019-08-21:
Do you really require the Mesa backend? I'm not sure in which state it is currently. I think most users rely on the FDO backend or RDK.

clopez, 2019-08-21:
> Do you really require the Mesa backend? I'm not sure in which state it is currently. I think most users rely on the FDO backend or RDK.

Indeed. Even if your have Mesa based drivers it is better to use the FDO backend nowadays.

concreted, 2019-08-21:
Ah, I was under the impression that Mesa was required for better graphics driver support on Intel devices. Do you know of users with the Intel NUC or similar devices that are getting good performance with FDO?

concreted, 2019-08-21:
Would you recommend adding the `meta-intel` (http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel/about/?h=sumo) layer to my build and switching to FDO backend?

clopez, 2019-08-21:
You can try the FDO backend and if you are not satisfied with the performance then you can try with other backends.

Seems the mesa one needs a bit of love in the recipes to build (as it is failing to build for you), you can send a patch here if you manage to fix the build.

A third backend you might want to try is the wayland one inside wpebackend-rdk.

But if FDO performs as expected, then I recommend to use it as it is the backend that is receiving more attention from upstream developers.

concreted, 2019-08-21:
@clopez Thanks for the additional info. I am trying a build with `meta-intel` and will see how that goes. 

I tested on my Compute Stick with non-Yocto compiled WPE Webkit, and weston appears to be using the correct Intel graphics drivers according to the start logs (see https://gist.github.com/concreted/498ef33faabc76c27fa102d6e569a3c7). A single Cog window with one of the examples here runs around 60fps, but opening more starts to tank the frame rate - with 6 windows open i'm getting about 10FPS on each. Is there a good way to confirm whether WPE is using the correct graphics drivers?

I saw you had a demo from 2015 (http://blog.neutrino.es/2015/meta-webkit-yocto-layer/) where you ran WPE Webkit on an Intel NUC which is pretty similar to the Compute Stick. Is the performance I'm seeing in line with what you'd expect for this type of device? The performance looks pretty good in your video.

clopez, 2019-08-22:
> I saw you had a demo from 2015 (http://blog.neutrino.es/2015/meta-webkit-yocto-layer/) where you ran WPE Webkit on an Intel NUC which is pretty similar to the Compute Stick. Is the performance I'm [...]

Lots of things have changed in this 4 years. On that post there is also a link to the image i used for recording that video, not sure if it will run on your hardware. If you try to boot it you will find inside it several commands that start with w4w-* and launch each ones of the demos
```
$ find usr/|grep w4w
usr/bin/w4w-webgl-materials2
usr/bin/w4w-postercircle
usr/bin/w4w-webgl-materiasl-cubemap-dynamic2
usr/bin/w4w-webgl-shader-lava
usr/bin/w4w-webgl-linescolors
usr/bin/w4w-webgl-shader2
```
Those are simply shells script that open the browser to show the demo that is shipped locally on /var/www.
I have extracted that demos from there and uploaded them here in case you want to test with the very same demos to see if the performance is comparable or not.
Check http://ftp.neutrino.es/w4w-demos/

concreted, 2019-08-29:
@clopez Update from my end, I was able to get WPEBackend-mesa to build against latest WPE by doing a very naive patch (see https://github.com/WebPlatformForEmbedded/WPEBackend-mesa/compare/master...mirainc:wpe1.0?expand=1) and it produced `libWPEBackend-mesa.so`. After I did `ln -s libWPEBackend-mesa.so /usr/lib/libWPEBackend-default.so`, running `cog <url>` crashes over and over on launch: 

```
(cog:1489): Cog-WARNING **: 01:11:16.259: <http://youtube.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 01:11:16.491: <http://youtube.com/> Load started.
Cog-Message: 01:11:16.493: <http://youtube.com/> Loading...
Cog-Message: 01:11:16.505: <http://youtube.com/> Loaded successfully.
```

Is there a way to get more detailed debug info? I also tried doing `cog -P mesa` but it warns `Could not load: libcogplatform-mesa.so (possible cause: No such file or directory).` 

I do want to try Mesa because the graphics performance on Intel Compute Stick with FDO doesn't seem to be using the GPU properly - running a site with basic CSS animations spikes the GPU usage (using `intel_gpu_top`) to 80-90%, whereas GPU utilization is much lower with Electron/Chromium and performance is similar with a slight edge to Electron/Chromium.

philn, 2019-08-29:
It's probably related with a badly handled `nullptr` now being returned in the mesa.cpp that you updated. Why did you remove keyboard handling anyway?

If something is broken in -fdo, it needs to be investigated. Switching to another backend is probably a good workaround for you but not for the community at large :)

concreted, 2019-08-29:
Ah that makes sense, will see if I can fix it up.

I took out keyboard handling because `wpe_input_key_mapper_interface` appears to not exist anymore in WPE so it was throwing a compile error. Also I don't really need keyboard support since I'm using WPE in a signage use case.

mykhani, 2019-11-27:
@concreted Hello, I have run into the same issue but I am using a rockchip platform. Atm, I am trying to use your patches. Did you manage to get the desired results with wpebackend-mesa? I would greatly appreciate if you could share your experience with wpebackend-mesa. Thank you.

closed 2021-02-05 by psaavedra