Igalia/meta-webkit #124: i.MX6 video acceleration not working within WPE

dumanovskis, 2019-09-03:
I'm trying to get hardware-acceleration of videos on an i.MX6 based system. When I load a web page containing a video, it clearly plays without HW acceleration. If I invoke gstreamer directly to play videos, everything is fine.

The Linux image is based on Yocto thud, with Vivante drivers. I use wpebackend-rdk. 

Output of `cog --version` is `0.3.2+git (WPE WebKit 2.24.2)`. gstreamer is `1.14.4`.

My `local.conf` has:

```
IMAGE_INSTALL_append = " wpewebkit cog gstreamer1.0-plugins-imx  gstreamer1.0-plugins-imx-meta "
```

GStreamer plugins for using the GPU seem to be present:

```
root@ax-mx6:/usr/lib# gst-inspect-1.0 | grep imx
imxv4l2video:  imxv4l2videosrc: V4L2 CSI Video Source
imxv4l2video:  imxv4l2videosink: V4L2 CSI Video Sink
imxeglvivsink:  imxeglvivsink: Freescale EGL video sink
imxpxp:  imxpxpvideosink: Freescale PxP video sink
imxpxp:  imxpxpvideotransform: Freescale PxP video transform
imxaudio:  imxuniaudiodec: Freescale i.MX uniaudio decoder
imxaudio:  imxmp3audioenc: Freescale i.MX MP3 encoder
imxg2d:  imxg2dvideosink: Freescale G2D video sink
imxg2d:  imxg2dvideotransform: Freescale G2D video transform
imxg2d:  imxg2dcompositor: Freescale G2D video compositor
imxg2d:  imxg2dtextoverlay: Text overlay
imxg2d:  imxg2dtimeoverlay: Time overlay
imxg2d:  imxg2dclockoverlay: Clock overlay
imxg2d:  imxg2dtextrender: Text renderer
imxvpu:  imxvpudec: Freescale VPU video decoder
imxvpu:  imxvpuenc_h263: Freescale VPU h.263 video encoder
imxvpu:  imxvpuenc_h264: Freescale VPU h.264 video encoder
imxvpu:  imxvpuenc_mpeg4: Freescale VPU MPEG-4 video encoder
imxvpu:  imxvpuenc_mjpeg: Freescale VPU motion JPEG video encoder
imxipu:  imxipuvideotransform: Freescale IPU video transform
imxipu:  imxipuvideosink: Freescale IPU video sink
imxipu:  imxipucompositor: Freescale IPU video compositor
```

If I play a video with `gst-launch-1.0 playbin uri=file:/root/test.mp4` then the video is clearly played with acceleration.

I tried seeing whether the browser loads the gstreamer plugins at all when playing video, and I think it does, as follows

```
root@ax-mx6:~# lsof -p $(pidof WPEWebProcess) | grep gstream
WPEWebPro 1830 root  mem       REG     179,34    83408  11756 /usr/lib/gstreamer-1.0/libgstdeinterlace.so
WPEWebPro 1830 root  mem       REG     179,34    25912  11739 /usr/lib/gstreamer-1.0/libgstvideoscale.so
WPEWebPro 1830 root  mem       REG     179,34    17772  11744 /usr/lib/gstreamer-1.0/libgstvideoconvert.so
WPEWebPro 1830 root  mem       REG     179,34    38560  11689 /usr/lib/gstreamer-1.0/libgstvideofilter.so
WPEWebPro 1830 root  mem       REG     179,34    63196  11633 /usr/lib/gstreamer-1.0/libgstimxvpu.so
WPEWebPro 1830 root  mem       REG     179,34    96152  11772 /usr/lib/gstreamer-1.0/libgstaudioparsers.so
WPEWebPro 1830 root  mem       REG     179,34   154104  11803 /usr/lib/gstreamer-1.0/libgstvideoparsersbad.so
WPEWebPro 1830 root  mem       REG     179,34   375056  11760 /usr/lib/gstreamer-1.0/libgstisomp4.so
WPEWebPro 1830 root  mem       REG     179,34    77212  11640 /usr/lib/gstreamer-1.0/libgsttypefindfunctions.so
WPEWebPro 1830 root  mem       REG     179,34   408260  11785 /usr/lib/gstreamer-1.0/libgstplayback.so
WPEWebPro 1830 root  mem       REG     179,34   281196  11679 /usr/lib/gstreamer-1.0/libgstcoreelements.so
WPEWebPro 1830 root  mem       REG     179,34    83356  11767 /usr/lib/gstreamer-1.0/libgstpulseaudio.so
WPEWebPro 1830 root  mem       REG     179,34    95716  11706 /usr/lib/gstreamer-1.0/libgstaudiofx.so
WPEWebPro 1830 root  mem       REG     179,34    21840  11738 /usr/lib/gstreamer-1.0/libgstvolume.so
WPEWebPro 1830 root  mem       REG     179,34    26016  11704 /usr/lib/gstreamer-1.0/libgstaudioresample.so
WPEWebPro 1830 root  mem       REG     179,34    21868  11795 /usr/lib/gstreamer-1.0/libgstaudioconvert.so
WPEWebPro 1830 root  mem       REG     179,34    17888  11669 /usr/lib/gstreamer-1.0/libgstautodetect.so
WPEWebPro 1830 root  mem       REG     179,34     5404  11766 /usr/lib/gstreamer-1.0/libgstapp.so
WPEWebPro 1830 root  mem       REG     179,34   153652  11649 /usr/lib/gstreamer-1.0/libgstopengl.so
WPEWebPro 1830 root  mem       REG     179,34   867740  12898 /usr/lib/libgstreamer-1.0.so.0.1404.0
```

In particular I see `libgstimxvpu.so` there, which is the library for accelerated iMX6 GPU if I understand correctly.

I'm not sure how to investigate the issue further, and if cog/wpe have useful debug output somewhere.

philn, 2019-09-03:
Out of curiosity, which WPEBackend are you using?

dumanovskis, 2019-09-03:
wpebackend-rdk/wayland. I haven't been able to get fdo working and would prefer not to use it in either case.

I made another attempt now, using the Freescale fsl-wayland as a base, instead of fslc-wayland as before. The difference seems to boil down to a bunch of `PREFERRED_VERSION_gstreamer1.0 = "1.14.imx"` That seems to result in a different version of gstreamer plugins, and gstreamer is `1.14.0` now if I run `gstreamer --version`. 

Still the same results, although I'm getting more output. When I play the video with `gst-play`, there's a bunch of output like this:

```
Prerolling...
====== AIUR: 4.4.2 build on Sep  3 2019 15:09:40. ======
        Core: MPEG4PARSER_06.14.02  build on Mar 27 2018 02:55:00
 file: /usr/lib/imx-mm/parser/lib_mp4_parser_arm11_elinux.so.3.2
------------------------
    Track 00 [video_0] Enabled
        Duration: 0:00:10.000000000
        Language: und
    Mime:
        video/x-h264, parsed=(boolean)true, alignment=(string)au, stream-format=(string)avc, width=(int)960, height=(int)540, framerate=(fraction)25/1, codec_data=(buffer)014d4028ffe1001c674d4028eca07808bf780b501010140000030004000afc803c60c65801000568eae132c8 
------------------------
------------------------
    Track 01 [audio_0] Enabled
        Duration: 0:00:10.004000000
        Language: eng
    Mime:
        audio/mpeg, mpegversion=(int)4, channels=(int)2, rate=(int)48000, bitrate=(int)160359, stream-format=(string)raw, codec_data=(buffer)119056e500 
------------------------

====== BEEP: 4.4.2 build on Sep  3 2019 15:09:46. ======
        Core: AAC decoder Wrapper  build on Dec  7 2017 18:13:49
 file: /usr/lib/imx-mm/audio-codec/wrap/lib_aacd_wrap_arm12_elinux.so.3
CODEC: BLN_MAD-MMCODECS_AACD_ARM_03.09.00_CORTEX-A8  build on Sep 20 2017 15:02:14.
display(/dev/fb0) resolution is (1920x1080).
====== OVERLAYSINK: 4.4.2 build on Sep  3 2019 15:09:59. ======
display(/dev/fb0) resolution is (1920x1080).
[INFO]  bitstreamMode 1, chromaInterleave 0, mapType 0, tiled2LinearEnable 0
[WARN]  VPU iram is less than needed, some parts don't use iram
display(/dev/fb0) resolution is (1920x1080).
0:00:00.0 / 0:00:10.0       
===!!! Current pulsesink device is auto_null !!!===
```

When I run `cog` to view a webpage with the same video, the output is exactly the same, except these 3 lines are missing:

```
display(/dev/fb0) resolution is (1920x1080).
====== OVERLAYSINK: 4.4.2 build on Sep  3 2019 15:09:59. ======
display(/dev/fb0) resolution is (1920x1080).
```

That makes some sense because gstreamer should be rendering to the browser's canvas and not to `/dev/fb0`, as I understand it. So I'm puzzled as to why the in-browser playback is slow when it seems to be doing the same thing.

philn, 2019-09-03:
Have you considered trying the etnaviv driver?

philn, 2019-09-03:
Anyway, my colleague @zdobersek had some success with hw-accelerated playback on i.MX6 with gst-imx. Maybe he can comment here :)

dumanovskis, 2019-09-04:
Using the etnaviv driver instead might be too much trouble at this point, but I've had some interesting progress. I played the video with `gst-launch` using every video sink that `gst-inspect` reports, and then started getting rid of sinks individually.

I found that removing the OpenGL sink (`libgstopengl.so`) dramatically improves performance in WPE. I do now get a complaint like the following:

```
** (WPEWebProcess:1077): WARNING **: 13:44:00.612: WebKit wasn't able to find the GStreamer opengl plugin. Hardware-accelerated zero-copy video rendering can't be enabled without this plugin.
```

That changes the code path in `MediaPlayerPrivateGStreamerBase::createVideoSinkGL`. Video playback is much better after this. Previously I had some color distortion and probably around 1FPS, now the video has no distortion and plays at what looks like 11-12 FPS. This is of course very surprising, as I understand the OpenGL functionality should provide a considerable speedup.

philn, 2019-09-04:
It would be nice to figure out the issue in libgstopengl :) How's the CPU usage with the fallback video sink?

philn, 2019-09-04:
Does this work? `gst-launch-1.0 playbin uri=file:///root/test.mp4 video-sink=glimagesink`

dumanovskis, 2019-09-04:
With opengl removed, CPU usage of the WPE process is very high (as before), despite the nearly-acceptable performance. So that seems to suggest the GPU is still not or underutilized.

Trying to run with `glimagesink` as you suggest doesn't work - there's no video output, and it prints the following, which doesn't look great:

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayEGL\)\ gldisplayegl0";
Missing element: MPEG-4 AAC decoder
WARNING: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0: No decoder available for type 'audio/mpeg, mpegversion=(int)4, framed=(boolean)true, stream-format=(string)raw, level=(string)2, base-profile=(string)lc, profile=(string)lc, codec_data=(buffer)1190, rate=(int)48000, channels=(int)2'.
Additional debug info:
../../../gst-plugins-base-1.14.4/gst/playback/gsturidecodebin.c(921): unknown_type_cb (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0
[INFO]  bitstreamMode 1, chromaInterleave 0, mapType 0, tiled2LinearEnable 0
[WARN]  VPU iram is less than needed, some parts don't use iram
Got context from element 'playsink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayEGL\)\ gldisplayegl0";
ERROR: from element /GstPlayBin:playbin0/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink: Failed to convert multiview video buffer
Additional debug info:
../../../gst-plugins-base-1.14.4/ext/gl/gstglimagesink.c(1740): gst_glimage_sink_prepare (): /GstPlayBin:playbin0/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
```

philn, 2019-09-04:
Can you try again after removing all the freescale gst patches and keeping gst-imx unmodified as well?
These downstream modifications usually are not of great quality, I'm afraid.

dumanovskis, 2019-09-04:
I'll give it a try, I'm not really sure what the entire recipe is like now to be honest. Do you essentially mean that my gst should be from Yocto-Poky, and not any meta-freescale layer?

zdobersek, 2019-09-11:
What kind of iMX6 hardware are you using?

What is the video resolution of the video you're trying to play? HD, FullHD?

One issue we experienced on this hardware was with decoding of high-quality videos taking up a lot of GPU memory, to the point that additional memory was being allocated from other sources. Moving around this extra memory around is very expensive and ended up being a big bottleneck on performance.

This can be avoided by specifying a bigger GPU memory area on the kernel command line.
Something like this would allocate 256MB of GPU memory, but also requires resizing CMA and vmalloc areas accordingly, assuming the board has enough memory to afford such big allocations:
`galcore.contiguousSize=268435456 cma=512M vmalloc=768M`

You can see the current state of GPU memory allocations by dumping the `/sys/kernel/debug/gc/meminfo` file. In short, once memory is being allocated outside of the gcvPOOL_SYSTEM pool, you have a problem.

Another performance problem with video playback inside WPE is the YUV-to-RGB conversion that has to be done. This alone requires a per-frame rendering pass that does the conversion, and the cost of that is proportional to the video size. This is not yet fixed in 2.24 or 2.26 versions, but just recently we were able to land initial work that removes the need for this extra conversion for most (but not all) YUV formats. I420 is the YUV format used by default on iMX, and as long as decoder remains outputting frames in that format, the conversion will not be required, and the extra cost is avoided.

dumanovskis, 2019-09-11:
My test video is a typical 720p H264-encoded video, and I'm on a Variscite system with dual Cortex-A9 and a Vivante GC2000, so it should not be having performance issues, although I'll verify your suggestions.

It also shouldn't be the YUV-to-RGB - I edited the WPE source code to remove `glcolorconvert` while investigating, which didn't help, leading me to conclude that `glupload` specifically is the pipeline element that creates the dramatic slowdown I see.

clopez, 2019-09-25:
Last time I worked with WPE on i.MX6 (around 2 years ago) video acceleration was working fine.
Maybe this is a reggresion? If you have the time and you are willing to, then it maybe helpful if you can try with older versions of wpewebkit to check if any of them works for video acceleration on i.MX6.

To build older versions of wpewebkit set on local.conf:

```
PREFERRED_VERSION_wpewebkit = '$version'
# maybe is also needed to lower cog version if fails to build with old wpewebkit
# PREFERRED_VERSION_cog = ''$version'
# and maybe is also needed to lower libwpe and wpebackend versions
```
To find the available versions look inside metawebkit:
```
$ find|grep -P '(cog|wpewebkit).*bb$'
./recipes-browser/cog/cog_0.4.0.bb
./recipes-browser/cog/cog_0.1.1.bb
./recipes-browser/cog/cog_git.bb
./recipes-browser/cog/cog_0.3.0.bb
./recipes-browser/cog/cog_0.2.0.bb
./recipes-browser/wpewebkit/wpewebkit_2.20.2.bb
./recipes-browser/wpewebkit/wpewebkit_2.22.4.bb
./recipes-browser/wpewebkit/wpewebkit_trunk.bb
./recipes-browser/wpewebkit/wpewebkit_2.24.3.bb
./recipes-browser/wpewebkit/wpewebkit_downstream.bb
./recipes-browser/wpewebkit/wpewebkit_2.26.0.bb
```

Hope this helps

closed 2021-04-12 by github-actions[bot]

vnd, 2022-08-08:
FYI: switching to open-source stack (IMX_DEFAULT_BSP = "mainline") solved video playback issues for me, with NXP stack there was no hardware acceleration.