Igalia/meta-webkit #132: thud - qt-wpe-simple-browser

ZikiHero, 2019-10-08:
I can see that wpewebkit-qtwpe-qml-plugin is in recepies but I am getting this error:
```
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openstlinux_weston-linux-gnueabi"
MACHINE              = "stm32mp1"
DISTRO               = "openstlinux-weston"
DISTRO_VERSION       = "2.6-snapshot-20191008"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
DISTRO_CODENAME      = "thud"
ACCEPT_EULA_stm32mp1 = "1"
GCCVERSION           = "8.%"
PREFERRED_PROVIDER_virtual/kernel = "linux-stm32mp"
meta-oe              
meta-gnome           
meta-xfce            
meta-initramfs       
meta-multimedia      
meta-networking      
meta-webserver       
meta-filesystems     
meta-perl            
meta-python          = "HEAD:f1511d254632a34c1deb51f4bf8b8c21e7423f51"
meta-webkit          = "thud:6c60f62253e27a8d446989349a2371eb064e1da0"
meta-st-stm32mp      = "HEAD:dcc955d6992cc46ce07440caaa5a17942776bbe6"
meta-qt5             = "HEAD:fd1f0f47312e7ee80e5ca65b8b2992cb068b1656"
meta-st-openstlinux  = "HEAD:4e36cf0c7c7bdb24700a990f73afcda84cdcf0cf"
meta                 = "HEAD:f0394e80a37f1da47042a1aa0487594f390603f9"

Initialising tasks: 100% |###############################################################################################| Time: 0:00:10
Sstate summary: Wanted 459 Found 446 Missed 13 Current 3160 (97% match, 99% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: wpewebkit-2.24.2-r0 do_configure: QA Issue: wpewebkit: invalid PACKAGECONFIG: wpewebkit-qtwpe-qml-plugin [invalid-packageconfig]
ERROR: st-image-weston-1.0-r0 do_rootfs: Unable to install packages. Command '/opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-openstlinux_weston-linux-gnueabi/st-image-weston/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated apt dpkg fb-test libgtk-3.0 gtk+3-demo kmscube libdrm2 libdrm-tests libv4l libx11-6 libx11-locale libxcb1 libxcursor1 packagegroup-base-extended packagegroup-core-boot packagegroup-core-eclipse-debug packagegroup-core-ssh-dropbear packagegroup-core-tools-profile packagegroup-framework-tools-audio packagegroup-framework-tools-audio-base packagegroup-framework-tools-core packagegroup-framework-tools-core-base packagegroup-framework-tools-kernel packagegroup-framework-tools-kernel-base packagegroup-framework-tools-network packagegroup-framework-tools-network-base packagegroup-framework-tools-python2 packagegroup-framework-tools-python2-base packagegroup-framework-tools-python3 packagegroup-framework-tools-python3-base packagegroup-framework-tools-ui packagegroup-framework-tools-ui-base packagegroup-gstreamer1-0 packagegroup-optee-core packagegroup-optee-test psplash-drm qt-wpe-simple-browser rc-keymaps resize-helper run-postinsts systemd-mount-partitions systemd-networkd-configuration libtiff5 weston weston-examples weston-init weston-xwayland wpewebkit xclock xeyes xf86-input-evdev xf86-input-keyboard xf86-input-mouse xf86-video-modesetting xinput xkbcomp xorg-minimal-fonts xserver-xorg-xwayland xterm' returned 100:
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qt-wpe-simple-browser : Depends: wpewebkit-qtwpe-qml-plugin but it is not installable
E: Unable to correct problems, you have held broken packages.

ERROR: st-image-weston-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/stm32mp1-openstlinux_weston-linux-gnueabi/st-image-weston/1.0-r0/temp/log.do_rootfs.4549
ERROR: Task (/opt/stlinux-weston/layers/meta-st/meta-st-openstlinux/recipes-st/images/st-image-weston.bb:do_rootfs) failed with exit code '1'

```

philn, 2019-10-08:
Here's what I have:
```
IMAGE_INSTALL_append = " qtbase-plugins qtwayland-plugins"
IMAGE_INSTALL_append = " wpewebkit-qtwpe-qml-plugin qt-wpe-simple-browser"
```
I haven't tried in thud recently though. The QA issue gives a good hint, your PACKAGECONFIG is incorrect. Here's the right one: `PACKAGECONFIG_append_pn-wpewebkit = " qtwpe"`

ZikiHero, 2019-10-08:
You are a lifesaver. I had to remove `qtwayland-plugins` but I was able to finally build it. I am new to yocto/bitbake and was struggling to get some web browser to stm for a week. Thanks for your help.

philn, 2019-10-08:
Cool. Please let us know how WPE performs on your platform :) I'll close this specific issue though, now that it appears to be fixed on your side.

ZikiHero, 2019-10-08:
So result is: Probably because I removed `qtwayland-plugins` the `qt-wpe-simple-browser` is unable to run with this error:
```
qt.qpa_xcb xcb_shm_create_segment() failed for size 33177600
```
When I try to build it with `qtwayland-plugins` I get following:
```
ERROR: qtwayland-5.11.2+gitAUTOINC+297bcd005c-r0 do_configure: Error calling /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/recipe-sysroot-native/usr/bin/qmake -makefile -o Makefile        /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/git/qtwayland.pro  --   -no-feature-drm-egl-server -no-feature-libhybris-egl-server -no-feature-wayland-brcm -feature-wayland-client -feature-wayland-egl -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx
ERROR: qtwayland-5.11.2+gitAUTOINC+297bcd005c-r0 do_configure: Function failed: do_configure (log file is located at /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/temp/log.do_configure.4144)
ERROR: Logfile of failure stored in: /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/temp/log.do_configure.4144
Log data follows:
| DEBUG: Executing shell function qmake5_base_preconfigure
| DEBUG: Shell function qmake5_base_preconfigure finished
| DEBUG: Executing shell function do_configure
| NOTE: qmake prevar substitution: '   '
| NOTE: qmake configure substitution: '  -no-feature-drm-egl-server -no-feature-libhybris-egl-server -no-feature-wayland-brcm -feature-wayland-client -feature-wayland-egl -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx'
| 
| Running configuration tests...
| Done running configuration tests.
| 
| Configure summary:
| 
| Qt Wayland Drivers:
|   EGL .................................... yes
|   Raspberry Pi ........................... no
|   XComposite EGL ......................... yes
|   XComposite GLX ......................... no
|   DRM EGL ................................ no
|   libhybris EGL .......................... no
| Qt Wayland Client ........................ yes
| Qt Wayland Compositor .................... yes
| Qt Wayland Compositor Layer Plugins:
|   VSP2 hardware layer integration ........ no
| 
| ERROR: Feature 'xcomposite-glx' was enabled, but the pre-condition 'features.wayland-client && features.opengl && !features.opengles2 && features.xlib && libs.xcomposite && libs.glx' failed.
| 
| ERROR: Feature 'xcomposite-glx' was enabled, but the pre-condition 'features.wayland-server && features.opengl && !features.opengles2 && libs.xcomposite && libs.glx' failed.
| 
| Check config.log for details.
| ERROR: Error calling /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/recipe-sysroot-native/usr/bin/qmake -makefile -o Makefile        /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/git/qtwayland.pro  --   -no-feature-drm-egl-server -no-feature-libhybris-egl-server -no-feature-wayland-brcm -feature-wayland-client -feature-wayland-egl -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /opt/stlinux-weston/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_weston-linux-gnueabi/qtwayland/5.11.2+gitAUTOINC+297bcd005c-r0/temp/log.do_configure.4144)
ERROR: Task (/opt/stlinux-weston/layers/meta-qt5/recipes-qt/qt5/qtwayland_git.bb:do_configure) failed with exit code '1'
```
Since I just downloaded repo from STM from this [tutorial](https://visualgdb.com/tutorials/linux/stm32mp1/sdk/) and added single layer `meta-webkit`, I suppose that STM has their image somehow configured and I shouldn't set it differently. I don't even know how.

I found out that in some layer is mentioned epiphany or minibrowser, so I am giving it a try with webkitgtk but that one has also some flaws:
```
error: invalid static_cast from type 'GLNativeWindowType' {aka 'long long unsigned int'} to type 'EGLNativeWindowType' {aka 'void*'}
```
I hope that, that one is solved in this [issue](https://github.com/Igalia/meta-webkit/issues/1)

philn, 2019-10-08:
Can you enable ` DRM EGL` in QtWayland?
I haven't tried WPE with the XCB QPA at all. It should work in EGLFS and Wayland-EGL QPAs though.

ZikiHero, 2019-10-08:
Like I mentioned above I am a newbie in this so I have no idea how to enable it. It is in recipe of QtWayland or I have to do something with `PACKAGECONFIG` and local.conf

philn, 2019-10-08:
From the log you included above, it looks like  `xcomposite-glx` conflicts with the wayland features. So try something like this maybe? `PACKAGECONFIG_remove_pn-qtwayland = "xcomposite-glx"`

philn, 2019-10-08:
Is `x11`  in your `DISTRO_FEATURES`? Do you really need it? If not, try to remove it, might help. Or not.

ZikiHero, 2019-10-08:
My `DISTRO_FEATURES` has only wayland. There is no x11 mentioned in `local.conf`

ZikiHero, 2019-10-08:
So right now I am trying to build it with this setting:
```
DISTRO_FEATURES_remove = "x11"
PACKAGECONFIG_remove_pn-qtwayland = "xcomposite-glx"
PACKAGECONFIG_append_pn-wpewebkit = " qtwpe"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL_append = " qtbase-plugins qtwayland-plugins"
IMAGE_INSTALL_append = " wpewebkit-qtwpe-qml-plugin qt-wpe-simple-browser"
```

philn, 2019-10-08:
I don't follow. If X11 is disabled, why are you trying to use the XCB QPA?
I think you should consult with whoever maintains the STM-related meta layers you're using.

ZikiHero, 2019-10-09:
Build was successful. After adding `export QT_QPA_PLATFORM=wayland` I was able to run the application, application window pop up, but the browser is still not running.
```
root@stm32mp1:~# qt-wpe-simple-browser 
Using Wayland-EGL
Using the 'xdg-shell-v6' shell integration
veglBindWaylandDisplay (334): failed to authentify gbm_fd=0
```

ZikiHero, 2019-10-09:
Nothing with wayland
```
root@stm32mp1:~# ps -aux | grep weston 
root       502  0.0  0.3   3824  1628 tty2     Ss+  07:21   0:00 weston-launch -- --log=/var/log/weston.log
root       507  0.1  3.4 111208 15144 tty2     Sl+  07:21   0:06 /usr/bin/weston --log=/var/log/weston.log
root       532  0.0  1.5  16684  6984 tty2     S+   07:21   0:00 /usr/libexec/weston-keyboard
root       533  0.0  2.7  21848 12264 tty2     S+   07:21   0:01 /usr/libexec/weston-desktop-shell
root       985  0.4  2.2  19036 10008 ?        Ss   07:23   0:14 /usr/bin/weston-terminal
root      5299  0.0  0.0   1776   392 pts/1    S+   08:19   0:00 grep weston
```

philn, 2019-10-09:
Weston is a Wayland compositor. So the browser should launch, but i've never seen this `veglBindWaylandDisplay` error...

philn, 2019-10-09:
Looks like some STM-specific thing?

ZikiHero, 2019-10-09:
I was trying to search for it but I have nothing. I suppose the best action now would be contact the maintainer of STM layer which you suggested before, right?

philn, 2019-10-09:
Right.

ZikiHero, 2019-10-09:
Ok, thank you for all your help you provided. If I found any solution I will write it here.

boti9723, 2020-11-19:
Hi! Have you found any solution to this problem? I've been struggling with the same problem for a while. Please let me know, if you have found out anything. Thank you!

closed 2021-11-27 by github-actions[bot]