Igalia/meta-webkit #137: What are the differences between meta-webkit and https://github.com/WebPlatformForEmbedded/meta-wpe

emblord, 2020-01-16:
What are the differences between meta-webkit and https://github.com/WebPlatformForEmbedded/meta-wpe

It seems to me both are  Igalia maintained repos?

clopez, 2020-01-16:
> What are the differences between meta-webkit and https://github.com/WebPlatformForEmbedded/meta-wpe

meta-webkit uses the upstream version of WPE (from https://wpewebkit.org). This version of WPE its released each 6 months from the code on https://github.com/WebKit/webkit (that is just the git mirror of the https://webkit.org repository)
meta-webkit also includes recipes for the WebKitGTK port.

meta-wpe uses the WPE version from https://github.com/WebPlatformForEmbedded/WPEWebKit which is a fork of the upstream one, and it includes several adaptations for the RDK platform, as well as different fixes for its supported devices. Check https://wiki.rdkcentral.com/display/RDK/WPE for more info about WPE in the RDK platform.

> It seems to me both are Igalia maintained repos?

There are people from Igalia working on both repos, but with different needs.

My advice would be:

* If you want a version of WPE that it includes the last version of WebKit and its close to upstream, then use meta-webkit. 
* If you want to use the RDK platform or you have a device that its better supported by meta-wpe (like an STB), then use meta-wpe.

philn, 2020-01-16:
See also https://github.com/WebPlatformForEmbedded/meta-wpe/issues/276

fberndl, 2020-01-20:
Thanks, can be closed.

closed 2020-01-27 by philn