Igalia/meta-webkit #138: Which backend for imx7dsabresd ( NXP MCIMX7 Sabre Board) in fullscreen mode

fberndl, 2020-01-20:
Hi,
we want to evaluate performance of the webkit running on imx7dsabresd on the NXP Sabre board.  

- We want to launch an website in fullscreen mode using the hdmi port.
- This board does not support any hw accelleration. 

Which backend and configuration should we choose. I read the wiki but I still don't know. 

Thanks a lot

zdobersek, 2020-01-31:
Probably easiest approach towards this setup would be to use Mesa with swrast, and WPEBackend-fdo on top of it. This would enable CPU-based rasterization.

Support for this was added in WPEBackend-fdo just recently: https://github.com/Igalia/WPEBackend-fdo/pull/90

From there on, what's missing is integration of displaying software-rasterized buffers via the display system. Cog must gain support for handling such exports, and then also find the proper way to handle their display.

In your case, per your description, it might be possible to use the DRM backend, copying the exported CPU memory containing rendered Web content into writable gbm_bo objects, and then display those via DRM/KMS.

jameshilliard, 2020-02-01:
> Probably easiest approach towards this setup would be to use Mesa with swrast, and WPEBackend-fdo on top of it. This would enable CPU-based rasterization.

I have a similar setup to this but I've been trying to run wpewebkit on top of a weston compositor(for display management), I've built mesa with just gallium swrast support only as my hardware(gma500 based) does not support hardware acceleration either.

> From there on, what's missing is integration of displaying software-rasterized buffers via the display system. Cog must gain support for handling such exports, and then also find the proper way to [...]

I've been attempting to run cog on top of weston using the DRM backend and pixman software rendering, is that a setup that should currently work or is cog missing some features needed still(I'm basically running master branches for all the wpewebkit project related software)? A similar setup to this seems to work fine for different hardware I support that does have hardware rendering(I'm running weston without pixman and building mesa with the i915 driver there but otherwise it's basically the same as what I'm trying to do with the gma500 hardware).

otavio, 2020-03-31:
Has this made any progress?

closed 2020-03-31 by fberndl