Igalia/meta-webkit #143: Rendering glitch when rendering animations

otavio, 2020-02-19:
We are enabling support for a device which is using:

Board: Nitrogen6X
GPU Driver: Etnaviv
Linux: 5.4.20
LibDRM: 2.4.100
Mesa: 19.3.3
Yocto Project: Zeus

We are very satisfied with the current CPU usage and general performance, however, as can be seen in the video, the rendering has some glitches[1] and we are wondering how we can fix it.

1. https://drive.google.com/file/d/1c5EEeLIFRowRKBUs4n3ijQAAx9koqSsw/view

For reference, in our distro we have:

```
DEFAULT_DISTRO_FEATURES = " \
    egl \
    multiarch \
    opengl \
    pam \
    systemd \
    vulkan \
    wayland \
"

DISTRO_FEATURES = "\
    ${DISTRO_FEATURES_DEFAULT} \
    ${DEFAULT_DISTRO_FEATURES} \
"

DISTRO_FEATURES_remove = "\
    x11 \
"

# WPE Settings
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "libwpe"
PREFERRED_VERSION_cog = "0.4.%"
PREFERRED_VERSION_wpewebkit = "2.26.2"
PREFERRED_VERSION_wpebackend-fdo = "1.4.%"

PACKAGECONFIG_append_pn-cairo = " glesv2 egl"
PACKAGECONFIG_append_pn-wpewebkit = " 2dcanvas"
```

Any idea what we can look at to avoid this rendering issues?

clopez, 2020-02-19:
Some suggestions:

* The first thing I would try its to disable 2dcanvas to see if that helps.
* The second thing I would try is to try to reproduce that on an x86_64 with intel Mesa graphics to see if its an issue related with the graphic drivers (with an image built with Yocto, using the same layers, but for desktop)
* The third thing I would try is to try to reproduce that on an x86_64 with intel Mesa graphics and using a development version of WPE instead of one built with Yocto (on a standard distro like Debian). Check: https://trac.webkit.org/wiki/WPE#BuildingWPEWebKitfromgit

If you can reproduce with the third option, then it is likely an issue on WPE itself, its worth reporting it upstream at https://bugs.webkit.org with all the possible details about how it can be reproduced. Select as component there "WPE WebKit"

clopez, 2020-02-20:
BTW; I just remembered we recently fixed a glitch with CSS 3D transforms and it seems we forgot to backport the fix to the 2.26.x stable branch :(
Can you try this patch to see if it helps https://trac.webkit.org/r254680 ? please let me know if it helps.

otavio, 2020-02-20:
> BTW; I just remembered we recently fixed a glitch with CSS 3D transforms and it seems we forgot to backport the fix to the 2.26.x stable branch :(
> Can you try this patch to see if it helps https://trac.webkit.org/r254680 ? please let me know if it helps.

Using the 2.26.4 *with* the https://trac.webkit.org/r254680 patch fixes the issue. It'd be nice to apply it on the layer so people is not affected by it.

clopez, 2020-02-20:
> Using the 2.26.4 _with_ the https://trac.webkit.org/r254680 patch fixes the issue.
Nice! I have proposed it for merging in the next 2.26.5 point release at https://trac.webkit.org/wiki/WebKitGTK/2.26.x

> It'd be nice to apply it on the layer so people is not affected by it.
Yes, do you want to send a PR? ;)

otavio, 2020-02-21:
We prepare it tomorrow.

otavio, 2020-03-31:
This can be closed as #145 was merged.

closed 2020-03-31 by otavio