Igalia/meta-webkit #144: Fail to build on wpebackend-fdo

otavio, 2020-02-20:
```
wpebackend-fdo/1.4.1+gitAUTOINC+b01aa039d1-r0/git/src/extensions/video-plane-display-dmabuf.cpp:237:70: error: invalid use of incomplete type 'struct wpe_video_plane_display_dmabuf_source_create(wpe_renderer_backend_egl*)::wpe_renderer_backend_egl_base'
|   237 |     auto* impl = new Impl::DmaBuf(*static_cast<WS::BaseBackend*>(base->interface_data));
|       |                                                                      ^~
```

otavio, 2020-02-20:
This happens with the current devupstream revision.

clopez, 2020-02-20:
Can you please describe your local.conf config and how to reproduce the failure?

otavio, 2020-02-21:
very same settings of #143.

clopez, 2020-02-21:
> very same settings of #143.

This seems to be caused by the line
```
PREFERRED_VERSION_wpebackend-fdo = "1.4.%"
```

* With that line, bitbake its picking wpebackend-fdo from the git development snapshoot instead of the release tarball
  ```
  $ bitbake -s|grep fdo
  wpebackend-fdo                                      :1.4.1-r0 :1.4.1+gitAUTOINC+b01aa039d1-r0
  ```

* Without it (the default) it picks the expected release tarball.

  ```
  $ bitbake -s|grep fdo
  wpebackend-fdo                                      :1.4.1-r0     
  ```

I'm still not familiar how this devupstream thing that was added in f81f284 works, but this doesn't seem right at first.
@JoshuaWatt  is this a expected behaviour? That by simply setting ```PREFERRED_VERSION_wpebackend-fdo = "1.4.%"``` bitbake picks the devupstream version?

otavio, 2020-02-21:
Ok, so we should use `1.4.1`. Good.

clopez, 2020-02-21:
> Ok, so we should use `1.4.1`. Good.

In general you should not need to modify the default versions on local.conf, bitbake should pick the last stable one by default. That is what we use and test. Other versions are not so well tested and may have issues

philn, 2020-02-21:
If you use git master of wpebackend-fdo you also need libwpe git master

otavio, 2020-02-21:
@philn that was my assumption. I think both need to be bumped.

JoshuaWatt, 2020-02-21:
> I'm still not familiar how this devupstream thing that was added in [f81f284](https://github.com/Igalia/meta-webkit/commit/f81f28436f6111a284d3471cb5969dd2fd8ac4e1) works, but this doesn't seem [...]

Yes that is correct. It make a "virtual" recipe with that version.

clopez, 2020-02-21:
> Yes that is correct. It make a "virtual" recipe with that version.

But I would expect to require something more evident like ```PREFERRED_VERSION_wpebackend-fdo = "gitAUTOINC%"```

And not simply ```PREFERRED_VERSION_wpebackend-fdo = "1.4.%" ```

That looks like a footgun to me. 

Also I see the file https://github.com/Igalia/meta-webkit/blob/master/conf/include/webkit-bleeding.inc has not aged well at all, it still pointing to ```PREFERRED_VERSION_wpebackend-fdo = "1.4.0+gitAUTOINC+%"``` and ```PREFERRED_VERSION_wpewebkit = "2.26.2+gitAUTOINC+%"``` which are already gone versions (1.4.0 and 2.26.2).

I think we need to make the version used for selecting the devupstream version more obvious and not make it depend on a specific release version. For example, selecting devupstream only when version its ```gitmaster``` or something like that

otavio, 2020-02-21:
Yes, using git or something makes it more obvious.

JoshuaWatt, 2020-02-21:
> > Yes that is correct. It make a "virtual" recipe with that version.
> 
> But I would expect to require something more evident like `PREFERRED_VERSION_wpebackend-fdo = "gitAUTOINC%"`
> 
> And not simply `PREFERRED_VERSION_wpebackend-fdo = "1.4.%" `
> 
> That looks like a footgun to me.

Ya, I re-read the comments and I see what you mean. I'm not precisely sure why this is happening, but I would speculate that the `"1.4.%"` causes bitbake to effectively ignore the (implicit) `DEFAULT_PREFERENCE = "-1"` in the devupstream recipe, when means it maybe falls back to the normal version parsing rules where `1.4.1+git...` sorts as a higher base version than `1.4.1`?

> Also I see the file https://github.com/Igalia/meta-webkit/blob/master/conf/include/webkit-bleeding.inc has not aged well at all, it still pointing to `PREFERRED_VERSION_wpebackend-fdo = [...]

Ya, we can set the PV to whatever is appropriate in the devupstream overrides

closed 2021-04-12 by github-actions[bot]