Igalia/meta-webkit #160: wpewebkit can't build without wpebackend-fdo for rpi3

kraj, 2020-04-01:
I am trying to build wpewebkit with wpebackend-rdk being the only backend and its failing

I have 
```shell
DISTRO_FEATURES_remove = "x11 wayland"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
DISABLE_VC4GRAPHICS = "1"
```
```
| CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
| Please set them or make sure they are set and tested correctly in the CMake files:
| /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.28.0-r0/wpewebkit-2.28.0/Source/WebCore/WPEBACKEND_FDO_INCLUDE_DIRS
|    used as include directory in directory /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.28.0-r0/wpewebkit-2.28.0/Source/WebCore
| /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.28.0-r0/wpewebkit-2.28.0/Source/WebKit/WPEBACKEND_FDO_INCLUDE_DIRS
|    used as include directory in directory /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.28.0-r0/wpewebkit-2.28.0/Source/WebKit
| WPEBACKEND_FDO_LIBRARIES (ADVANCED)
|     linked by target "qtwpe" in directory /mnt/b/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/wpewebkit/2.28.0-r0/wpewebkit-2.28.0/Source/WebKit
|
| -- Configuring incomplete, errors occurred!
```

Is this unsupported configuration. ?

clopez, 2020-04-01:
The code suggest that you need wpebackend-fdo to enable the QtWPE:

```
# in WPEWebKit at Source/cmake/OptionsWPE.cmake
if (ENABLE_WPE_QT_API OR USE_WPE_VIDEO_PLANE_DISPLAY_DMABUF)
    find_package(WPEBackend_fdo 1.5.0 REQUIRED)
endif ()
```

Didn't CMake complained more loudly about this above?

kraj, 2020-04-01:
@clopez and I see that wpebackend-fdo requires wayland, am I right ? I was looking for build it with out wayland or X11, similar to like QT's eglfs mode.

clopez, 2020-04-01:
> @clopez and I see that wpebackend-fdo requires wayland, am I right ?

That's right. I can work without wayland (directly via KMS/DRM) but it uses the wayland protocol to communicate, so you need to have wayland as DISTRO_FEATURE

clopez, 2020-04-01:
> to communicate

I mean it uses the wayland protocol to share graphic buffers between the WebProcess and UIProcess

kraj, 2020-04-01:
I tried it and yes it builds but it does not run

Do I need specific settings during build for wpebackend-rdk ?

```
root@raspberrypi4:~# cog -P rdk http://google.com

** (cog:1875): WARNING **: 21:50:54.959: Could not load: libcogplatform-rdk.so (possible cause: Resource temporarily unavailable).

Wayland::Display: failed to connect

(cog:1875): Cog-WARNING **: 21:50:55.165: <http://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Wayland::Display: failed to connect

(cog:1875): Cog-WARNING **: 21:50:55.321: <http://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Wayland::Display: failed to connect

(cog:1875): Cog-WARNING **: 21:50:55.475: <http://google.com/> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Wayland::Display: failed to connect

````

clopez, 2020-04-01:
> I tried it and yes it builds but it does not run

What you tried exactly? can you paste here the complete local.conf?

kraj, 2020-04-01:
> > I tried it and yes it builds but it does not run
> 
> What you tried exactly? can you paste here the complete local.conf?

I use yoe distro so everything is not in local.conf 

but I have wayland opengl in DISTRO_FEATURES
``` PREFERRED_RPROVIDER_virtual/wpebackend ??= "wpebackend-fdo"
DISABLE_VC4GRAPHICS = "1"
IMAGE_INSTALL_append = "  wpewebkit \
  cog \
"
```

It start ok when I have weston running, but I was hoping it to work without weston.

clopez, 2020-04-02:
For the RPi you can build this 2 combinations:
* 32-bits, propietary userland-driver, wpebackend-rdk (cog runs directly over the framebuffer, no weston compositor involved). Cog runs full-screen but you need to pass some magical environment variables (check the wiki below) to enable the keyboard and mouse, etc.
* 32 or 64-bits, open source mesa vc4-driver, wpebackend-fdo: Cog runs inside a Weston  window. It perhaps can run directly over the framebuffer instead of over weston, but I'm not sure if the support for that its still there for the RPi.

I documented this here https://github.com/Igalia/meta-webkit/wiki/RPi years ago, some things may have changed since then.

clopez, 2020-04-02:
> I realized that I should have set `DISABLE_VC4GRAPHICS = "1"` as well to enable userland graphics instead of vc4graphics. Trying that now I see wpebackend-fdo fails to compile

wpebackend-fdo can't work with userland-drivers. To use userland-drivers you need to use wpebackend-rdk with the rpi packageconfig option enabled.

kraj, 2020-04-02:
> For the RPi you can build this 2 combinations:
>
>     * 32-bits, propietary userland-driver, wpebackend-rdk (cog runs directly over the framebuffer, no weston compositor involved). Cog runs [...]

yes, I am building that now. Somehow wpebackend-fdo is still trying to build, I need to unreak the dependency somewhere.. on it now

>     * 32 or 64-bits, open source mesa vc4-driver, wpebackend-fdo: Cog runs inside a Weston  window. It perhaps can run directly over the framebuffer instead of over weston, but I'm not sure if the support for that its still there for the RPi.
> 

this I have got working.

> 
> I documented this here https://github.com/Igalia/meta-webkit/wiki/RPi years ago, some things may have changed since then.

clopez, 2020-04-02:
> yes, I am building that now. Somehow wpebackend-fdo is still trying to build, I need to unreak the dependency somewhere.. on it now

Check that you don't have the meta-qt layer on the mix, since it triggers an automatic build of QtWPE that needs wpebackend-fdo

kraj, 2020-04-02:
> > yes, I am building that now. Somehow wpebackend-fdo is still trying to build, I need to unreak the dependency somewhere.. on it now
> 
> Check that you don't have the meta-qt layer on the mix, since it triggers an automatic build of QtWPE that needs wpebackend-fdo

hmm yes, I think we need to convert that into a explicit setting

kraj, 2020-04-02:
OK I can now build options 1 as you described, it still fails to run

```
root@raspberrypi4:~# cog http://google.com
wpe: could not load the impl library. Is there any backend installed?: /usr/lib/libWPEBackend-default.so: undefined symbol: wl_egl_window_destroy
Aborted
root@raspberrypi4:~# cog -P rdk http://google.com

** (cog:350): WARNING **: 01:04:13.867: Could not load: libcogplatform-rdk.so (possible cause: Operation not permitted).

wpe: could not load the impl library. Is there any backend installed?: /usr/lib/libWPEBackend-default.so: undefined symbol: wl_egl_window_destroy
Aborted
```

kraj, 2020-04-02:
ah I See 

`PACKAGECONFIG ?= "wayland" `

in `recipes-browser/wpebackend-rdk/wpebackend-rdk.inc`

I need to override that not append to it.

clopez, 2020-04-02:
Two questions:
1) its /usr/lib/libWPEBackend-default.so a symlink to ibWPEBackend-rdk.so ?
2) what packageconfig options are you passing to wpebackend-rdk? Only one of them should be enabled, so you should only enable the rpi one (don't enable the wayland option).
I see that on WPEBackend-rdk ```wl_egl_window_destroy``` its not used on the rpi backend, if you have it there I guess its because you have enabled other PACKAGECONFIG options than just the rpi one.
```
WPEBackend-rdk $ git grep wl_egl_window_destroy
src/realtek-wl-egl/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
src/wayland-egl/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
src/westeros/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
```

kraj, 2020-04-02:
> Two questions:
> 
>     1. its /usr/lib/libWPEBackend-default.so a symlink to ibWPEBackend-rdk.so ?
> 

its not a symlink

> 2. what packageconfig options are you passing to wpebackend-rdk? Only one of them should be enabled, so you should only enable the rpi one (don't enable the wayland option).
>        I see that on [...]

yeah the catch was I was using `PACKAGECONFIG_append = 'rpi'` which was also leaving wayland in there, it expects to be overridden in this case not appended.

Now the undefined symbol error is gone. but it still does not render. perhaps something else
is missing ( btw I am on pi4/32bit has it been tested on pi4 ? )

> 
> 
> ```
> WPEBackend-rdk $ git grep wl_egl_window_destroy
> src/realtek-wl-egl/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
> src/wayland-egl/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
> src/westeros/renderer-backend.cpp:        wl_egl_window_destroy(m_window);
> ```

kraj, 2020-04-02:
OK eglfs works fine with pi3 but not with pi4

clopez, 2020-04-02:
> ( btw I am on pi4/32bit has it been tested on pi4 ? )

I only tested wpebackend-rdk+userland on rpi2 and rpi3 (32-bits), but not on rpi4.

clopez, 2020-04-02:
There its a third combination:

32 or 64-bits, open source mesa vc4-driver, wpebackend-fdo + cog rendering directly to the framebuffer with DRM/KMS. I see now that cog has a  `drm` option in the `PACKAGECONFIG` options. I guess you have to enable the support to render to work directly with the framebuffer with wpebackend-fdo and without weston. Note: I have not tested it.

kraj, 2020-04-02:
> There its a third combination:
>
> 32 or 64-bits, open source mesa vc4-driver, wpebackend-fdo + cog rendering directly to the framebuffer with DRM/KMS. I see now that cog has a `drm` option in the [...]

interesting let me give drm/kms a shot, this should work on pi4 too

kraj, 2020-04-02:
> OK eglfs works fine with pi3 but not with pi4

seems to not work 

```
root@raspberrypi3:~# cog -P drm http://yoedistro.org

** (cog:323): WARNING **: 03:54:52.583: Platform setup failed: Failed to initialize DRM

** (cog:323): CRITICAL **: 03:54:52.586: WebKitWebViewBackend* webkit_web_view_backend_new(wpe_view_backend*, GDestroyNotify,
 gpointer): assertion 'backend' failed

** (cog:323): ERROR **: 03:54:52.586: Could not instantiate any WPE backend.
```

closed 2021-04-11 by github-actions[bot]