Igalia/meta-webkit #162: Delete old recipes and just keep the last stable version

psaavedra, 2020-04-01:
To keep all the WebKit GTK and WPE recipes (a part of the rest of components) have been manifested like a bad idea. This complicates the CI and there is not a strong justification since the moment that the repository is being branch every time that an new poky version is released.

It could be more simple just to proceed like the others layers just keeping the last stable version for each component.

There is related discussion in this PR: https://github.com/Igalia/meta-webkit/pull/130

clopez, 2020-04-01:
I did that for WebKitGTK in https://github.com/Igalia/meta-webkit/pull/159

psaavedra, 2020-04-03:
I  just created a pull request addressing this issue

PR #164 (2020-04-03): Delete old recipes and just keep the last stable and old-stable versions
* wpewebkit: remove the old  versions but the last two newest
* wpebackend-fdo: remove the old versions but the last two newest
* wpebackend-rdk: remove the old versions but the last two newest
* libwpe: remove the old versions but the last two newest
* cog: remove the old versions but the last two newest
* ci: update CI tasks.

Fix: #162
commit 18488ef3 (2020-04-03): Delete old recipes and just keep the last stable and old-stable versions

* wpewebkit: remove the old  versions but the last two newest
* wpebackend-fdo: remove the old versions but the last two newest
* wpebackend-rdk: remove the old versions but the last two newest
* libwpe: remove the old versions but the last two newest
* cog: remove the old versions but the last two newest
* ci: update CI tasks.

Fix: #162
closed 2020-04-03 by psaavedra