Igalia/meta-webkit #165: Rpi3 + WPE support for GStreamer 1.16.2 + wpebackend-fdo backend + Enabling accelerated video decoding + Yocto zeus

aurelihein, 2020-04-03:
I try to find a working build with accelerated video decoding with wpe
The problem is that the rendering(not sure it is the correct work) doesn't work

I already have tried rdk backend with accelerated video decoding but the video played is not enough fluent ...

Here is the way I create my yocto build :

Get repos :
```
mkdir 20200402-wpe-original-meta-webkit-zeus
cd 20200402-wpe-original-meta-webkit-zeus
git clone -b zeus https://github.com/aurelihein/meta-bouin sources/meta-bouin
git clone -b zeus git://github.com/OSSystems/meta-gstreamer1.0.git sources/meta-gstreamer1.0
git clone -b zeus git://git.openembedded.org/meta-openembedded sources/meta-openembedded
git clone -b zeus git://git.yoctoproject.org/meta-raspberrypi sources/meta-raspberrypi
git clone -b master git://github.com/Igalia/meta-webkit sources/meta-webkit
git clone -b zeus https://git.yoctoproject.org/git/poky sources/poky

MACHINE="raspberrypi3" && source sources/poky/oe-init-build-env build
```
Now we should be into build dir
Let's update local.conf :
```
cat << 'EOTEOT' >> conf/local.conf
MACHINE = "raspberrypi3"
MACHINE_FEATURES_append = " vc4graphics"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL_append = " wpewebkit cog"
IMAGE_INSTALL_append = " htop iotop tree"
IMAGE_INSTALL_append = " gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly"
LICENSE_FLAGS_WHITELIST = "commercial_ffmpeg commercial_x264 commercial_gstreamer1.0-plugins-ugly"
LICENSE_FLAGS_WHITELIST += "commercial"
EOTEOT
```
Let's update bblayers.conf
```
cat << 'EOTEOT' >> conf/bblayers.conf
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /workdir/sources/poky/meta \
  /workdir/sources/poky/meta-poky \
  /workdir/sources/poky/meta-yocto-bsp \
  /workdir/sources/poky/../meta-openembedded/meta-oe \
  /workdir/sources/poky/../meta-openembedded/meta-python \
  /workdir/sources/poky/../meta-openembedded/meta-multimedia \
  /workdir/sources/poky/../meta-openembedded/meta-networking \
  /workdir/sources/poky/../meta-openembedded/meta-webserver \
  /workdir/sources/poky/../meta-gstreamer1.0 \
  /workdir/sources/poky/../meta-raspberrypi \
  /workdir/sources/poky/../meta-webkit \
  "
EOTEOT
```
Let's create a new meta to add some custom parameters :
```
bitbake-layers create-layer ../sources/meta-bouin
bitbake-layers add-layer ../sources/meta-bouin
```

#into meta-bouin :

```
mkdir -p recipes-multimedia/gstreamer/
cat << 'EOTEOT' > recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.16.%.bbappend
DEPENDS += "fdk-aac"
EXTRA_OECONF_remove = "--disable-fdk_aac"
EXTRA_OECONF += "--enable-fdk_aac"

DEPENDS += "wpebackend-fdo wpewebkit"
EXTRA_OECONF_remove = "--disable-wpe"
EXTRA_OECONF += "--enable-wpe"
EOTEOT

mkdir -p recipes-multimedia/fdk-aac/
cat << 'EOTEOT' > recipes-multimedia/fdk-aac/fdk-aac_2.0.1.bb
SUMMARY = "fdk-aac package provides the Fraunhofer FDK AAC library"
LICENSE = "FDK_AAC_Codec_Library_for_Android"
LIC_FILES_CHKSUM = "file://NOTICE;beginline=1;endline=20;md5=047e46453be4812e5a7f22881ce2edba"

SRC_URI = "https://downloads.sourceforge.net/opencore-amr/fdk-aac-2.0.1.tar.gz"

SRC_URI[md5sum] = "e8b0b38e837df455b8a1ba75417ff0ad"
SRC_URI[sha256sum] = "840133aa9412153894af03b27b03dde1188772442c316a4ce2a24ed70093f271"

inherit autotools pkgconfig
EOTEOT
```

I launch the build with :
`bitbake core-image-weston`

FYI the git SHA I am using last zeus branches from today 3th of April 2020 :
```
#Build Configuration:
#BB_VERSION           = "1.44.0"
#BUILD_SYS            = "x86_64-linux"
#NATIVELSBSTRING      = "ubuntu-18.04"
#TARGET_SYS           = "arm-poky-linux-gnueabi"
#MACHINE              = "raspberrypi3"
#DISTRO               = "poky"
#DISTRO_VERSION       = "3.0.2"
#TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
#TARGET_FPU           = "hard"
#meta                 
#meta-poky            
#meta-yocto-bsp       = "zeus:5531ffc5668c2f24b9018a7b7174b5c77315a1cf"
#meta-oe              
#meta-python          
#meta-multimedia      
#meta-networking      
#meta-webserver       = "zeus:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
#meta-gstreamer1.0    = "master:bf562ab7bb4dd68e486c71be6ebcf9d872f63b33"
#meta-raspberrypi     = "zeus:0e05098853eea77032bff9cf81955679edd2f35d"
#meta-webkit          = "zeus:57955c3a39ef5452db70176102a526f09588a546"
```

The way I burn the image :
```
SDCARD_IMAGE=tmp/deploy/images/raspberrypi3/core-image-weston-raspberrypi3.rpi-sdimg
ls -lhH ${SDCARD_IMAGE}
SDCARD_SLOT=/dev/sde
lsblk ${SDCARD_SLOT}
sudo dd if=${SDCARD_IMAGE} of=${SDCARD_SLOT} bs=4M conv=notrunc,noerror status=progress ; sync
```

Once the board is booted :

We can verify wpe support for gstreamer :
```
root@raspberrypi3:~# gst-inspect-1.0 | grep wpe
wpe:  wpesrc: WPE source
```
```
root@raspberrypi3:~# gst-inspect-1.0 --version
gst-inspect-1.0 version 1.16.2
GStreamer 1.16.2
Unknown package origin
```

When I launch :
```
export WAYLAND_DISPLAY="wayland-0"
export XDG_RUNTIME_DIR="/run/user/0"
gst-launch-1.0 -v wpesrc location="https://www.google.fr" ! queue ! glimagesink

```

This is What I get :
[image]
[image]

Another problem is when I try to play a video from a web server with this page into it :
```

cat <<'EOTEOT' > page.html
<html>
<head>
</head>
<body>
	<video width="1920" height="1080" autoplay preload="auto" loop >
		<source src="big_buck_bunny_1080p_surround.mp4" type="video/mp4">
		Your browser does not support the video tag.
	</video> 
</body
</html>
EOTEOT
```

and the well know big_buck_bunny_1080p_surround.avi converted into mp4

#start debugging with :

GST_DEBUG="3,webkit*:6" GST_DEBUG_FILE=/tmp/gst.log GST_DEBUG_DUMP_DOT_DIR=/tmp cog -P fdo http://url_to/page.html

This is what I get :
```
Cog-Message: 15:47:46.603: <http://192.168.1.101/benchvideo/page.html> Load started.
Cog-Message: 15:47:46.890: <http://192.168.1.101/benchvideo/page.html> Loading...

(cog:1248): Cog-WARNING **: 15:48:00.346: <http://192.168.1.101/benchvideo/page.html> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 15:48:02.755: <http://192.168.1.101/benchvideo/page.html> Load started.
Cog-Message: 15:48:02.778: <http://192.168.1.101/benchvideo/page.html> Loading...
Cog-Message: 15:48:02.843: <http://192.168.1.101/benchvideo/page.html> Loaded successfully.
```

```
root@raspberrypi3:~# cat /tmp/gst.log 
0:00:00.176042604  1295 0x6e0038a0 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:462:gst_element_factory_make: no such element factory "avdec_aac"!
0:00:00.176197291  1295 0x6e0038a0 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:462:gst_element_factory_make: no such element factory "avdec_aac_fixed"!
0:00:00.176250260  1295 0x6e0038a0 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:462:gst_element_factory_make: no such element factory "avdec_aac_latm"!
```

Thank you for your support / help / suggestions / time
Regards,

Aurelien BOUIN

aurelihein, 2020-04-05:
Yes `gltestsrc` display the mire and is working : 
`gst-launch-1.0 -v gltestsrc pattern=smpte ! glimagesink`
[image]

And yes cog seems to work until I want to play video :
GST_DEBUG="3,webkit*:6" GST_DEBUG_FILE=/tmp/gst.log GST_DEBUG_DUMP_DOT_DIR=/tmp cog -P fdo https://youtube.com
[image]
[image]

Here is the debug tmp folder : 
[tmp_youtube_fail.zip](https://github.com/Igalia/meta-webkit/files/4433144/tmp_youtube_fail.zip)
Not much into `gst.log` file than previously

May I miss any gstreamer plugin to bring with the yocto image ?

philn, 2020-04-05:
Try a simple use-case before youtube please. This for instance: https://www.w3.org/2010/05/video/mediaevents.html

It's odd the logs are almost empty, this isn't expected.

philn, 2020-04-05:
`gst-inspect-1.0 > plugins.log` and attach the file here please.

aurelihein, 2020-04-05:
I thought that the page.html I wrote on the first part of my issue was not that complicate, since it just asked to play a video
Here is the plugins.log :
[plugins.log](https://github.com/Igalia/meta-webkit/files/4434280/plugins.log)
Thank you for your help and support

philn, 2020-04-05:
Ok, we wouldn't a stack-trace of the crash then.

aurelihein, 2020-04-05:
I don't understand, what do you want ?

philn, 2020-04-05:
`(cog:1248): Cog-WARNING **: 15:48:00.346: <http://192.168.1.101/benchvideo/page.html> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.`

This means WPEWebProcess has crashed. Can you get a stack-trace with gdb/gdbserver?

aurelihein, 2020-04-06:
For an unknown reason it seems that the instance is killed while debugging :
I launch it with :
`gdb --args cog -P fdo http://192.168.1.101/benchvideo/page.html`
Here is the log : 
[20200406-gdb_debug_killed.txt.txt](https://github.com/Igalia/meta-webkit/files/4437134/20200406-gdb_debug_killed.txt.txt)

Is there any stable version somewhere ? Do you have any git sha as reference for zeus ?

philn, 2020-04-06:
This is supposed to be the stable version.
What you need to debug is the WebProcess, not cog:

1. Remove `autoplay` from your html page. Add `controls`
2. Start cog
3. In another shell, attach to `WPEWebProcess` with gdb
4. Press play in the web-page

aurelihein, 2020-04-06:
I just have removed `autoplay` and add `controls`
I have verified the behavior on my host computer that it waits that we press play for playing :+1: 
When I launch my cog command : `cog -P fdo http://192.168.1.101/benchvideo/page.html`
I don't see anything and already have the crash :
```
root@raspberrypi3:~# export WAYLAND_DISPLAY="wayland-0"
root@raspberrypi3:~# export XDG_RUNTIME_DIR="/run/user/0"
root@raspberrypi3:~# cog -P fdo http://192.168.1.101/benchvideo/page.html
Cog-Message: 08:54:58.226: <http://192.168.1.101/benchvideo/page.html> Load started.
Cog-Message: 08:54:58.271: <http://192.168.1.101/benchvideo/page.html> Loading...

(cog:507): Cog-WARNING **: 08:55:02.931: <http://192.168.1.101/benchvideo/page.html> Crash!: The renderer process crashed. Reloading the page may fix intermittent failures.
Cog-Message: 08:55:03.424: <http://192.168.1.101/benchvideo/page.html> Load started.
Cog-Message: 08:55:03.434: <http://192.168.1.101/benchvideo/page.html> Loading...
Cog-Message: 08:55:03.484: <http://192.168.1.101/benchvideo/page.html> Loaded successfully.
```

aurelihein, 2020-04-06:
@philn : Ok I propose to use the url :  https://www.w3.org/2010/05/video/mediaevents.html the start doesn't fail... Sorry for that question but how do you attach to `WPEWebProcess` with gdb

aurelihein, 2020-04-06:
Ok I figured out, to whom is looking for it :

aurelihein, 2020-04-06:
@philn I also receive a Killed : 
```
root@raspberrypi3:~# gdb -p $(pidof WPEWebProcess)
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-poky-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 1666
[New LWP 1670]
[New LWP 1671]
[New LWP 1673]
[New LWP 1674]
[New LWP 1685]
[New LWP 1686]
[New LWP 1688]
[New LWP 1689]
[New LWP 1690]
[New LWP 1695]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
Killed
```

philn, 2020-04-06:
Once the process was killed, type this in the `gdb` prompt... `thread apply all bt`

aurelihein, 2020-04-06:
Once killed I am out of gdb prompt ...
So I also try this without sucess : 
`ulimit -c unlimited `
`cog -P fdo https://www.w3.org/2010/05/video/mediaevents.html`
#Click to start video and get the crash
`gdb --batch -ex "thread apply all bt full" /usr/libexec/wpe-webkit-1.0/WPEWebProcess core &> backtrace.txt`
here is the backtrace.txt : I don't get anything in it :
```
root@raspberrypi3:~# cat backtrace.txt 
[New LWP 2222]
[New LWP 2219]
[New LWP 2204]
[New LWP 2220]
[New LWP 2223]
[New LWP 2208]
[New LWP 2221]
[New LWP 2229]
[New LWP 2207]
[New LWP 2232]
[New LWP 2227]
[New LWP 2228]
[New LWP 2235]
[New LWP 2234]
[New LWP 2230]
[New LWP 2236]
[New LWP 2231]
[New LWP 2209]
[New LWP 2233]
[New LWP 2218]
[New LWP 2199]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
```

aurelihein, 2020-04-06:
This seems to be a RAM memory issue, this is OOM killer that kill `gdb`, so I have disabled it : 
`vm.oom-kill = 0`
But even though it seems to not have enough memory ...
Once I launch `gdb -p $(pidof WPEWebProcess)`
Using `while true; do free -mh; sleep 1; done` I see the memory going down to 0 , and we see the gdb crashing ...
Any idea ?

michaelhanin, 2020-04-06:
Hello,

I have the same issue with the last wpewebkit 2.28 ( using buildroot )

I can use cog ( weston + fdo  ) to view website -> it work

But it crash when <video> is in html

play video with gst-play-1.0 -> it work

---------
ps : previous build with wpewebkit 2.27 was OK with video

--> i test with raspberry pi 4

philn, 2020-04-06:
`gst-play-1.0 --videosink glimagesink /path/to/video.mp4` works?

philn, 2020-04-06:
@mickdev this issue is about yocto, not buildroot. Please don't hijack this bug, yours might be unrelated.

aurelihein, 2020-04-06:
@philn : wov crash with gst-play-1.0 !
```
root@raspberrypi3:~# gst-play-1.0 --videosink glimagesink 10_24fps_5M_nb-fastdecode.mp4 
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/root/10_24fps_5M_nb-fastdecode.mp4
0:00:00.0 / 0:02:00.0       
** (gst-play-1.0:1710): CRITICAL **: 11:00:47.238: gst_gl_memory_setup_buffer: assertion '!wrapped_data || n_mem * views == n_wrapped_pointers' failed
Segmentation fault0.0
```

aurelihein, 2020-04-06:
Also `gst-launch-1.0 -v filesrc location=10_24fps_5M_nb-fastdecode.mp4 ! parsebin name=p ! v4l2h264dec capture-io-mode=4 ! 'video/x-raw, format=(string)RGB16' ! glimagesink` give the same blinky image than the one trying to display google webpage
Any idea ?

philn, 2020-04-06:
Please apply these patches to your gst-plugins-base recipe: https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/-/merge_requests/615

aurelihein, 2020-04-06:
@philn patches applied this following line doesn't hang anymore but ... still this issue : 
`gst-launch-1.0 -v filesrc location=10_24fps_5M_nb-fastdecode.mp4 ! qtdemux ! queue ! h264parse ! v4l2h264dec capture-io-mode=4 ! 'video/x-raw, format=(string)RGB16' ! glimagesink`
[IMG_2619.zip](https://github.com/Igalia/meta-webkit/files/4438253/IMG_2619.zip)
And speaking of cog there is still this crash : 
`Crash!: The renderer process crashed`

philn, 2020-04-06:
I don't know, something looks wrong with the graphics driver. I don't have a rpi3/4 so can't help further. Sorry!

aurelihein, 2020-04-06:
@philn Thank you very much for all the time spent for me already ! As soon as I solve this I will propose a PR
If you have any suggestion on the way I should look at ... I will look for older Rpi graphics driver
If you need any yocto image to test please ask
Best regards,
Aurelien BOUIN

psaavedra, 2020-04-07:
Using the wpe from trunk and the meta-perf-browser (https://gitlab.com/browsers/meta-perf-browser) layer the Youtube works so far.  

Note that the meta-perf-browser layer includes several patches in mesa and gstreamer which could be related with the rpi3 user experience (@philn , could you confirm/discard this?):

* https://gitlab.com/browsers/meta-perf-browser/-/tree/nightly/recipes-multimedia/gstreamer
* https://gitlab.com/browsers/meta-perf-browser/-/tree/nightly/recipes-graphics/mesa

``` sh
MACHINE=raspberrypi3-mesa
DISTRO=browsers
BUILD_ENV=${MACHINE}-wpe-nightly
cd /home/bot/yocto
repo init -u https://gitlab.com/browsers/meta-perf-browser.git -m manifest-zeus.xml -b nightly
repo sync --force-sync
source setup-environment ${BUILD_ENV} ${MACHINE} ${DISTRO} default mesa-wpe-nightly --update-config
IMAGE=core-image-weston-wpe
bitbake ${IMAGE}
```

(prebuilt image: https://wk-contrib.igalia.com/yocto/meta-perf-browser/browsers/nightly/images/raspberrypi3-mesa/core-image-weston-browsers-raspberrypi3-mesa-20200405041232.rootfs.rpi-sdimg)

```
export WAYLAND_DISPLAY=wayland-0
export XDG_RUNTIME_DIR=/tmp/xdg-900-runtime-dir/
cog -P fdo https://youtube.com
```

* Youtube Video: https://mega.nz/file/CI53lISA#IfHElvKYKi0MJkKg7YfLP4iEyGzKy3WUH0h3_uMZEeU
* HTML5 Video Events and API (https://www.w3.org/2010/05/video/mediaevents.html): https://mega.nz/file/rYxQUArY#totohfr4bW1pLc1NhgtJ1vyfsKeMpm-JpMJM4hxRKDo

Could you verify if works for 2.28 as well. I suppose it will work but I didn't test it.

``` bash
source setup-environment ${BUILD_ENV} ${MACHINE} ${DISTRO} default mesa-wpe-2_28    --update-config
```

The bitbake meta-layers hashes:

```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-10"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3-mesa"
DISTRO               = "browsers"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "zeus:5531ffc5668c2f24b9018a7b7174b5c77315a1cf"
meta-oe              
meta-multimedia      
meta-perl            
meta-python          
meta-networking      
meta-filesystems     
meta-webserver       = "HEAD:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta-clang           = "HEAD:c8c1d5f915eeddf28b690ae4f7b3908d1d40660f"
meta-browser         = "HEAD:830ef438e81ba5fc915b1855e69f02b2c286b21a"
meta-rust            = "HEAD:1a3625364cdca31cb37f5c47b678523277b1374c"
meta-updater         = "HEAD:416a934b0555a1131490644a7edcbb644a934e76"
meta-security        = "HEAD:ecd8c30af6b1b957b4fa814d3f25c40f060b3ac1"
meta-raspberrypi     = "HEAD:0e05098853eea77032bff9cf81955679edd2f35d"
meta-freescale       = "HEAD:4e7e8ef4676c45f502dd2a0bd5179c3b777362a1"
meta-freescale-3rdparty = "HEAD:076c3132bb244b73c0f54b698d64c48b1ffbb58f"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-intel           = "HEAD:faced19dda5332cce9164903b250db5aa9b86259"
meta-cloud-services  = "HEAD:b8c25f63ab9124fe911165205d17808b9a7dead0"
meta-gstreamer1.0    = "HEAD:bf562ab7bb4dd68e486c71be6ebcf9d872f63b33"
meta-webkit          = "issue_162:c282f2a600870b627a9e5958930907110d5f2191"
meta-perf-browser    = "HEAD:9f7012d27ba66be10ab954bd991ab7748cbef655"
```

Below, the gstreamer selected packages:

```
=== Matching recipes: ===
gstreamer1.0:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-libav:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
  meta-freescale       1.14.0
gstreamer1.0-meta-base:
  meta-gstreamer1.0    1.0
  meta                 1.0
gstreamer1.0-omx:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-plugins-bad:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-plugins-base:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-plugins-good:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-plugins-ugly:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-python:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-rtsp-server:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
gstreamer1.0-vaapi:
  meta-gstreamer1.0    1.16.2
  meta                 1.16.1
```

philn, 2020-04-07:
Ah yeah the mesa vc4 patches are relevant. For some reason I thought they were already in meta-webkit

aurelihein, 2020-04-07:
@philn FYI I am still alive on this, I recompile my yocto build on zeus but with a MACHINE="raspberrypi3-64" instead of MACHINE="raspberry3" and I don't have the rendering problem anymore.
The gst-launch are now playing well ...
The `gst-launch-1.0 -v wpesrc location="https://www.google.fr" ! queue ! glimagesink` is now rendering well also
But I still have the crash when trying to play html5 video ... And I still can't attach a gdb on WPEWebProcess (oom killer do its job too well ...)
However I have a strange behavior with my rpi image. I can boot it only one time ! Then I need to reflash the sdcard image to retry ... never seen that before ... have you experienced such behavior before ? and I have checked on the already booted sdcard any files changes ... I can't find any ...

aurelihein, 2020-04-07:
@psaavedra I didn't see your message before posting mine ! Thank you very much I will give it a try tomorrow ! Thank you !!!

aurelihein, 2020-04-08:
@psaavedra & @philn Thank you very much, video acceleration is now working !
I had to :
1. add the mesa 20.0.0 version like you suggested @psaavedra 
2. add the gstreamer glupload fix like you suggested @philn 
3. add a patch for wpewebkit 2.28 to avoid the cairo crash on version 1.16.0
@philn what is the best way to share this ? the The purpose is to share a stable build on rpi3 in branch zeus using your meta-webkit
PS : @psaavedra nice meta-perf-browser !

mickdev, 2020-04-09:
@aurelihein hello,

i cannot find this patch "add a patch for wpewebkit 2.28 to avoid the cairo crash on version 1.16.0", you created it, can you share it ? thx  ?

If the patch is about wpwebkit,  you can fill a request directly to webkit.org, and propose it upstream here for a patch for wpewebkit in meta-webkit.

@psaavedra i downloaded from your link core-image-weston-browsers-raspberrypi3-mesa-20200405041232.rootfs.rpi-sdimg  but it seems to be a webkit 2.27.** not 2.28 ( it work on rpi3 )

psaavedra, 2020-04-09:
> @psaavedra i downloaded from your link core-image-weston-browsers-raspberrypi3-mesa-20200405041232.rootfs.rpi-sdimg but it seems to be a webkit 2.27.** not 2.28 ( it work on rpi3 )
source setup-environment ${BUILD_ENV} ${MACHINE} ${DISTRO} default mesa-wpe-2_28    --update-config

Actually the image is built from WebKit trunk (date: 2020-04-05 04:12:32 CEST) so it should be something more like 2.29.X.

You can try to build your own image just using the following commands:

``` sh
MACHINE=raspberrypi3-mesa
DISTRO=browsers
BUILD_ENV=${MACHINE}-wpe-nightly
cd /home/bot/yocto
repo init -u https://gitlab.com/browsers/meta-perf-browser.git -m manifest-zeus.xml -b nightly
repo sync --force-sync
source setup-environment ${BUILD_ENV} ${MACHINE} ${DISTRO} default mesa-wpe-2_28    --update-config

IMAGE=core-image-weston-wpe
bitbake ${IMAGE}
```

More info:
* https://gitlab.com/browsers/meta-perf-browser/-/wikis/core-image-weston-browsers-for-rpi3
* https://gitlab.com/browsers/meta-perf-browser/-/blob/master/README.md

aurelihein, 2020-04-09:
@mickdev I am talking about this one : https://github.com/WebPlatformForEmbedded/WPEWebKit/commit/71aab10b92eede51527e86b6c01abb21fe7d4f96

aurelihein, 2020-04-10:
If interested I just updated the instruction on my README.md : https://github.com/aurelihein/meta-bouin

mickdev, 2020-04-14:
@aurelihein thank you, i just compiled the raspi-3-64bits-mesa with your readme

-> only the syntel video work on the w3.org (?)
-> no detection of event media ended on the w3.org
-> maybe problem with the loop attribute (?) ( syntel video loop .. but should not .. )
-> between 25% and 50% cpu usage ( between 5% and 10% with omx ... )

wezzynl, 2021-02-22:
@psaavedra Hey Pablo, those references you're mentioning seem to be 404'ing at the moment. Is that meta-perf-browser layer accessible somewhere?

psaavedra, 2021-02-23:
> @psaavedra Hey Pablo, those references you're mentioning seem to be 404'ing at the moment. Is that meta-perf-browser layer accessible somewhere?

Hello, that repo is not longer available but the changes commented in this issue are nowadays mostly integrated in the upstream of each component: mesa, webkit ...

wezzynl, 2021-02-23:
> upstream of each component: mesa, webkit

Are you sure? I'm on Dunfell and I needed to patch `mesa` otherwise it would crash, luckily @aurelihein had those patches in his repo. Might be worthwhile to mention this in the wiki maybe? It does mention this issue and, I guess, eventually people are going to figure out what is going on but still... ;-)

Thanks for your response though, much appreciated.

psaavedra, 2021-02-23:
For the case of Mesa, the patch is this
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/3997.

closed 2021-11-09 by psaavedra