Igalia/meta-webkit #178: Hard to decode error while building either webkit - rpi3 warrior

blakelloyd-coderus, 2020-04-20:
Currently trying to build wpe for a rpi3 32bit (have tried both options with hardcoded values and all sorts) but I keep getting to the same error:

[3655/4217] ccache /home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++  -DBUILDING_WITH_CMAKE=1 -DBUILDING_WPE__=1 -DBUILDING_WebCore -DGETTEXT_PACKAGE=\"WPE\" -DHAVE_CONFIG_H=1 -DJSC_GLIB_API_ENABLED -DSTATICALLY_LINKED_WITH_PAL=1 -IDerivedSources/ForwardingHeaders -I. -IDerivedSources/WebCore -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/airplay -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/applepay -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/applepay/paymentrequest -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/applicationmanifest -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/beacon -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/cache -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/credentialmanagement -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/encryptedmedia -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/encryptedmedia/legacy -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/entriesapi -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/fetch -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/geolocation -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/indexeddb -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/indexeddb/client -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/indexeddb/server -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/indexeddb/shared -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediacapabilities -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediacontrols -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediarecorder -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediasession -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediasource -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediastream -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/mediastream/libwebrtc -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/navigatorcontentutils -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/notifications -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/paymentrequest -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/plugins -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/quota -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/speech -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/streams -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webaudio -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webauthn -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webauthn/cbor -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webauthn/fido -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webdatabase -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webdriver -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webgpu -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webgpu/WHLSL -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webgpu/WHLSL/AST -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/websockets -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/Modules/webvr -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/accessibility -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/accessibility/isolatedtree -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/animation -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/bindings -I/home/jenkins/workspace/branch/poky/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/wpewebkit/2.26.2-r0/wpewebkit-2.26.2/Source/WebCore/bindings/js -I/home/jenkins/workspace/bra [... cut ...] bCore/CMakeFiles/WebCore.dir/__/__/DerivedSources/WebCore/unified-sources/UnifiedSource-c57e08af-7.cpp.o -MF Source/WebCore/CMakeFiles/WebCore.dir/__/__/DerivedSources/WebCore/unified-sources/UnifiedSource-c57e08af-7.cpp.o.d -o Source/WebCore/CMakeFiles/WebCore.dir/__/__/DerivedSources/WebCore/unified-sources/UnifiedSource-c57e08af-7.cpp.o -c DerivedSources/WebCore/unified-sources/UnifiedSource-c57e08af-7.cpp
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.

clopez, 2020-04-20:
Since the WebKit build its parallelized the error will not be on the last line.
You need to find the complete error log file (the log.do_compile file) and then look on that file for lines containing the word error (case insensitive)

blakelloyd-coderus, 2020-04-20:
Ahh, well that makes way more sense!

`FAILED: Source/JavaScriptCore/CMakeFiles/JavaScriptCore.dir/dfg/DFGSpeculativeJIT.cpp.o `

`arm-poky-linux-gnueabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.`

Seems to be the actual issue

clopez, 2020-04-20:
> arm-poky-linux-gnueabi-g++: fatal error: Killed signal terminated program cc1plus compilation terminated.

This usually mean you ran out of ram and the kernel killed the gcc process. You need at least 2GB of RAM per CPU vcore to build WebKit. If you don't have enough then try to lower the number of build threads.

For example you can set in local.conf
```
# Default to setting automatically based on cpu count
BB_NUMBER_THREADS = "4"

# Default to setting automatically based on cpu count
PARALLEL_MAKE = "-j 4"
```

With that number (4) you would need 8GB of RAM to build WebKit

blakelloyd-coderus, 2020-04-20:
Ahh ok, I thought I was just missing the error that was there!

It was running on 12 cores with 10GB RAM so that may well have been the issue, currently rebuilding. Thanks!

blakelloyd-coderus, 2020-04-20:
Well that seems to fix it, thanks for the help!

closed 2020-04-20 by blakelloyd-coderus