Igalia/meta-webkit #179: Touchscreen cursor does not move

blakelloyd-coderus, 2020-04-24:
Currently running meta-webkit with cog on warrior on a Pi 3 with the userland drivers and having some issues with the touch screen.

Even after setting
export WPE_BCMRPI_TOUCH=1
export WPE_BCMRPI_CURSOR=1
before starting cog, I can't seem to get the touchscreen to work

Checking /dev/input shows readings as expected, and if I plug a mouse in, it moves the cursor (even to the point of if I move the cursor with the mouse and then touch the screen it will simulate a click but no movement).

zdobersek, 2020-04-30:
What kind of touch device are you using?

blakelloyd-coderus, 2020-04-30:
I'm using an ELO 1593L

blakelloyd-coderus, 2020-05-28:
Anybody have any ideas for this? As the touchscreen works fine in the full sato build.

psaavedra, 2020-06-08:
What kind of `PREFERRED_PROVIDER` you have for `virtual/wpebackend`?

blakelloyd-coderus, 2020-06-08:
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"

psaavedra, 2021-02-05:
even with some delay, @blakelloyd-coderus this reply in https://github.com/Igalia/meta-webkit/issues/217 could be helpful to answer your doubts.

closed 2021-02-05 by psaavedra